Koenkk/zigbee2mqtt #2278: Documentation issue -> peering CC2531 in router mode

Maik-100, 2019-11-04:
Hi.
I try to peer a CC2531 in router mode to an other CC2531 in coordinator mode.
The LED told me, no peering occurred.

I flashed a second CC2531 to router mode, perhaps the first one had a defect. But same behaviour. With USB and diag support I recognize, right channel and right panID, but no peering.

Accidently I changed in the room the position of the CC2531, both CC2531 are more close to each other. Suddently green LEDs comes and long red flashing -> peering occurred.

In documentation of my other zigbee devices (they comes from IKEA), they write, **if you want to peer, you have to bring devices close to each other.**

So, please write in your documentation this info in a short way. Thanks.

Koenkk, 2019-11-05:
Could you make a pr? The link in the docs shows which file to edit

Maik-100, 2019-11-05:
Hi Koenkk,

first this page need the info:
https://github.com/Koenkk/Z-Stack-firmware/tree/master/router
Point: Pairing

And I think same for the CC2530, but I do not have such device, so I am not able to test:
https://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html
Point 4. Done!

And of course:
https://ptvo.info/cc2531-based-router-firmware-136/
But I think this is not your Web-Page, right?

What is a "pr",  pull request?
like: https://github.com/Koenkk/zigbee2mqtt/issues/1086#issuecomment-469373021

Koenkk, 2019-11-05:
I've checked but this is not a requirement in order to pair this device. So I'm not sure what happened in your case.

Maik-100, 2019-11-07:
Yes, that is the problem, no dokumentation to this toppic (distance between coordinator and router in the moment of pairring).
I put the dokumentation of my devices
[IKEA_Tradfri.pdf](https://github.com/Koenkk/zigbee2mqtt/files/3818256/IKEA_Tradfri.pdf)
into this issue. (5 cm in the moment of pairring and normal/operation distance 10 m).
If I pairred the motion detector, the LED-light, and the remote control with the CC2531 in coordinator mode I respected the small distance and no problems. But last saturday I tried to pair CC2531 in router mode, I needed hours, because I tryed normal/operation distance to pair.

closed 2020-01-13 by stale[bot]