Koenkk/zigbee2mqtt #2279: Help Adding New Device Lightify Dimming Switch

warllo54, 2019-11-05:
I am trying to add the Sylvania Lightify Dimming Switch to Zigbee2mqtt. I was able to pair the device per the logs. I have uploaded the logs to pastbin

https://pastebin.com/j7RViMeY

I added the following to my devices.js

`    {
        zigbeeModel: ['LIGHTIFY Dimming Switch'],
        model: '73743',
        vendor: 'Sylvania',
        description: 'Lightify Dimming Switch',
        fromZigbee: [], // We will add this later
        toZigbee: [],
    },`

I then restarted zigbee2mqtt and tried to perform an action however nothing appears in the log. What am I missing? Here is my docker config in case it may be helpful.

https://pastebin.com/ctggktxH

Kryzek, 2019-11-05:
You should have logging level set to debug to see more information.
I'm not familiar with docker environment, but zigbee2mqtt configuration.yaml -file
has an option to set log_level under advanced -section.
```
advanced:
  # Optional: Logging level, options: debug, info, warn, error (default: info)
  log_level: debug
```

warllo54, 2019-11-05:
@Kryzek Thanks for the suggestion. I enabled debug logging but I'm still not seeing anything. I'm guessing it has something to do with the text below but I'm not sure how to figure out what additional config is needed.

`In case your device is not reporting anything, it could be that this device requires additional configuration. This can be done by adding a configure: section (examples here). In case your device is a contact or motion sensor, it could be that this is an IAS device. Example of an IAS configure: section:

configure: (ieeeAddr, shepherd, coordinator, callback) => {
    const device = shepherd.find(ieeeAddr, 1);
    const actions = [
        (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
        (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
    ];

    execute(device, actions, callback);
},`

warllo54, 2019-11-07:
I was able to make some changes to my config and make a different remote/switch work. I no longer care to add support for this.

closed 2019-11-07 by warllo54