Koenkk/zigbee2mqtt #2280: zigbee2mqtt crashes after a while

Bruceforce, 2019-11-05:
# Bug Report
## What happened
zigbee2mqtt crashes after a while after upgrade from docker tag `latest` to `latest-dev`

## What did you expect to happen
zigbee2mqtt shouldn't crash

## How to reproduce it (minimal and precise)
Just let it run for a few hours / days.

## Debug Info
zigbee2mqtt version: via docker tag `latest-dev`
CC253X firmware version: 20190608

Sorry I can't provide debug logs since the crash happens only after a few days. Do these logs help?
docker-compose log: https://gist.github.com/Bruceforce/7475b61be16e2c0a6772d145662ad056
npm debug log: https://gist.github.com/Bruceforce/fbbf4c475c9912ce69b7089a7c112e17

Koenkk, 2019-11-05:
These logs don't help, I need the logs at the moment it crashes.

Bruceforce, 2019-11-05:
Anything special to debug this or should I just set log level to debug in zigbee2mqtt and wait till it crashes?

Koenkk, 2019-11-06:
Run the container with `-e DEBUG=zigbee-herdsman:*`.

Bruceforce, 2019-11-06:
I started it with just loglevel set to debug without the environment variable yesterday, before you posted this. I got this log: 
https://gist.githubusercontent.com/Bruceforce/23818fd9a557041f3e681896b82a0262/raw/fae3b64764b48c2e5c19bd1ee4c7b0cd13e549ec/log.txt

Does this already help or do you still need me to set the environment variable? I'll do so anyways, but I can't get back to you until it crashes again.

Koenkk, 2019-11-06:
Can you disable the report and availability feature?

Bruceforce, 2019-11-06:
Should I first wait until it crashes again with the environment variable set or disable these features right away?

I want to emphasize that these features didn't cause a crash on the old zigbee2mqtt versions and it would be great if they would work with zigbee herdsman as well.

Koenkk, 2019-11-06:
I only expect that they can provide some troubles on startup, but I don't expect it to have anything to do with the crash after some uptime. So if zigbee2mqtt starts fine don't disable them.

Bruceforce, 2019-11-06:
Yes they indeed cause some trouble at the start (devices show unavailable for a few minutes) but this settles after around 5 minutes max. 

So I let it run with the environment set and post the new log when it crashes again. After that I'll try to disable report and availability just to make sure.

Bruceforce, 2019-11-07:
Today it crashed again. Unfortunately zigbee2mqtt somehow didn't create a logfile by itself, so all I've got for you is the compose log. But I think it should contain the same as the zigbee2mqtt log. I included only a few minutes before the crash into the gist. If you want the full log, see the here [compose_logs.txt.gz](https://github.com/Koenkk/zigbee2mqtt/files/3821691/compose_logs.txt.gz). 

FYI: I noticed that it crashed right after I turned on power of three GU10 bulbs (ceiling spots).

https://gist.github.com/Bruceforce/3604d48e12afc9ce61657b7159d40d8d

Does this help or should I try something else?

Koenkk, 2019-11-08:
The stick crashes, please disable the report and availability feature and check if things improve.

Bruceforce, 2019-11-08:
Seems to be working - at least for now. But will this be the long time solution? That would be really bad, because without reporting enabled I very often get the wrong state of the lights. For example if I toggle them via Google Assistant.

Koenkk, 2019-11-08:
If this works you can try enabling reporting again (without availability, as that will put the most stress on the CC2531). If you really want to keep using both, you might want to take a look at the CC2652R (#1429), which will be supported by the next zigbee2mqtt release.

Bruceforce, 2019-11-08:
Okay. Nevermind. It crashed again with reporting and availability disabled. But I forget to collect debug logs. I'll enable them again and send you the logs.

Koenkk, 2019-11-08:
How many devices do you have in your network?

Bruceforce, 2019-11-08:
6 Ikea gu 10 bulbs
2 Ikea smart plugs
2 Osram smart plugs
1 Ikea E14 bulb
2 Xiaomi human body sensors 
1 Xiaomi Switch 
1 Ikea Dimmer 

So I think about 15 or maybe 16.

Koenkk, 2019-11-09:
That's not a very big network, I guess this should also work with the CC2531, did you already try the source routing firmware? This should also improve stability: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

Bruceforce, 2019-11-10:
Today it crashed again with all reporting disabled.

Here are the logs: 
Compose log: https://gist.githubusercontent.com/Bruceforce/42c380e9fcadd5405a011ccfec804b38/raw/b56b0995b7c32afe088a95c5ddbabe4d7d13310c/compose.log

zigbee2mqtt log in `logs` folder does not contain the extra logs from `DEBUG=zigbee-herdsman:*` that's why I provided the compose log.

I will now flash the source routing firmware and try again.

Koenkk, 2019-11-10:
You cannot control this but in practice I have never seen this go wrong. My assumption is that it prefers connections to routers instead of end devices.

Bruceforce, 2019-11-17:
I tried the source routing firmware and it's working until now (with reporting and availability disabled. Do you use these features in your z2m network or did you disable them?

If it keeps running with source routing firmware for another week I'll close here.

Koenkk, 2019-11-17:
I dont have them enabled but I dont use the CC2531 but the CC2652R (which is powerful enough to handle this)

Bruceforce, 2019-11-28:
Closing this since it's working with the source routing firmware. So for conclusion: CC2531 isn't powerful enough for all zigbee2mqtt features like reporting and availability if the network gets bigger.

closed 2019-11-28 by Bruceforce