Koenkk/zigbee2mqtt #2281: Docker compose error "no such file or directory"

877dev, 2019-11-05:
HI,

I had the sniffer working on my Pi3 using conventional install.
I am now trying to install with Docker compose.

Here is the yaml:
'''
zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    volumes:
      - ./volumes/zigbee2mqtt/data:/app/data
    devices:
    # CC251
      - /dev/ttyUSB_cc2531:/dev/ttyACM0
    # CC2530 / GBAN GB2530S
    # - /dev/ttyUSB_cc2530:/dev/ttyACM0
    restart: always
    network_mode: host
    environment:
      - TZ=Europe/Amsterdam
'''
When I try and run the stack:
'''
 docker-compose up -d
'''
It first downloaded the container but did not start it up. Every time I try this happens:

'''
Recreating zigbee2mqtt ... error

ERROR: for zigbee2mqtt  Cannot start service zigbee2mqtt: b'error gathering device information while adding custom device "/dev/ttyUSB_cc2531": no such file or directory'

ERROR: for zigbee2mqtt  Cannot start service zigbee2mqtt: b'error gathering device information while adding custom device "/dev/ttyUSB_cc2531": no such file or directory'
ERROR: Encountered errors while bringing up the project.

'''

Koenkk, 2019-11-06:
Change `- /dev/ttyUSB_cc2531:/dev/ttyACM0` to `- /dev/ttyACM0:/dev/ttyUSB_cc2531`

877dev, 2019-11-06:
Thank you I think that is working now :)

closed 2019-11-06 by 877dev