Koenkk/zigbee2mqtt #2282: Help for adding new devices?

ChristianPorsche, 2019-11-05:
Hi Koen, 

I am starting a home automation business in the Netherlands and bought a few zigbee devices which are not compatible with zigbee2mqtt. However, I really want to use your plugin to control these devices through home assistant etc. 

Can you help me to configure these devices? 

Thanks in advance. 

Christian

warllo54, 2019-11-05:
Have you looked at the following documentation? If you can post logs we might be able to help you.

https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

ChristianPorsche, 2019-11-09:
I did, but I can't figure it out with the following documentation. Has nobody made a video or anything to show how to add unsupported products?

SkyHyperV, 2019-11-13:
Just to piggyback off this as I am new to this as well. The following isn't working and I could use some assistance. I have a Iris (MagicStar 1117-S Motion Sensor). I did the following without success:

devices.js:

    {
    zigbeeModel: ['lumi.sensor_motion'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'iL07_1', // Vendor model number, look on the device for a model number
    vendor: 'Iris', // Vendor of the device (only used for documentation and startup logging)
    description: 'Iris Motion Sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'motion, temperature and humidity', // Actions this device supports (only used for documentation)
    fromZigbee: [fz.iris_occupancy], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    },

fromZigbee.js (copied template from another motion sensor as documentation recommended):
    iris_occupancy: {
        // This is for occupancy sensor that send motion start AND stop messages
        // Note: options.occupancy_timeout not available yet, to implement it will be
        // needed to update device report intervall as well, see devices.js
        cluster: 'msOccupancySensing',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options) => {
            if (msg.data.hasOwnProperty('occupancy')) {
                return {occupancy: msg.data.occupancy === 1};
            }
        },
    },

Still getting error about unsupported device. Yes, I restarted the service.

Koenkk, 2019-11-14:
You need to fill in the correct zigbee model (the one you use is still from the tutorial) see step 1 of https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

I will close this as the OP has send me the devices (physically)

closed 2019-11-14 by Koenkk