Koenkk/zigbee2mqtt #2285: Ikea tradfri led

cima37, 2019-11-06:
Hi!  Is it possible to set the tradfri led's somewhere if you switch it off in the dimmed state with a switch, the next turn on has normal brightness again?  I have zigbee2mqtt running on a raspberry pi with cc2531 usb stick!

nldroid, 2019-11-06:
You can do that from whatever home automation system you're using. I assume that you trigger the lamp via such a system.

cima37, 2019-11-06:
yes I use fhem!  but the "problem" is that I homatic 2 channel switch I have installed and then the current to the led free and when I turn off then the led keeps the status and I can change this again when the switch is turned on again!  With the hue lamps there is an adjustment in the app that allows the lamps to give their own status when they get power again!  That's what I wanted to know if this is also in here, or how this is realized!

lolorc, 2019-11-06:
I don't quite understand your setup, do you use a "normal" switch to cut the power to the bulb ?
If not, are you using a zigbee switch bound to your bulb ? in that case the switch communicates directly with the bulb, I would also like to know how to setup a default brightness level.
Or, if it's fhem telling the bulb to switch on, you could use the brightness attribute in the mqtt message.

cima37, 2019-11-06:
I have a normal switch to cut the power!

nldroid, 2019-11-06:
That's not something thats part of zigbee2mqtt. You could check if a message is sent over the mqtt channel when you put power back on. I'm not sure if that is the case but if there is a message, you could make a 'trigger' in fhem that catches it. The delay would be terrible probably.

I never cut the power off the bulbs but use a zigbee switch to turn it on and off.

cima37, 2019-11-06:
Ok thanks! You can not use the bulb without zigbee, is that correct?

nldroid, 2019-11-06:
Sure you can. But using switches that pass messages though a home automation system gives you more flexibility. One you cut the power, you cannot 'talk' to the bulb anymore.

cima37, 2019-11-06:
Ok thanks! I close this!

closed 2019-11-06 by cima37