Koenkk/zigbee2mqtt #2286: zigbee2mqtt fresh install not starting

mattsmithuk, 2019-11-06:
# Bug Report
## What happened
Original install of zigbee2mqtt stopped working - not starting with daemon script. Error message status stated that zigbee2mqtt required node.js v10, version 12 installed on system (see log at https://pastebin.com/mKyB03mT)

zigbee2mqtt reinstalled into virtual enviroment with recommended node.js version. 'npm start' fails (see log at https://pastebin.com/ALPHRvQY)

System was previously working, no change to hardware

## What did you expect to happen
zigbee2mqtt to start following npm start command

## How to reproduce it (minimal and precise)
Activate virtual environment and try to start service

## Debug Info
zigbee2mqtt version: 1.6.0
CC253X firmware version: 20190608

Koenkk, 2019-11-07:
The log you provided doesnt show the actual problem, please provide the direct output when starting with npm start.

mattsmithuk, 2019-11-08:
I've worked out my problem. Had to add my user to the 'dialout' group.

Thanks for the reply

closed 2019-11-08 by mattsmithuk