Koenkk/zigbee2mqtt #2288: Pairing does not work any more. Troubleshoot / reset / unpair instructions?

amund7, 2019-11-07:
I am using a CC2531, version '20190315'
zigbee2mqtt for hassio
home assistant 0.101.2
About 10 xiaomi aquara temperature / hydro sensors
At some point my stick stopped pairing new devices, also stopped re-adding the xiaomi aquara sensors when they fall off. They were always possible to re-pair, but not anymore. 

Some rumours tell me I may have paired too many items, and this would happen then.
The old ones that are still paired still work.
There is written about limits between 21 and 40 devices. I never had this many, but I may have been up to 12 or 15.
I have tried 
- press reset
- unplug
- reboot
- unplug, replug, then reboot
- unplug, reboot, then replug, etc other combinations of this
- send MQTT topic zigbee2mqtt/bridge/config/remove
- delete DB
- delete json
- delete devices.yaml
- uninstall/reinstall zigbee2mqtt
- different MQTT brokers
- manually delete entities from Home Assistant
- manually delete MQTT retained topics
- 2 different CC2531 sticks - amazingly, they both work with the same xiaomi sensors without re-pairing. so the pairing is not stored in the stick either
- buying a router stick
- using zigbee2mqtt assistant. That one actually manages to delete devices! But I can't re-pair them afterwards.
- switching back and forth between DEV branches of both zigbee2mqtt and zigbee2mqtt assistant
- bought new sensors

I have searched
- google
- github (zigbee2mqtt, zigbee2mqtt
- home assistant forums

I have asked
- discord (response: press reset button)
- home assistant forums (response: none)

The only thing I haven't tried, that I hear rumours about sometimes, is to re-flash the stick. I don't have a debugger (but I have ordered one now, as I am getting impatient)

I even bought 5 new xiaomi sensors, in case something was wrong with them. None of them can pair. There is nothing in the logs when I try adding, not even with debug on. (Yes I have enabled adding)

So in summary,
- how to unpair and delete devices
- how to troubleshoot inability to pair
- where is the pairing actually stored
- can you just swap USB sticks, or is the pairing stored in the stick?

alwashe, 2019-11-07:
How to remove
https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html
```zigbee2mqtt/bridge/config/remove

Allows you to remove devices from the network. Payload should be the friendly_name, e.g. 0x00158d0001b79111. On successful remove a device_removed message is send.
```

stored in configuration.yaml and database.db
you can manualy delete devices there also

just try to delete and repair and see the logs

i think you can swap, but i am not shure

Koenkk, 2019-11-07:
They can probably pair again once the stick is reflashed.

amund7, 2019-11-07:
Thanks. Yes, the mqtt remove command works, I suppose that is also what zigbee2mqtt Assistant uses, which also works.

However I still can't re-pair them after deleting, then they are gone forever.

So really the only way to reset the stick is to re-flash the firmware?

Koenkk, 2019-11-07:
If you are using zigbee2mqtt 1.6, they are probably force removed, in the next release this will be an explicit action: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/information/mqtt_topics_and_message_structure.md#zigbee2mqttbridgeconfigremove

But yes, basically the only way to reset is to reflash for now.

alwashe, 2019-11-07:
1. When keeping configuration.yaml and database.db and I swap the sticks do I need to rejoin all devices?
2. When keeping configuration.yaml and database.db and I reflash the stick do I need to rejoin all devices?
```
{"version":"1.6.0","commit":"7669682","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}},"log_level":"debug","permit_join":true}
```

Koenkk, 2019-11-07:
No and no (https://www.zigbee2mqtt.io/information/what_does_and_doesnt_require_repairing.html)

closed 2020-01-13 by stale[bot]