Koenkk/zigbee2mqtt #2289: Trouble adding Philips Hue 9290011370B

alwashe, 2019-11-07:
I have already 20 Bulb with this modelnumber working.
```
{"version":"1.6.0","commit":"7669682","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}},"log_level":"debug","permit_join":true}
```
```
Nov 07 18:13:07 zigbee npm[27360]: zigbee2mqtt:info  2019-11-07T17:13:07: Device '0x0017880104cbee6a' joined
Nov 07 18:13:07 zigbee npm[27360]: zigbee2mqtt:info  2019-11-07T17:13:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x0017880104cbee6a"}}'
Nov 07 18:13:07 zigbee npm[27360]: zigbee2mqtt:info  2019-11-07T17:13:07: Starting interview of '0x0017880104cbee6a'
Nov 07 18:13:07 zigbee npm[27360]: zigbee2mqtt:info  2019-11-07T17:13:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x0017880104cbee6a"}}'
Nov 07 18:13:07 zigbee npm[27360]: zigbee2mqtt:debug 2019-11-07T17:13:07: Device '0x0017880104cbee6a' announced itself
Nov 07 18:13:09 zigbee npm[27360]: zigbee2mqtt:debug 2019-11-07T17:13:09: Received Zigbee message from '0x0017880104cbee6a', type 'readResponse', cluster 'genBasic', data '{"modelId":"LWB010","manufacturerName":"Philips","powerSource":1}' from endpoint 11 with groupID 0
Nov 07 18:13:09 zigbee npm[27360]: zigbee2mqtt:debug 2019-11-07T17:13:09: No converter available for '8718696449691' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"LWB010","manufacturerName":"Philips","powerSource":1}'
Nov 07 18:13:09 zigbee npm[27360]: zigbee2mqtt:debug 2019-11-07T17:13:09: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Nov 07 18:13:09 zigbee npm[27360]: zigbee2mqtt:info  2019-11-07T17:13:09: MQTT publish: topic 'homeassistant/light/0x0017880104cbee6a/light/config', payload '{"brightness":true,"schema":"json","command_topic":"zigbee2mqtt/0x0017880104cbee6a/set","state_topic":"zigbee2mqtt/0x0017880104cbee6a","name":"0x0017880104cbee6a_light","unique_id":"0x0017880104cbee6a_light_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x0017880104cbee6a"],"name":"0x0017880104cbee6a","sw_version":"Zigbee2mqtt 1.6.0","model":"Hue White Single bulb B22 (8718696449691)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt/bridge/state"}'
Nov 07 18:13:09 zigbee npm[27360]: zigbee2mqtt:info  2019-11-07T17:13:09: MQTT publish: topic 'homeassistant/sensor/0x0017880104cbee6a/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x0017880104cbee6a","json_attributes_topic":"zigbee2mqtt/0x0017880104cbee6a","name":"0x0017880104cbee6a_linkquality","unique_id":"0x0017880104cbee6a_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x0017880104cbee6a"],"name":"0x0017880104cbee6a","sw_version":"Zigbee2mqtt 1.6.0","model":"Hue White Single bulb B22 (8718696449691)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt/bridge/state"}'
Nov 07 18:13:09 zigbee npm[27360]: zigbee2mqtt:debug 2019-11-07T17:13:09: Received Zigbee message from '0x0017880104cbee6a', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1,"appVersion":2,"stackVersion":1}' from endpoint 11 with groupID 0
Nov 07 18:13:09 zigbee npm[27360]: zigbee2mqtt:debug 2019-11-07T17:13:09: No converter available for '8718696449691' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1,"appVersion":2,"stackVersion":1}'
Nov 07 18:13:09 zigbee npm[27360]: zigbee2mqtt:debug 2019-11-07T17:13:09: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

way2many, 2019-11-11:
Someone correct me if I'm wrong but I already saw many discussions about Osram plugs disrupting the network.

alwashe, 2019-11-14:
I made a fresh install and kopied configuration.yaml and database.db
then I rejoind all Philips Bulbs, I had to reset some Bulbs 5 times, minimum was 3 times.

But now it works.

It can't be a zigbee2mqtt problem, I think it is about frequency interferences or maybe other  disturbing devices (like the osram Plugs)

closed 2019-11-14 by alwashe