Koenkk/zigbee2mqtt #2290: zigbee-shepherd died, also on fresh installs: clueless

jelmerkk, 2019-11-07:
# Bug Report
## What happened
Zigbee-Shepherd stopped working, also on fresh install (both direct install and as docker container) can't get it to work anymore. Log generally looks like this: https://pastebin.com/mXJU87uP

## What did you expect to happen
zigbee2mqtt to start I suppose

## How to reproduce it (minimal and precise)
- Fresh install directly on Raspbian
- Fresh docker container

## Debug Info
zigbee2mqtt version: 1.6.0
CC253X firmware version: CC2531 stick was provided, don't know how to check the version

jelmerkk, 2019-11-07:
npm start debug log: https://pastebin.com/piD2scjK

jelmerkk, 2019-11-07:
hmm I just used my stick on another device ant get the same `Error: SRSP - SYS - osalNvRead after 6000ms\` error. Afraid that the stick itself is toast.

Koenkk, 2019-11-07:
Reflashing should fix the issue, I would recommend with the source routing firmware (more stable)

jelmerkk, 2019-11-07:
Will do. Since it's the first time I'm going to flash it I haven't looked at the different firmwares. You mean something else than in this one?

https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20190608.zip

jelmerkk, 2019-11-07:
Thanks Koen, will try this.

closed 2020-01-14 by stale[bot]