Koenkk/zigbee2mqtt #2292: Eurotronic Spirit SPZB0001 : only 4 devices

a20gard, 2019-11-07:
Hello

First of all, I would like to thank you because I'm using this plugin and it works perfectly.

I'm writing here, not because of a "big issue" but more for a simple question

I'm using 4 Eurotronic Spirit SPZB0001 thermostats that paired well into Domoticz but that are only showing only 4 devices :
- Link Quality
- Battery Voltage
- Temperature
- Setpoint

Is it normal ? or should i find 11 devices as I could read in some forums ?

For information, I'm using : 
'version': '1.6.0', 'coordinator': 20190425, 'commit': 'e26ad2a'

Should I update ? the coordinator ?

carldebilly, 2019-11-08:
Converter : https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L4161
Definition for HASS: https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/homeassistant.js#L642

Maybe there's a Home Assistant definition missing...

Koenkk, 2019-11-08:
Are you using https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin? If so, please create an issue there so that more devices can be exposed. However for Home Assistant this is indeed what is currently exposed.

a20gard, 2019-11-09:
> Are you using https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin? If so, please create an issue there so that more devices can be exposed. However for Home Assistant this is indeed what is currently exposed.

No, I'm using koenkk / zigbee2mqtt :-) with Domoticz into a Synology server

Koenkk, 2019-11-10:
How are the devices discovered in Domitcz? Via the Home Assistant discovery protocol?

a20gard, 2019-11-10:
via npm start with pairing authorization, following the instructions of the site : 
https://www.zigbee2mqtt.io/getting_started/pairing_devices.html

Koenkk, 2019-11-11:
It's still unclear to me how the 4 devices, which you mentioned in the OP, are discovered into domoticz. However the 4 sensors you describe are the expected ones when integrating via Home Assistant, so nothing strange there.

a20gard, 2019-11-11:
Hello Koenkk
In the Hardware location, i added the type "Zigbee2MQTT" relative to zigbee2mqtt project (pointing to your github), filling mosquitto server address and port. I didn't use Home Assistant (or otherwise I don't know where) >> how can I be sure ?

a20gard, 2019-11-12:
well, after having a look to your links, I admit that I ended by using stas-demydiuk plugin : I didn't know that this plugin was created by a different person, I thought it was your plugin. Sorry for the missunderstanding.

Koenkk, 2019-11-12:
Yes, so if you want to have more entities exposed it should be added there. Create an issue here: https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues

closed 2019-11-12 by Koenkk