Koenkk/zigbee2mqtt #2294: Startup problem with latest dev

Kryzek, 2019-11-08:
I just update to latest dev version and I encountered this error.
```
zigbee2mqtt:debug 2019-11-08T16:15:50: Using zigbee-herdsman with settings: '{"network":{"panID":6755,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[12],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM1"}}'
winston:file written true false
  serialport/bindings loading LinuxBinding +0ms
  serialport/stream .list +0ms
winston:file logged 80 info  2019-11-08T16:15:50: Starting zigbee2mqtt version 1.6.0 (commit #6c45373)

winston:file logged 135 info  2019-11-08T16:15:50: Starting zigbee-herdsman...

winston:file logged 564 debug 2019-11-08T16:15:50: Using zigbee-herdsman with settings: '{"network":{"panID":6755,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[12],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM1"}}'

  zigbee-herdsman:adapter Path '/dev/ttyACM1' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[7,0,5,1,1,5,0,7,7,1,5,5,1,7,6,9],"panID":6755,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[12]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM1"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening with /dev/ttyACM1 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  serialport/stream opening path: /dev/ttyACM1 +99ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +1ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyACM1 +0ms
  serialport/stream _read reading { start: 0, toRead: 65536 } +0ms
  serialport/binding-abstract read +2ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +3ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  serialport/stream _write 1 bytes of data +2ms
  serialport/binding-abstract write 1 bytes +1ms
  serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +0ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixWrite write returned: wrote 1 bytes +1ms
  serialport/bindings/unixWrite Finished writing 1 bytes +0ms
  serialport/stream binding.write write finished +1ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
  serialport/bindings/poller Polling for "readable" +3ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +1s
  serialport/stream _write 5 bytes of data +1s
  serialport/binding-abstract write 5 bytes +1s
  serialport/bindings/unixWrite Starting write 5 bytes offset 0 bytesToWrite 5 +1s
  serialport/bindings/unixWrite write returned: wrote 5 bytes +1ms
  serialport/bindings/unixWrite Finished writing 5 bytes +0ms
  serialport/stream binding.write write finished +2ms
zigbee2mqtt:error 2019-11-08T16:15:57: Error while starting zigbee-herdsman
winston:file written true false
zigbee2mqtt:error 2019-11-08T16:15:57: Failed to start zigbee
winston:file written true false
zigbee2mqtt:error 2019-11-08T16:15:57: Exiting...
winston:file written true false
zigbee2mqtt:error 2019-11-08T16:15:57: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
winston:file written true false
```

zigbee2mqtt version 1.6.0 (commit #6c45373)

Coordinator firmware version: zStack3x0 20191015 with CC2652R

Koenkk, 2019-11-08:
Are you sure that the path to the CC2652R is correct? Note that it exposes two ports: the JTAG flasher and the actual CC2652R. Try `ttyACM0`

Kryzek, 2019-11-08:
I did not change configuration at all while updating. I had it as ttyACM1 originally. I did, however, flashed newer firmware (20191106) and I got it running with my old version (#7669682) after I changed ttyACM1 to ttyACM0 in configuration.
```
info  2019-11-08T16:55:23: Starting zigbee2mqtt version 1.6.0 (commit #7669682)
info  2019-11-08T16:55:23: Starting zigbee-herdsman...
debug 2019-11-08T16:55:23: Using zigbee-herdsman with settings: '{"network":{"panID":6755,"extenedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt-backup/data/database.db","backupPath":"/opt/zigbee2mqtt-backup/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
info  2019-11-08T16:55:33: zigbee-herdsman started
info  2019-11-08T16:55:33: Coordinator firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'
```

And after this it did also start with new dev version (#6c45373) with ttyACM0 in configuration file.
```
info  2019-11-08T17:04:38: Starting zigbee2mqtt version 1.6.0 (commit #6c45373)
info  2019-11-08T17:04:38: Starting zigbee-herdsman...
debug 2019-11-08T17:04:38: Using zigbee-herdsman with settings: '{"network":{"panID":6755,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
info  2019-11-08T17:04:40: zigbee-herdsman started
info  2019-11-08T17:04:40: Coordinator firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'
debug 2019-11-08T17:04:40: Zigbee network parameters: {"panID":6755,"extendedPanID":"0xdddddddddddddddd","channel":11}
info  2019-11-08T17:04:40: Currently 18 devices are joined:
```

Koenkk, 2019-11-08:
Great! A solution for this in the future might be to use the autodetect port option (currently undocumented but can be achieved by setting `port: null`.

closed 2019-11-08 by Koenkk