Koenkk/zigbee2mqtt #2296: deviceAvailability publishes message even if there is no change in availability

sjorge, 2019-11-08:
# Bug Report
## What happened
I configured availability_timeout of 300 (5 minutes) as I am really only interested in the polling of the color after some of my hue bulbs lose power.

* startup: each device publishes the available = online
* active devices that keep publishing message: no update of available state because timer keeps getting reset (great)
* devices that don't publish often: publishes available = online after interval
* device goes offline: publishes available = offline after interval
* device comes online: publishes available = online + query all properties (yay)

## What did you expect to happen
I would expect NOT to get an available = online repeat message in the if one was already sent.

Currently devices like the ikea repeater publish the available message every 5 minutes due to the timer, other bulbs that publish reports never post available = online after the one at startup.

## How to reproduce it (minimal and precise)
Set a long timeout and you'll notice if you have devices that are very active vs one that are not.

## Debug Info
zigbee2mqtt version: `1.6.0 (commit #00bc4de)`
CC253X firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}}`

sjorge, 2019-11-08:
This doesn't seem to hard to fix, I will open a PR for this next week when i have a bit more time. (Probably on monday)

The current idea is to a hash with the name -> availability mapping and then check [here](https://github.com/Koenkk/zigbee2mqtt/blob/74cdde23b912578b1f50a6762e268b81e402277e/lib/extension/deviceAvailability.js#L102)  if the current state differs from the stored one. If so publish and update the hash.

That way we should only get available mqtt message when the state has changed.

sjorge, 2019-11-09:
Better yet, it seems this info is already available in `this.state`. Will do some more digging around the extension to understand it better to implement it as cleanly as I can.

sjorge, 2019-11-09:
PR opened, the change was minimal

Koenkk, 2019-11-10:
Thanks! merged, therefore closing this.

PR #2303 (2019-11-10): Only post deviceAvailability updates if the status has changed.
This fixes #2296 so that the /availability MQTT message is only sent when the state has  change...
- offline -> online
- online -> offline
closed 2019-11-10 by Koenkk