Koenkk/zigbee2mqtt #2297: RTCGQ11LM no_occupancy : 0 sends every hour update with true occupancy

flok, 2019-11-08:
# Bug Report
## What happened
The RTCGQ11LM has an option for an occupancy_timeout, which when its set to 0 the device will never send a false occupancy anymore and only if motion is detected. The RTCGQ11LM also sends a update every 55 Minutes with data, the occupancy variable in there will always be true if the occupancy_timeout is set to 0
## What did you expect to happen
55 Minute update should still send occupancy : false even when occupancy_timeout is set to 0
## How to reproduce it (minimal and precise)
set occupancy_timeout of the device inside configuration.yaml to 0 and look at the update every 5 minutes.
## Debug Info
zigbee2mqtt version: 1.6.0

andreasbrett, 2019-11-11:
Thanks so much for your observation! I noticed this too but couldn't pin it down to the real cause. I modified my RTCGQ11LM to send updates every 5sec (hardware modification) and right after that modified the occupation_timeout to 0. I thought the modification caused my lights to go on "randomly".

Koenkk, 2019-11-11:
This is indeed a bug, but I don't know what is the best way to solve it, the following happens:
- Sensor detects occupancy -> publishes occupancy: true
- As `occupancy_timeout` is `0` no occupancy: false is published
- Every hour the sensor send a battery status update, at this moment the last known occupancy state is true, results in publish of battery + occupancy: true

Possible solutions
- After occupancy: true, immediately publish a `{"occupancy": ""}` (easy solution)
- Only include the occupancy property when occupancy changed (technically a hard solution)
- Allowing a minimal value of 1 -> occupancy: false is then reported after 1 second (easy solution)

flok, 2019-11-12:
The Second one would obviously be the best one but the third also sounds good. If i understand it correctly the sensor only publishes true occupancy and the zigbee2mqtt handles the false after X minutes ?

Koenkk, 2019-11-12:
@flok yes that is correct, I would also opt for solution 3 as otherwise we will have messages without the occupancy property (this will probably throw some errors on the home assistant side again).

closed 2020-01-19 by stale[bot]