Koenkk/zigbee2mqtt #2298: Crashing after pairing Drayton Wiser radiator thermostat

arcanefoam, 2019-11-08:
# Bug Report
## What happened
After adding a new entry for the Drayton valve the application crashes  after restart

## What did you expect to happen
Receive some data from the valve

## How to reproduce it (minimal and precise)
Add device to devices.json
// Drayton Wiser
    {
        zigbeeModel: ['iTRV'],
        model: 'WV704R0A0902',
        vendor: 'Drayton Wiser',
        description: 'Wiser Radiator Thermostat',
        supports: 'temperature', // Actions this device supports (only used for documentation)
        fromZigbee: [], // We will add this later
        toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    },

then restart zigbee2mqtt

## Debug Info
zigbee2mqtt version: 1.6.0
CC253X firmware version: Latest

## Log
```
...
  zigbee2mqtt:debug 11/8/2019, 11:27:36 PM zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":"11","panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"ieeeAddr":"0x00124b000be8a0ed","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190608},"startTime":1573255656,"joinTimeLeft":0}
  zigbee-shepherd Backup is only supported for Z-Stack 3, skipping... +0ms
  zigbee2mqtt:info 11/8/2019, 11:27:36 PM Currently 1 devices are joined:
  zigbee2mqtt:info 11/8/2019, 11:27:36 PM 0x086bd7fffe7ed5e4 (0x086bd7fffe7ed5e4): WV704R0A0902 - Drayton Wiser Wiser Radiator Thermostat (EndDevice)
...
  zigbee-shepherd Register zApp, epId: 11, profId: 260  +116ms
  zigbee2mqtt:debug 11/8/2019, 11:27:36 PM Mounted the cieApp (epId 11)
  zigbee2mqtt:info 11/8/2019, 11:27:46 PM Zigbee publish to device '0x086bd7fffe7ed5e4', genBasic - readRsp - [] - {"direction":1,"seqNum":4,"disDefaultRsp":1} - 1
/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/zcl/utils.js:49
        throw new Error(`Cluster with key '${key}' does not exist`);
        ^

Error: Cluster with key '65027' does not exist
    at Object.getCluster (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/zcl/utils.js:49:15)
    at Object.getAttributeLegacy (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/zcl/index.js:39:27)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/shepherd.js:122:36
    at Object._.forEach (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/busyman/index.js:411:27)
    at ZShepherd.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/shepherd.js:121:11)
    at ZShepherd.emit (events.js:198:13)
    at dispatchIncomingMsg (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/af.js:565:41)
    at Controller.zclIncomingMsgHandler (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/af.js:631:12)
    at Controller.emit (events.js:198:13)
    at Immediate.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/af.js:640:23)
```

arcanefoam, 2019-11-09:
From what I can tell, line 49-51 in zcl/utils.js makes the addition of new devices hard.
I would rather have the caller of getCluster support a "missingCluster Object", hence, line 50 could then be replaced by:
```
return {
        ID: name,  // key
        attributes: cluster.attributes,
        "A new cluster with the given ID must be added to .../cluster.js",
        commands: {},
        commandsResponse: {},
```

Probably we would like users to be able to provide a local cluster.js that can be merged with the dist one. That way it is easier to debug, fix incrementally and avoid having to wait for releases to get devices supported.

Koenkk, 2019-11-09:
This error should already be fixed in the latest dev branch, herdsman will just emit the cluster number.

messismore, 2019-11-15:
> This error should already be fixed in the latest dev branch
Can you confirm this @arcanefoam ? Thinking of ordering the same TRV…

arcanefoam, 2019-11-17:
Hi @messismore, I have managed to pair the TRV with the latest software and I am in the process of figuring out what configuration it needs in order to work properly. I have managed to getting it to report the temperature, but still need to figure out how to change the setting point and explore the other valve features. Feel free to email me (`<username>` at gmail) if you want to exchange information and figure this one out together.

closed 2019-11-17 by arcanefoam