Koenkk/zigbee2mqtt #2299: Can pair bulbs, supported smart plugs - not so much

nanobra1n, 2019-11-09:
# Bug Report
## What happened
I am transitioning from my WInk 2 hub to zigbee2mqtt and have been incrementally pairing my devices. So far it has been mostly smooth, Aqara vibration and temperature sensors as well as Cree Connected Bulbs paired on the first few tries. An Osram LED strip gave me more trouble and I filed an [issue](https://github.com/Koenkk/zigbee2mqtt/issues/2213) where during troubleshooting it finally paired, leading me to believe that I just gave up too soon and if you just hang in there long enough, it will happen. But now I am trying to pair 2 different supported devices and after 50+ tries each we are nearing the insanity definition. 

The devices are:
- a [Securify Peanut](https://www.zigbee2mqtt.io/devices/PP-WHT-US.html) Smart plug
- a [Centralite 4256251-RZHAC](https://www.zigbee2mqtt.io/devices/4256251-RZHAC.html) smart plug 

I tried pairing centimeters next to the stick or the other side of the room - same result  

Peanut 
Log: https://pastebin.com/2bSLWDQS
Sniffed traffic: https://www.dropbox.com/s/4bm017xy4t0pqao/peanutpair.pcapng?dl=0

4256251-RZHAC
Log: https://pastebin.com/RnNcNrBy
Sniffed traffic: https://www.dropbox.com/s/mhmxso1hoef4ny7/4256251-RZHACpair.pcapng?dl=0
 
## What did you expect to happen
either one of them pairs  
## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: z2m-edge 1.6.0 36c7a27e6c632e0450db18279fc46a1d01365579
CC253X firmware version: 20190608

Koenkk, 2019-11-10:
- Looking at your log, it looks that there are 2 separate zigbee network operating on the same network? Is the wink hub still powered on?
- I see that the device joins successfully, but leaves immediately. In the sniffer logs I see that a device is broadcasting leave requests in the network (asking devices to leave the network). The network address of this device is `42924`, can you find it in your `database.db`?

nanobra1n, 2019-11-10:
- yes, the WInk hub was still on. I tried pairing with it off, but the result was the same. 
- no 42924 in my `database.db` the closest I have is several `"profId":49246`.

[database.db](https://pastebin.com/k7pgzZJj)

Koenkk, 2019-11-10:
But are all the devices that are connected to the wink hub also powered off? Otherwise the zigbee network still exists.

nanobra1n, 2019-11-10:
Well, d'oh. No, they weren't

I'll try again with them powered off.

nanobra1n, 2019-11-10:
Didn't work either. Just tried the peanut. Sniffed traffic is here: https://www.dropbox.com/s/aaw54f39sli9eof/peanutpair_noWink.pcapng?dl=0

Koenkk, 2019-11-10:
I don't know why, but immediately after joining the devices leaves the network (while no-one requested this). You can see it on ID 214, 215 and 216. Are you sure you are using the correct pairing procedure and don't reset the device twice? A second reset could lead to a device leaving the network again.

nanobra1n, 2019-11-10:
The reset procedure for the peanut is to press the pair button 10 s, then release it (at whhich point I get "device has left the network" message in the log , disconnect from power and then connect power again (I use a power strip with an on/off button). I only do another reset once I got the error in the log 

```
2019-11-10T19:19:22.832Z zigbee-herdsman:controller:error Interview failed for '0x000d6f000a79260e with error 'Error: AREQ - ZDO - nodeDescRsp after 10000ms'
zigbee2mqtt:error 2019-11-10T19:19:22: Failed to interview '0x000d6f000a79260e', device has not successfully been paired
```
and nothing more happens in the log for a bit (sometimes the interview process gets retried)

Koenkk, 2019-11-11:
I honestly have no clue why they immediately leave (resulting in a failing interview). @thardie @ChrisSantiago82 @roger-  does this device pair correctly in zigbee2mqtt 1.7.0?

ChrisSantiago82, 2019-11-11:
Hello,
I'm using 7 Peanut Smart Plugs and they work all without any problems. The only thing is when pairing them you have to add the modelId manually.

nanobra1n, 2019-11-11:
So here is what I did (no idea if that would have an impact on my problem, but getting desperate):

- Wink Hub and all its paired devices powered off - no change
- reflashed the stick  - no change
- switched to 1.7 - no change
- removed all devices, powered all Zigbee devices off except the plug - no change
- ordered a new plug in case the one I have is just defective (will be here tomorrow)

I re-added all my other devices and none gave me any major problems. 🤷‍♂

ChrisSantiago82, 2019-11-11:
The question is if you have followed the pairing step in the documentation?
The problem is the Peanut Smart Plugs is not sending the id when it's doing the pairing and because of that the stick won't know what device it is. After pairing you must enter the id by hand.
https://www.zigbee2mqtt.io/devices/PP-WHT-US.html

nanobra1n, 2019-11-11:
The problem is that I don't even get it to pair. So nothing in `database.db` to edit.

ChrisSantiago82, 2019-11-11:
I see. But you are getting the lights on the plug?

nanobra1n, 2019-11-11:
Yes, pressing the pair button 10 s gives me solid red, after removing it from power and back in blinking red lights for a few second (until interview starts?), but the interview fails every time and pairing is unsuccessful

ChrisSantiago82, 2019-11-11:
Well, in that case I can't help you :(
I'm sorry. I have the same version and the same plug and here it works fine.
And other devices are working? Just not the  Peanut Smart Plugs?

nanobra1n, 2019-11-11:
Smart plugs don't like me

Not working:
- 1 Securify Peanut Smart plug
- 1 Centralite 4256251-RZHAC smart plug

Working:

- 5 CREE Connected bulb (B00TN589ZG) 
- 1 OSRAM Flex RGBW (4052899926110)
- 1 Xiaomi Aqara vibration sensor (DJT11LM)
- 1 Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM) 
- 1 CC2531 router

nanobra1n, 2019-11-11:
More tries:

- tuned my wifi networks off (I, umm, might have running Tx power higher than the default setting) to rule out interference - no change
- turned off ethernet over power line in case of interference - no change 
- I got my CC1352P-2 in today, tried to pair just it with it - no luck either

So I gave up for now (going back to the CC2531) and tried to pair an [OSRAM AA70155](https://www.zigbee2mqtt.io/devices/AA70155.html) and I can't get to pair that one either.
The weird thing is that bulb blinks when the interview process starts, and then again 4 times a bit later. On my Wink hub that means the bulb paired, but the log says pairing failed.

nanobra1n, 2019-11-11:
Changed channels, just tried to pair the bulb still failed, but not until after Homeassistant got the bulbs config: https://pastebin.com/DRcNfpA5

Looks like zigbee2mqtt gives up too early?

Koenkk, 2019-11-12:
@nanobra1n something different happens for this one, can you provide the sniffing logs and logs when running with `DEBUG=zigbee-herdsman:controller*`?

nanobra1n, 2019-11-13:
So the second Peanut came in and also no success pairing it :(
Log: https://pastebin.com/bKfWJbeg
Sniff: https://www.dropbox.com/s/todl3csyan5o54o/2ndpeanutpair.pcapng?dl=0

Just plugged it in out of the box, went through several interview rounds without me needing to reset it.

Tried the Osram bulb as well again - also no success
Log: https://pastebin.com/XCU9UTSk
Sniff: https://www.dropbox.com/s/y91zgva9vr3iaf8/Osrampair.pcapng?dl=0

Reset the bulb once, but wasn't very good at it, therefore the deveral "device announced itself"s

BTW, is there a better way to capture the log than copy & pasting the screen output as the debug output of `zigbee_herdsman_debug: true` doesn't make it into the log file

Koenkk, 2019-11-14:
Again I see the same weird behavior for the OSRAM bulb, it leaves before zigbee2mqtt can interview it.

As I don't believe the devices itself are the problem (you tried with many different ones now) we should start to isolate things.

Can you setup a clean zigbee2mqtt instance on a different system, with a freshly flashed stick and try to pair the OSRAM bulb to it. Before doing this make sure that all zigbee devices in your house are powered off.

nanobra1n, 2019-11-14:
Thanks for taking the time to look what appears a unique problem :)

I have a RPi laying around I can resurrect and install just zigbee2mqtt on it. I live in an apartment complex, so I can't control what my neighbors are doing (while I doubt they run HA with zigbee2mqtt. there are probably some Echo Plus devices around), but I can shut down everything I own. Come to think of it I can sniff traffic and see if something else is around.  

The OSRAM bulb is weird, when I paired it back with my wink hub 2 it connected so quick that Wink said pairing failed, but it then showed up anyway.

nanobra1n, 2019-11-15:
So partial success:

- cloned zigbee2mqtt - dev on a RPi
- reflashed the stick
- turned off my wifi
- turned off all my zigbee devices (sniffed traffic and there was none)
- started zigbee2mqtt
- tried to pair OSRAM bulb: immediate success
- deleted database
- paired again: success after the second try: https://pastebin.com/ZLRNdD4R

Emboldened, I now tried the Peanut, but there wasn't even a message that the plug was detected and and or even that an interview process was attempted:  https://pastebin.com/thJ7MvDk

Traffic was sniffed again: https://www.dropbox.com/s/sr0fh64ekyf6zr2/peanutpair_rpi.pcapng?dl=0

Koenkk, 2019-11-15:
The device doesn't even announce itself in the log, can you try the centralite plug?

nanobra1n, 2019-11-15:
So when I plug in the OSRAM first and then try to pair the plug at least the interview process gets going. It fails, but at least it now starts: https://pastebin.com/Ppi44pq6

nanobra1n, 2019-11-15:
And wouldn't you believe it, the peanut finally paired: https://pastebin.com/4qBCg53x

Started zigbee2mqtt with the unpaired OSRAM bulb on, then reset the Peanut and it paired.

Sniff while that happened: https://www.dropbox.com/s/byfqlsut1dxn8nd/Peanutsuccess.pcapng?dl=0

Now back to my HA system and see if the OSRAM magic works with the stable add-on.

nanobra1n, 2019-11-16:
Yeah, that was too ambitious. The lack of feedback in stable makes it an effort in futility and even in -edge I am unable to reproduce my earlier pairing success.

nanobra1n, 2019-11-16:
So since it appears that the peanut plug would rather use the bulb as a relay then connect with the stick directly I tried and flashed the router fw onto a CC2351, paired it and plugged it in right next to the smart plug. Tried it with the Centralite first and it paired on the first try. As did both Peanuts I had . I don't know enough why that one worked ¯\_(ツ)_/¯

Koenkk, 2019-11-16:
Thats great, so all devices are paired on your production system now?

nanobra1n, 2019-11-16:
Yes, they are

closed 2019-11-16 by nanobra1n