Koenkk/zigbee2mqtt #2300: Ikea ICTC-G-1 paired but no messages in lates dev with Z-Stack 3 on CC2652R

nldroid, 2019-11-09:
I succesfully paired the Ikea ICTC-G-1 (https://www.zigbee2mqtt.io/devices/ICTC-G-1.html)

But when i turn the dial, nothing happens in zigbee2mqtt. Nothing is shown in the log.

I've tried new batteries, re-pairing, restarting etc. but it won't send any messages. 

Anything changed in herdsman for this device?

Koenkk, 2019-11-10:
Can you share your `database.db`?

nldroid, 2019-11-10:
Here is the dimmer part:

`{"id":10,"type":"EndDevice","ieeeAddr":"0x000b57fffe902bb0","nwkAddr":46964,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI wireless dimmer","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":2064,"inClusterList":[0,1,3,9,2821,4096],"outClusterList":[3,4,6,8,25,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI wireless dimmer","manufacturerName":"IKEA of Sweden","powerSource":3,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":30,"dateCode":"20170612","swBuildId":"1.2.248"}}},"binds":[{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b001caab8c3","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001caab8c3","endpointID":1}]}},"appVersion":17,"stackVersion":87,"hwVersion":30,"dateCode":"20170612","swBuildId":"1.2.248","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1}}`

If you need the rest, let me know,.

I planned to sniff the traffic but i have to find tim for it..

Koenkk, 2019-11-10:
Are you also using the latest cc2652r firmware? Out of my head: 20191015 or 20191106 (latest) is required for this device

nldroid, 2019-11-10:
Yes. Had 20191015 but flashed 20191106 yesterday when it didn't work,

nldroid, 2019-11-10:
Don't spend too much time on it. I'll find some time this week to flash a sniffer firmware and see what it sends.

nldroid, 2019-11-11:
Not sure whats going on with this dimmer. I got a lot of 'Orphan' messages in wireshark as soon as i put the battery in.

I've replaced he battery (again), deleted the database entry and re-paired. It seems to work for now but not not sure for how long.

I'll monitor it and see what it does the next couple of days.

closed 2019-11-11 by nldroid