Koenkk/zigbee2mqtt #2305: Cannot get the Node Descriptor of the Device: 0xc8d2c1XXXXXXXXXX (Error: Timed out after 10000 ms)

Inshakoor, 2019-11-10:
##Bug Report
When I pair a device for the first time on a fresh instance of Z2M (The only devices I have are 3 IKEA outlets and 2 Lightify RGBW BR30 bulbs), I start getting the following errors constantly:

zigbee2mqtt:error 11/9/2019, 5:48:41 PM Cannot get the Node Descriptor of the Device: 0xc8d2c10000005b30 (Error: Timed out after 10000 ms)
zigbee2mqtt:error 11/9/2019, 5:49:01 PM Cannot get the Node Descriptor of the Device: 0xc8d2c100000051ac (Error: Timed out after 10000 ms)
zigbee2mqtt:error 11/9/2019, 5:49:21 PM Cannot get the Node Descriptor of the Device: 0xc8d2c1000000494d (Error: Timed out after 10000 ms)
zigbee2mqtt:error 11/9/2019, 5:49:41 PM Cannot get the Node Descriptor of the Device: 0xc8d2c100000051a8 (Error: Timed out after 10000 ms)
zigbee2mqtt:info 11/9/2019, 5:51:17 PM Successfully reenabled joining

The first 6 characters of the device IDs all point to Jetlun devices (Range extenders?) but I don't have any of those devices (is it possible this is coming from a neighbor?).  I added the IDs to the ban and blacklist sections in the configuration.yaml to no avail.  Once i start getting these errors, it's next to impossible to add any more devices.  Some will add in a couple hours, some overnight, and some I can't get to pair at all, but certainly not within a minute or 2.

## What did you expect to happen
I expected to be able to continue pairing devices.

## How to reproduce it (minimal and precise)
1) Remove all devices
2) Change the wireless channel to any other channel (Once changed you can change it back.  The logs will be clean, no errors)
3) Add a new device
4) The new device adds immediately and then the errors posted above start showing up and pairing is incredibly temperamental.

## Debug Info
zigbee2mqtt version:  zigbee2mqtt version 1.6.0 (commit #e26ad2a)
CC253X firmware version:  Coordinator firmware version: '20190619'

##Notes
I've seen similar posts here but those seem to be with the errors coming from the device being paired.  The IDs in the errors are not my devices.

Thank you.

Koenkk, 2019-11-10:
Please update to zigbee2mqtt 1.7.0, you shouldn't be getting those errors there.

Inshakoor, 2019-11-11:
Thank you very much! Upgrading to 1.7.0 (zigbee2mqtt version 1.7.0 (commit #551a58a)) now allows me to add devices almost instantly, but now devices take between 5-15 seconds to respond (IKEA outlets around 5 seconds, Lightify bulbs around 13 seconds).  I will do more testing to see if the slowness is anything on my side.

Unfortunately, errors for the unknown devices continue.  After the upgrade, Z2M added the foreign devices to my devices list in configuration.yaml.  The logs, albeit showing different errors, still show the device errors over and over:

zigbee2mqtt:info  2019-11-11T07:08:24: Starting interview of '0xc8d2c100000051ac'
zigbee2mqtt:info  2019-11-11T07:08:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xc8d2c100000051ac"}}'
zigbee2mqtt:error 2019-11-11T07:08:53: Failed to interview '0xc8d2c10000005b30', device has not successfully been paired
zigbee2mqtt:info  2019-11-11T07:08:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0xc8d2c10000005b30"}}'
zigbee2mqtt:error 2019-11-11T07:08:53: Failed to interview '0xc8d2c1000000494d', device has not successfully been paired
zigbee2mqtt:info  2019-11-11T07:08:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0xc8d2c1000000494d"}}'
zigbee2mqtt:info  2019-11-11T07:08:55: Starting interview of '0xc8d2c1000000494d'
zigbee2mqtt:info  2019-11-11T07:08:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xc8d2c1000000494d"}}'
zigbee2mqtt:info  2019-11-11T07:08:58: Starting interview of '0xc8d2c10000005b30'
zigbee2mqtt:info  2019-11-11T07:08:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xc8d2c10000005b30"}}'
zigbee2mqtt:error 2019-11-11T07:09:03: Failed to interview '0xc8d2c100000051a8', device has not successfully been paired
zigbee2mqtt:info  2019-11-11T07:09:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0xc8d2c100000051a8"}}'

I then added the devices to the ban list in configuration.yaml, restarted Z2M, and I get the following errors over and over:

zigbee2mqtt:info  2019-11-11T07:06:08: Rejecting joining banned device '0xc8d2c100000051ac'
zigbee2mqtt:info  2019-11-11T07:06:10: Rejecting joining banned device '0xc8d2c1000000494d'
zigbee2mqtt:info  2019-11-11T07:06:10: Rejecting joining banned device '0xc8d2c1000000494d'
zigbee2mqtt:info  2019-11-11T07:06:11: Rejecting joining banned device '0xc8d2c100000051a8'
zigbee2mqtt:info  2019-11-11T07:06:11: Rejecting joining banned device '0xc8d2c10000005b30'
zigbee2mqtt:info  2019-11-11T07:06:13: Rejecting joining banned device '0xc8d2c100000051ac'
(node:16) UnhandledPromiseRejectionWarning: Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
(node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 5709)

Any further suggestions on how to eliminate these errors or are they no big deal besides filling up the logs?  Any ideas as to the slow response times (while I still dig into this at my end)?  I again thank you for your great support.  The community is very grateful :).

Koenkk, 2019-11-11:
I guess this is also the reason for your slowness, some devices seem to join again and again. Is `permit_join: false` in `configuration.yaml`?

Inshakoor, 2019-11-12:
That was it!  By leaving permit_join: true in my configuration.yaml, it was bogging things down with those devices trying over and over to connect.  Thank you very much.  Things appear to be working great and adding devices takes 10 seconds.  Again, Thanks and Happy Holidays!

closed 2019-11-12 by Inshakoor