Koenkk/zigbee2mqtt #2307: zigbee2mqtt 1.7 problem with network map

jarzebski, 2019-11-10:
hi, after update to 1.7 I have issue with bad network map (zigbee2mqtt/bridge/networkmap/raw). some nodes missing link info, but they are physically connected correctly. After downgrade to 1.6 all is ok

FedeBZ, 2019-11-10:
hello, me too

Koenkk, 2019-11-10:
Should be fixed in latest dev (should give same behaviour as 1.6.0). (reference #1961 )

jasimancas, 2019-11-10:
Update to dev versión to fix the bug?

Koenkk, 2019-11-11:
yes

jasimancas, 2019-11-11:
[image]
Thanks!

Koenkk, 2019-11-11:
Looks great (meshing). Closing this as the issue has been fixed.

closed 2019-11-11 by Koenkk

lolorc, 2019-11-11:
[image]

Kryzek, 2019-11-11:
My network maps has never looked this good. Could be a bit sexier but some routers are not connected to mains currently. But mesh looks very good! Thanks you! 

[image]