Koenkk/zigbee2mqtt #2308: Error: SRSP - AF - dataRequest after 6000ms

spockster1985, 2019-11-10:
# Bug Report
## What happened
I started using Zigbee2Mqtt 3 days ago with 3 Ikea bulbs. When I modify the lights, they frequently gives this error and the light does not respond to the command.

## What did you expect to happen
Light to respond to the command and tunr on/off, change color, ...

## How to reproduce it (minimal and precise)
Toggle lights on/off or change color or change brigthness.

## Debug Info
zigbee2mqtt version: latest dev version (1.7.0+dev (commit #b617239))
CC253X firmware version: CC2652R with zStack3x0 20191106

error 2019-11-09T21:53:41: Publish 'set' 'state' to 'Matthijs_Wolk' failed: 'Error: SRSP - AF - dataRequest after 6000ms'
info  2019-11-09T21:53:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Matthijs_Wolk' failed: 'Error: SRSP - AF - dataRequest after 6000ms'","meta":{"friendly_name":"Matthijs_Wolk"}}'

I added log files, where you can see that it frequently failes to execute the command.
[log1.txt](https://github.com/Koenkk/zigbee2mqtt/files/3828678/log1.txt)
[log2.txt](https://github.com/Koenkk/zigbee2mqtt/files/3828679/log2.txt)

Koenkk, 2019-11-10:
Can you provide the log when running zigbee2mqtt with `DEBUG=zigbee-herdsman:* npm start`?

spockster1985, 2019-11-10:
Here is the new log file.
[Log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3828898/Log.txt)

Koenkk, 2019-11-11:
Does it only happen with the IKEA color bulb or also others (e.g. Buitenlamp)?

spockster1985, 2019-11-11:
It also happens to the other light.

Koenkk, 2019-11-12:
Very strange, haven't seen this before.

Can you try removing `coordinator_backup.json` and reflashing the CC2652R using https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html, after this you have to re-pair your devices.

spockster1985, 2019-11-13:
Reflashed. Still getting the same error.

Could it be the 'complex' setup? I'm using docker on a XPenology VM on ESX 6.5.

Koenkk, 2019-11-14:
I guess that could be.

Either:
- the cc2652r doesnt send a respond (unlikely)
- something bad happens with the serial communication (guess more likely given your setup)

Can you try on e.g. a linux laptop or pi?

spockster1985, 2019-11-14:
I switched to a Ubuntu VM. I fired a lot of commands towards the lights. So far I haven't got the error.

For me it's solved! Big thanks for the help.

closed 2019-11-14 by spockster1985

patriot1889, 2019-11-20:
I would just like to add that I just got this error. I tried to add a Wemo bulb and got this error. I was unable to do anything at all, nothing would respond. Multiple restarts and removing the wemo from the config did nothing. 

I finally got it all to work after reflashing the stick. It seems like the bulb completely corrupted the network. I'm leaving it out of the network for now.

oischinger, 2019-11-21:
I have the same issue with some innr bulbs
For me zigbee2mqtt stops doing anything as soon as they are added to the network. But it doesn't always seem to be the case. Seems to be a bit intermittent.

thardie, 2019-11-30:
I'm getting this randomly too. The even worse problem is restarting zigbee2mqtt doesn't clear it. I have to unplug the USB stick and plug is back in to clear it. Even pressing the reset button on it doesn't clear it.

thardie, 2019-12-16:
Just got this again. Mine's not VM related, since I don't have any VM anywhere in my loop. The system will also work for several days, and then go into this state. I removed the 1 innr appliance module I have, and it still happens.

I'm going to have to dump this whole platform if there's not a fix for this soon :((

Koenkk, 2019-12-16:
@thardie your stick crashes because it probably cannot handle your network. I would recommend switching to the source routing firmware if you are not already (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing). If that still doesn't solve the problem the only thing left to do is upgrading to a more powerful coordinator https://www.zigbee2mqtt.io/information/supported_adapters.html

cvhoang, 2020-08-03:
I'm experiencing the same problem on Texas Instruments LAUNCHXL-CC26X2R1 and 88 devices. The adapter would randomly crash every week or so. You have to unplug and restart Z2M.

My devices are mostly Hue lights, sensors, Samsung Smart Plugs and Xiaomi sensors. I previously ran Conbee II stick. It was really stable but I migrated because I think Z2M is the future. Not sure what I can do here since I already have the most powerful hardware?

thardie, 2020-08-04:
I think the stick is just very sensitive to USB timings. I was originally running mine over USB IP. When I switched to direct USB, these problems went away. I suggest trying a different USB controller, hub and cables and see if it's any more stable

cvhoang, 2020-08-06:
@thardie do you have your stick on a Raspberry Pi? Mine is connected directly to Intel NUC. I'm running Docker as well.

thardie, 2020-08-06:
Mine is on an openWRT device with an arm cpu. How are you getting the USB device into your docker container?

cvhoang, 2020-08-06:
configuration.yaml:

```
serial:
  port: >-
    /dev/serial/by-id/usb-Texas_Instruments_XDS110__03.00.00.13__Embed_with_CMSIS-DAP_L11008CR-if00
```

docker-compose.yaml:

```
  zigbee2mqtt:
    container_name: z2m
    volumes:
      - /opt/docker/zigbee2mqtt:/app/data
      - /opt/zigbee2mqtt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters
      - /opt/zigbee2mqtt/homeassistant.js:/app/lib/extension/homeassistant.js
    devices:
      - /dev/serial/by-id/usb-Texas_Instruments_XDS110__03.00.00.13__Embed_with_CMSIS-DAP_L11008CR-if00
    environment:
      - TZ
    image: koenkk/zigbee2mqtt
    restart: always
```

radinsky, 2021-08-29:
I have the same issue now, might be not enough power for this stick?

muspelheim, 2021-12-04:
@Koenkk @radinsky I had the same issue on my nuc with installed ubuntu 20.04. i found that stick works nicely from time to time and then unplug/plug is required to make it work.

In the log i found this one:
```
Zigbee2MQTT:debug 2021-12-04 12:59:39: Error: Command 0x60a423fffea3b006/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
```

I tried to solve this problem and i found that a lot of people with Nuc/Dell OptiPlex micro/etc platform faced with this problem and i found what is the reason of this behavior.

following the documentation:
```
USB autosuspend is a feature designed to preserve battery life, its' 18.04 default is "enabled", whether or not the device Ubuntu is being installed on even has a battery.
```

run the command in terminal `cat /sys/module/usbcore/parameters/autosuspend` if it will return 1-2 this is the thing why you got the error with the stick. to fix this - disable the autosustend:

```
sed -i 's/GRUB_CMDLINE_LINUX_DEFAULT="/&usbcore.autosuspend=-1 /' /etc/default/grub
update-grub
systemctl reboot
```

Koenkk, 2021-12-05:
@muspelheim thanks, added this to the FAQ

radinsky, 2021-12-05:
@muspelheim thanks for this finding! unfortunately, it didn't help me, still getting the adapter fails after some (short) time, as well I'm getting the same `SRSP - AF - dataRequestExt after 6000ms` error and cannot reconnect until I remove and plug back the adapter.

mbalko, 2021-12-10:
Well, I'm running zigbee2mqtt along with HA on RPi3B+ and have the same issues. After running `cat /sys/module/usbcore/parameters/autosuspend` I get `2`, so it seems like the problem really can be here. However, I can't find any working solution to turn it off on a Raspberry Pi (grub is not present in those).

muspelheim, 2021-12-10:
@mbalko on my raspberry it also 2, but it works on RPI. The thing that i said was that it only helps ubuntu (i have not tested on other distributives)
```
pi@rpi-02:~$ cat /sys/module/usbcore/parameters/autosuspend
2
```

sorentorp, 2022-02-09:
Getting this problem also, from one day to the other. Finds all devices on every restart, but cant get communication thru. Anyone solve this?
I'm on a TI-CC2531, Zigbee2MQTT Latest docker image, Linux Debian, autosuspend=-1

sorentorp, 2022-02-09:
Maybe this helps?

> zigbee2mqtt    | Zigbee2MQTT:info  2022-02-09 23:41:35: Checking if update available for 'Ikea Dimmer kælder'
> zigbee2mqtt    | Zigbee2MQTT:info  2022-02-09 23:41:35: Checking if update available for 'Kældertrappe 2'
> zigbee2mqtt    | Zigbee2MQTT:info  2022-02-09 23:41:35: Checking if update available for 'Karla natlampe'
> zigbee2mqtt    | Zigbee2MQTT:info  2022-02-09 23:41:35: Checking if update available for 'Kældertrappe 1'
> zigbee2mqtt    | Zigbee2MQTT:info  2022-02-09 23:41:35: Checking if update available for '0x5c0272fffe520cbd'
> zigbee2mqtt    | Zigbee2MQTT:info  2022-02-09 23:41:35: Checking if update available for 'radiator_stuen'
> zigbee2mqtt    | Zigbee2MQTT:info  2022-02-09 23:41:57: Starting network scan (includeRoutes 'false')
> zigbee2mqtt    | (node:7) UnhandledPromiseRejectionWarning: Error: Timeout - 39028 - 1 - null - 25 - 1 after 30000ms
> zigbee2mqtt    |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
> zigbee2mqtt    |     at listOnTimeout (internal/timers.js:557:17)
> zigbee2mqtt    |     at processTimers (internal/timers.js:500:7)
> zigbee2mqtt    | (Use `node --trace-warnings ...` to show where the warning was created)
> zigbee2mqtt    | (node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
> zigbee2mqtt    | (node:7) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
> zigbee2mqtt    | (node:7) UnhandledPromiseRejectionWarning: Error: Timeout - 5986 - 1 - null - 25 - 1 after 30000ms
> zigbee2mqtt    |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
> zigbee2mqtt    |     at listOnTimeout (internal/timers.js:557:17)
> zigbee2mqtt    |     at processTimers (internal/timers.js:500:7)
> zigbee2mqtt    | (node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
> zigbee2mqtt    | (node:7) UnhandledPromiseRejectionWarning: Error: Timeout - 7269 - 1 - null - 25 - 1 after 30000ms
> zigbee2mqtt    |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
> zigbee2mqtt    |     at listOnTimeout (internal/timers.js:557:17)
> zigbee2mqtt    |     at processTimers (internal/timers.js:500:7)
> zigbee2mqtt    | (node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 3)
> zigbee2mqtt    | (node:7) UnhandledPromiseRejectionWarning: Error: Timeout - 23168 - 1 - null - 25 - 1 after 30000ms
> zigbee2mqtt    |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
> zigbee2mqtt    |     at listOnTimeout (internal/timers.js:557:17)
> zigbee2mqtt    |     at processTimers (internal/timers.js:500:7)
> zigbee2mqtt    | (node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 4)
> zigbee2mqtt    | (node:7) UnhandledPromiseRejectionWarning: Error: Timeout - 52789 - 1 - null - 25 - 1 after 30000ms
> zigbee2mqtt    |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
> zigbee2mqtt    |     at listOnTimeout (internal/timers.js:557:17)
> zigbee2mqtt    |     at processTimers (internal/timers.js:500:7)
> zigbee2mqtt    | (node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 5)
> zigbee2mqtt    | (node:7) UnhandledPromiseRejectionWarning: Error: Timeout - 8228 - 1 - null - 25 - 1 after 30000ms
> zigbee2mqtt    |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
> zigbee2mqtt    |     at listOnTimeout (internal/timers.js:557:17)
> zigbee2mqtt    |     at processTimers (internal/timers.js:500:7)
> zigbee2mqtt    | (node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 6)

thardie, 2022-02-10:
In all the cases we've seen this it is due to some instability over USB. Like USB over IP. Faulty USB hub. Bad USB cable, etc.

Kimovitzh, 2022-04-27:
Brand new Raspberry pi 4. CC2652R directly in the the USB3 port. Using original raspberry USB-C power supply. Flashed with CC2652R_coordinator_20220219.hex file.

So only the the autosuspend thing is left.
Really no way of disabling this?
I almost has this error every day, sometimes twice a day.

grantbeattie, 2022-05-16:
> Brand new Raspberry pi 4. CC2652R directly in the the USB3 port. Using original raspberry USB-C power supply. Flashed with CC2652R_coordinator_20220219.hex file.
> 
> So only the the autosuspend thing is left. Really no way of disabling this? I almost has this error every day, sometimes twice a day.

Have you tried adding `usbcore.autosuspend=-1` to the kernel command line in `/boot/cmdline.txt`?

that does it for me on rpi3 w/ Debian bullseye:

```
# cat /sys/module/usbcore/parameters/autosuspend
-1
```

jeroenbeuz, 2023-12-21:
> @Koenkk @radinsky I had the same issue on my nuc with installed ubuntu 20.04. i found that stick works nicely from time to time and then unplug/plug is required to make it work.
>
> In the log i found [...]

Thank you so much! Have been looking everywhere for a fix, was just thinking about resetting my coordinator and rejoining my 80+ devices.

Some background info. I migrated my setup from a Raspberry Pi4 to an Intel Nuc. Made a backup of the zigbee network and everything worked without re-joining devices. However every other day I would notice that no zigbee devices were working for a couple of minutes. I really hope this will fix the issue. Will report back here in a couple of days to confirm.

TonTon-Macout, 2024-01-25:
the same error, but with Ham Geek POE Zigbee 3.0, tried via LAN and via Wifi and powered by USB and POE. I get this error once a day and everything stops working until I turn off the power. 

I put a wi-fi outlet and when unavailable devices appear, it switches the power, the last lines of the log are the result of this reboot. if you do not reboot, then gradually everything becomes inaccessible with this error

```
info  2024-01-23 22:17:30: MQTT publish: topic 'zigbee2mqtt/ВАННАЯ -  Зеркало', payload '{"consumption":1183.2899169921875,"current":8.22,"device_temperature":46,"energy":1183.29,"interlock":false,"last_seen":"2024-01-23T19:17:30.880Z","linkquality":72,"power":1383.89,"power_outage_count":1394,"power_outage_memory":null,"state_l1":"ON","state_l2":"OFF","update":{"installed_version":47,"latest_version":47,"state":"idle"},"update_available":false,"voltage":225.66}'
debug 2024-01-23 22:17:32: Received Zigbee message from 'ВАННАЯ -  Зеркало', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":1227.934814453125}' from endpoint 1 with groupID 0
info  2024-01-23 22:17:32: MQTT publish: topic 'zigbee2mqtt/ВАННАЯ -  Зеркало', payload '{"consumption":1183.2899169921875,"current":8.22,"device_temperature":46,"energy":1183.29,"interlock":false,"last_seen":"2024-01-23T19:17:32.834Z","linkquality":76,"power":1227.93,"power_outage_count":1394,"power_outage_memory":null,"state_l1":"ON","state_l2":"OFF","update":{"installed_version":47,"latest_version":47,"state":"idle"},"update_available":false,"voltage":225.66}'
debug 2024-01-23 22:17:36: Received Zigbee message from 'Гардеробная -  розетка 3д принтера', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":22648}' from endpoint 1 with groupID 0
info  2024-01-23 22:17:36: MQTT publish: topic 'zigbee2mqtt/Гардеробная -  розетка 3д принтера', payload '{"current":0,"last_seen":"2024-01-23T19:17:36.885Z","linkquality":105,"power":0,"state":"OFF","voltage":226.48}'
debug 2024-01-23 22:17:41: Received Zigbee message from 'ДЕТСКАЯ - Розетка - Лампа', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":100,"rmsVoltage":22917}' from endpoint 1 with groupID 0
info  2024-01-23 22:17:41: MQTT publish: topic 'zigbee2mqtt/ДЕТСКАЯ - Розетка - Лампа', payload '{"current":0.04,"last_seen":"2024-01-23T19:17:41.673Z","linkquality":72,"power":10,"state":"ON","voltage":229.17}'
debug 2024-01-23 22:17:42: Received Zigbee message from 'Прихожая - розетка зеркала', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":22984}' from endpoint 1 with groupID 0
info  2024-01-23 22:17:42: MQTT publish: topic 'zigbee2mqtt/Прихожая - розетка зеркала', payload '{"current":0,"last_seen":"2024-01-23T19:17:42.230Z","linkquality":72,"power":0,"state":"OFF","voltage":229.84}'
debug 2024-01-23 22:17:42: Received Zigbee message from 'Гирлянда - розетка', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
info  2024-01-23 22:17:42: MQTT publish: topic 'zigbee2mqtt/Гирлянда - розетка', payload '{"child_lock":"UNLOCK","current":0,"energy":0.46,"indicator_mode":"off/on","last_seen":"2024-01-23T19:17:42.720Z","linkquality":163,"power":0,"power_outage_memory":"off","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":225}'
debug 2024-01-23 22:17:44: Received Zigbee message from 'ВАННАЯ -  Зеркало', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":1095.651611328125}' from endpoint 1 with groupID 0
info  2024-01-23 22:17:44: MQTT publish: topic 'zigbee2mqtt/ВАННАЯ -  Зеркало', payload '{"consumption":1183.2899169921875,"current":8.22,"device_temperature":46,"energy":1183.29,"interlock":false,"last_seen":"2024-01-23T19:17:44.843Z","linkquality":80,"power":1095.65,"power_outage_count":1394,"power_outage_memory":null,"state_l1":"ON","state_l2":"OFF","update":{"installed_version":47,"latest_version":47,"state":"idle"},"update_available":false,"voltage":225.66}'
debug 2024-01-23 22:17:46: Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"warn"}},"transaction":"hu9ej-2"}'
warn  2024-01-24 06:21:46: Failed to ping 'ОСВЕЩЕНИЕ - Туалет' (attempt 1/2, Read 0xa4c138d9a47ac4a7/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 42878 - 1 - 49 - 0 - 1 after 10000ms))
warn  2024-01-24 06:22:13: Failed to ping 'ОСВЕЩЕНИЕ - Туалет' (attempt 2/2, Read 0xa4c138d9a47ac4a7/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 42878 - 1 - 50 - 0 - 1 after 10000ms))
warn  2024-01-24 06:32:27: Failed to ping 'ОСВЕЩЕНИЕ - Туалет' (attempt 1/1, Read 0xa4c138d9a47ac4a7/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed  [... cut ...]
warn  2024-01-25 12:03:41: Failed to ping 'ЗАЛ - Освещение' (attempt 2/2, Read 0xdc8e95fffef44318/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
error 2024-01-25 12:04:38: Adapter disconnected, stopping
error 2024-01-25 12:04:39: Failed to stop Zigbee2MQTT
```

fhollermayer, 2024-07-17:
If you're experiencing issues with USB autosuspend on a Raspberry Pi running Linux, it's worth noting that Raspberry Pi OS (and similar distributions) typically use `cmdline.txt` instead of GRUB for boot parameters.  To apply the solution described by @muspelheim on Raspberry Pi OS I did the following:

1. Verify that `autosuspend` is actually the problem (value 1 - 2)
```
cat /sys/module/usbcore/parameters/autosuspend
```
2. Edit `cmdline.txt`:

Use a text editor to edit the /boot/cmdline.txt file. For example, with nano:
```
sudo nano /boot/cmdline.txt
```
Add the following parameter to the file:
```
usbcore.autosuspend=-1
```
Ensure all parameters in `cmdline.txt` are on a single line, separated by spaces. Do not create new lines.

3. Reboot
```
sudo reboot
```

4. Verify
```
cat /sys/module/usbcore/parameters/autosuspend
```

Dinth, 2024-11-10:
Thanks for the tips above, but how can i set autosuspend=-1 on HASS?

slyglif, 2025-05-30:
I realize this is currently marked closed, but I'm hoping someone sees this.  Would it be possible to add a binary sensor such that when this error starts to happen, it sets the sensor to On?  That would make it a lot easier to detect it from HA and have an automation setup to do something to correct it.