Koenkk/zigbee2mqtt #2309: zigbee2mqtt v1.7.0 issue with friendly names

pkedvessy, 2019-11-10:
Sending zigbee message to Xiaomi wall switches using friendly name in the mqtt topic stopped working with 1.7.0. There was no issue in 1.6.0 at all.

Zigbee2Mqtt & coordinator versions:
`{"version":"1.7.0","commit":"551a58a","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}`

Device configuration in configuration.yml (base_topic: zigbee2mqtt):
`  '0x00158d0003151f2f':
    friendly_name: ground_floor/kitchen/wall_switch/2
    retain: true`

Mqtt topic
`zigbee2mqtt/ground_floor/kitchen/wall_switch/2/left/set`

Data
`{"state": "OFF"}`

Error:
`Nov 10 20:23:01 peetserver npm[3869]: zigbee2mqtt:error 2019-11-10T19:23:01: Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'ID' of undefined
Nov 10 20:23:01 peetserver npm[3869]:     at Object.entries.find (/opt/zigbee2mqtt/lib/zigbee.js:184:106)
Nov 10 20:23:01 peetserver npm[3869]:     at Array.find (<anonymous>)
Nov 10 20:23:01 peetserver npm[3869]:     at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:184:76)
Nov 10 20:23:01 peetserver npm[3869]:     at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/entityPublish.js:55:36)
Nov 10 20:23:01 peetserver npm[3869]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:320:44)
Nov 10 20:23:01 peetserver npm[3869]:     at Controller.onMQTTMessage (/opt/zigbee2mqtt/lib/controller.js:239:14)
Nov 10 20:23:01 peetserver npm[3869]:     at MQTT.emit (events.js:189:13)
Nov 10 20:23:01 peetserver npm[3869]:     at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.js:86:14)
Nov 10 20:23:01 peetserver npm[3869]:     at MqttClient.emit (events.js:189:13)
Nov 10 20:23:01 peetserver npm[3869]:     at MqttClient._handlePublish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:1162:12))`

Using the device id as a topic is still workin eg.:
`zigbee2mqtt/0x00158d0003151f2f/left/set`

Status reports are still available with the friendly name eg.:
`zigbee2mqtt/ground_floor/kitchen/wall_switch/2`

I tried to repair the switch but friendly name issue is still exists.

sjorge, 2019-11-10:
I am also using '/' as a room separator and had some issues during the dev phase, for me it was working after some changes done by Koen. But I only have 2 levels while you seem to have 4... this may be the problem.

Can you try with the friendly_name: `ground_floor/kitchen/wall_switch_2` to see if it is a divider limit or something else.

pkedvessy, 2019-11-10:
Good catch @sjorge ! Reducing the number of separators in the friendly name solved my issue.
@Koenkk is there a reason on limiting the depth of mqtt topic?

sjorge, 2019-11-10:
https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/entityPublish.js#L24 Seems to hint that it will accept 20 x '/' but something else might be causing the issue... I think it might be the end point selector since the switch seems to have a left and right endpoint.

I have no such devices so I did not run into it during the dev phase.

pkedvessy, 2019-11-10:
@sjorge I'm fine with reducing the depth of mqtt topics. But this restriction should be documented or fixed if there is no reason. Thanks for your help!

sjorge, 2019-11-10:
I think the idea still is to limit it to 20ish, yeah the exact value should be documented. But I don't think you should be hitting it yet either. I'm poking around in entityPublish a bit.

pkedvessy, 2019-11-10:
Yeah, 20ish should be more then enough 😄 Thanks again for your quick and usefull response.

sjorge, 2019-11-10:
After more digging it fails here: https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/zigbee.js#L184

Koenkk, 2019-11-11:
Fixed in the latest dev, the amount of slashes are not the cause of this issue, the bug occured due to `/2/`.

closed 2019-11-11 by Koenkk

viart, 2019-11-27:
After upgrading to the latest version I'm experiencing somewhat a similar issue.

I have a light bulb with a `friendly_name: middle/livingroom/bulb/1` and it's registering and reporting the state properly, but failing when I try to change it (either by manually publishing or through the Homeassistant) z2m complains:
```
2019-11-27T19:24:34: Entity 'middle/livingroom/bulb/1' is unknown
```

I see that it cuts `/1` due to this (hence it can't find the entity further down):
https://github.com/Koenkk/zigbee2mqtt/blob/b459c352191091af9b9eafec82a98f73069797e8/lib/zigbee.js#L146-L149

...I don't really know the back story of that postfix(es) logic, but would it make sense instead of trying to find a parsed topic in devices list try to find `id/friendly_name` in a given topic, somemthing like (simplified):
```js
const entity = devices.find(d => key.startsWith(d.id) || key.startsWith(d.friendlyName))
const postfix = entity ? key.replace(entity.friendlyName, '').replace(entity.id, '') : ''
```

In this case a whitelist of postfixes should not be needed, @Koenkk what do you think?

Koenkk, 2019-11-28:
Zigbee2mqtt interpets the `1` as the endpoint. Allowing this would cause a lot of ambiguity, so friendly names ending with only a `/NUMBER` should not be allowed. Todo: add to docs and update code.