Koenkk/zigbee2mqtt #2310: Heiman HS2SK: null value instead of 0 for all power metering data

TheSkater, 2019-11-10:
I switched just to the developer version of zigbe2mqtt. 
I am using a HS2SK (Heiman Plug with power metering) and have still "null" values in my MQTT messages. This leads to an error when I take this result as input to a simple influxdb output node in NodeRed:

> "Error: A 400 Bad Request error occurred: {"error":"unable to parse 'zigbee2mqtt/Keller/Wasserwerk/Zigbee/Schalter current=null,linkquality=15,power=null,state=\"ON\",voltage=null': invalid number"}↵"

See coressponding MQTT message: 
`info  2019-11-10T20:28:56: MQTT publish: topic 'zigbee2mqtt/Keller/Wasserwerk/Zigbee/Schalter', payload '{"state":"ON","linkquality":15,"power":null,"current":null,"voltage":null}`

It seems that usage data like power or voltage are not read from the device.
I have the following device:
`{"ieeeAddr":"0x000d6f000b7a7240","type":"Router","networkAddress":15652,"model":"HS2SK","friendly_name":"Keller/Wasserwerk/Zigbee/Schalter","manufacturerID":4619,"**manufacturerName":"Heiman","powerSource":"Mains (single phase)","modelID":"SmartPlug","hardwareVersion":16,"dateCode":"2018.1.11**","lastSeen":1573417736631}`
`

_Originally posted by @TheSkater in https://github.com/Koenkk/zigbee2mqtt/issues/2277#issuecomment-552232806_

Koenkk, 2019-11-10:
Please provide the debug log when receiving the message.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

TheSkater, 2019-11-11:
Here is the log file. The Heiman HS2SK device is name "Keller/Wasserwerk/Zigbee/Schalter"
[log-zigbee2mqtt.txt](https://github.com/Koenkk/zigbee2mqtt/files/3830292/log-zigbee2mqtt.txt)

Koenkk, 2019-11-11:
In this log the power values are not received from the device, the only publish I see is the one on startup but these are cached values:

```
info  2019-11-11T08:06:24: MQTT publish: topic 'zigbee2mqtt/Keller/Wasserwerk/Zigbee/Schalter', payload '{"state":"ON","linkquality":5,"power":null,"current":null,"voltage":null}'
```

TheSkater, 2019-11-11:
Thx, but how can I bring the device to publish the metering data?

Koenkk, 2019-11-11:
Attach a load to it.

TheSkater, 2019-11-11:
I did, but NULL values will not change.  Is there a message to reset the device or  to configure it?

TheSkater, 2019-11-11:
Also, the device does not send messages automatically. I have to send a "get" message first 
to receive any values.

Koenkk, 2019-11-11:
Can you share your `database.db`?

TheSkater, 2019-11-11:
for sure
[database.db.txt](https://github.com/Koenkk/zigbee2mqtt/files/3833072/database.db.txt)

Koenkk, 2019-11-12:
@matejzero does this work properly for you know with zigbee2mqtt 1.7.0, also does it automatically report the new power values after adding a load?

matejzero, 2019-11-12:
I somehow missed notifications from this thread. I'll upgrade to edge today in home assistant, check it and report back.

matejzero, 2019-11-12:
I can confirm that it's working with 1.7.0.

This is how my debug looks like now:
```
zigbee2mqtt:debug 2019-11-12T16:49:59: Received Zigbee message from '0x000d6f000b7ad5b0', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":23175}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2019-11-12T16:49:59: MQTT publish: topic 'zigbee2mqtt/0x000d6f000b7ad5b0', payload '{"state":"ON","linkquality":5,"power":0,"current":0,"voltage":231.75}'
```

It looks like zigbee2mqtt reports power and current correctly as 0 and not as null.

Koenkk, 2019-11-12:
@TheSkater are you sure you didn't map a custom zigbee-herdsman-converters?

TheSkater, 2019-11-12:
What do you mean with that and how should I test it?

TheSkater, 2019-11-12:
I have reset the device and diconnected it from zigbee2mqtt. I restartet zigbee2mqtt and paired the device newly and renamed it. Now I receive mqqt message like this without any meetering data:
`{"state":"ON","linkquality":0}`
So it seemes that is not recocknized as metering device and only as a plug?! Does that make sense?

Koenkk, 2019-11-14:
Before re-pairing, did you do https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove (this is crucial otherwise the device won't be configured again to provide power measurements)

TheSkater, 2019-11-14:
I remove the device completely (also with the force command). After re-pairing the device receives a new "friendly name" which is idential to its ieeeAdr. This should show that is was newly configured.
But I receive only message from zigbee2mqtt with state and linkquality values.
What else can I try?

Koenkk, 2019-11-15:
Can you sniff the log when pairing the device (remove it first) and then attaching a load to it (which should trigger power measurments): https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html.

PedroLamas, 2019-11-21:
I'm having the same problem with 2 HS2SK devices I received today. Freshly paired with the network, all I get is this:

```
zigbee2mqtt:info  2019-11-21T16:53:14: MQTT publish: topic 'zigbee2mqtt/SmartPlug4', payload '{"state":"OFF","linkquality":23}'
zigbee2mqtt:info  2019-11-21T16:53:14: MQTT publish: topic 'zigbee2mqtt/SmartPlug5', payload '{"state":"OFF","linkquality":28}'
```

I did notice something, shouldn't [this bind](https://github.com/Koenkk/zigbee-herdsman-converters/blob/bf7e5a7ade1e71b0f55077042302a95667c0fcd7/devices.js#L3812) include `haElectricalMeasurement` ?

PedroLamas, 2019-11-21:
Removing and adding smartplug5 resulted in this:

```
zigbee2mqtt:info  2019-11-21T17:05:52: Device 'SmartPlug5' joined
zigbee2mqtt:info  2019-11-21T17:05:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"SmartPlug5"}}'
zigbee2mqtt:info  2019-11-21T17:05:52: Starting interview of 'SmartPlug5'
zigbee2mqtt:info  2019-11-21T17:05:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"SmartPlug5"}}'
zigbee2mqtt:debug 2019-11-21T17:05:52: Device 'SmartPlug5' announced itself
zigbee2mqtt:debug 2019-11-21T17:05:54: Received Zigbee message from 'SmartPlug5', type 'readResponse', cluster 'genBasic', data '{"modelId":"SmartPlug","manufacturerName":"Heiman","powerSource":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-11-21T17:05:54: No converter available for 'HS2SK' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"SmartPlug","manufacturerName":"Heiman","powerSource":1}'
zigbee2mqtt:debug 2019-11-21T17:05:54: Received Zigbee message from 'SmartPlug5', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2,"appVersion":17}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-11-21T17:05:54: No converter available for 'HS2SK' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":2,"appVersion":17}'
zigbee2mqtt:debug 2019-11-21T17:05:54: Received Zigbee message from 'SmartPlug5', type 'readResponse', cluster 'genBasic', data '{"hwVersion":16,"dateCode":"2018.1.11"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-11-21T17:05:54: No converter available for 'HS2SK' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":16,"dateCode":"2018.1.11"}'
zigbee2mqtt:info  2019-11-21T17:05:54: Successfully interviewed 'SmartPlug5', device has successfully been paired
zigbee2mqtt:info  2019-11-21T17:05:54: Device 'SmartPlug5' is supported, identified as: HEIMAN Smart metering plug (HS2SK)
zigbee2mqtt:info  2019-11-21T17:05:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"SmartPlug5","model":"HS2SK","vendor":"HEIMAN","description":"Smart metering plug","supported":true}}'
zigbee2mqtt:info  2019-11-21T17:05:54: Configuring 'SmartPlug5'

...

zigbee2mqtt:info  2019-11-21T17:05:55: Succesfully configured 'SmartPlug5'
```

PedroLamas, 2019-11-21:
Here's the database entry:

```
{
    "id": 3,
    "type": "Router",
    "ieeeAddr": "0x000d6f000be64a15",
    "nwkAddr": 29397,
    "manufId": 4619,
    "manufName": "Heiman",
    "powerSource": "Mains (single phase)",
    "modelId": "SmartPlug",
    "epList": [
      1
    ],
    "endpoints": {
      "1": {
        "profId": 260,
        "epId": 1,
        "devId": 81,
        "inClusterList": [
          0,
          3,
          4,
          6,
          9,
          1794,
          2820
        ],
        "outClusterList": [
          3,
          25
        ],
        "clusters": {
          "genBasic": {
            "attributes": {
              "modelId": "SmartPlug",
              "manufacturerName": "Heiman",
              "powerSource": 1,
              "zclVersion": 2,
              "appVersion": 17,
              "hwVersion": 16,
              "dateCode": "2018.1.11"
            }
          }
        },
        "binds": [
          {
            "cluster": 6,
            "type": "endpoint",
            "deviceIeeeAddress": "0x00124b0019366c03",
            "endpointID": 1
          }
        ]
      }
    },
    "appVersion": 17,
    "hwVersion": 16,
    "dateCode": "2018.1.11",
    "zclVersion": 2,
    "interviewCompleted": true,
    "meta": {
      "configured": 2
    }
  }
```

Seems way too "light"...

Koenkk, 2019-11-21:
> I'm having the same problem with 2 HS2SK devices I received today. Freshly paired with the network, all I get is this:
>
> ```
> zigbee2mqtt:info  2019-11-21T16:53:14: MQTT publish: topic [...]

- with zigbee-herdsman less info is stored in the `database.db`, no need to worry about this it looks fine
- The missing `haElectricalMeasurement` in binds is a good point, can you try adding it?

PedroLamas, 2019-11-21:
Oh, I *did* add it before and nothing happened, but seems that pairing after the change does provide values!

```
zigbee2mqtt:info  2019-11-21T19:55:32: Device 'SmartPlug5' joined
zigbee2mqtt:info  2019-11-21T19:55:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"SmartPlug5"}}'
zigbee2mqtt:info  2019-11-21T19:55:32: Starting interview of 'SmartPlug5'
zigbee2mqtt:info  2019-11-21T19:55:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"SmartPlug5"}}'
zigbee2mqtt:debug 2019-11-21T19:55:32: Device 'SmartPlug5' announced itself
zigbee2mqtt:debug 2019-11-21T19:55:33: Received Zigbee message from 'SmartPlug5', type 'readResponse', cluster 'genBasic', data '{"modelId":"SmartPlug","manufacturerName":"Heiman","powerSource":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-11-21T19:55:33: No converter available for 'HS2SK' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"SmartPlug","manufacturerName":"Heiman","powerSource":1}'
...
zigbee2mqtt:debug 2019-11-21T19:55:33: Received Zigbee message from 'SmartPlug5', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2,"appVersion":17}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-11-21T19:55:33: No converter available for 'HS2SK' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":2,"appVersion":17}'
zigbee2mqtt:debug 2019-11-21T19:55:33: Received Zigbee message from 'SmartPlug5', type 'readResponse', cluster 'genBasic', data '{"hwVersion":16,"dateCode":"2018.1.11"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-11-21T19:55:33: No converter available for 'HS2SK' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":16,"dateCode":"2018.1.11"}'
zigbee2mqtt:info  2019-11-21T19:55:33: Successfully interviewed 'SmartPlug5', device has successfully been paired
zigbee2mqtt:info  2019-11-21T19:55:33: Device 'SmartPlug5' is supported, identified as: HEIMAN Smart metering plug (HS2SK)
zigbee2mqtt:info  2019-11-21T19:55:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"SmartPlug5","model":"HS2SK","vendor":"HEIMAN","description":"Smart metering plug","supported":true}}'
zigbee2mqtt:info  2019-11-21T19:55:33: Configuring 'SmartPlug5'
...
zigbee2mqtt:info  2019-11-21T19:55:34: Succesfully configured 'SmartPlug5'
zigbee2mqtt:debug 2019-11-21T19:55:35: Received Zigbee message from 'SmartPlug5', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":23729}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2019-11-21T19:55:35: MQTT publish: topic 'zigbee2mqtt/SmartPlug5', payload '{"state":"OFF","linkquality":47,"voltage":237.29}'
```

PedroLamas, 2019-11-21:
Confirmed it, needs `haElectricalMeasurement` on the bind! Will send a PR in a few minutes...

PedroLamas, 2019-11-21:
I did notice it takes about 5 seconds between updates, is this expected?

Turning off and seeing the state change is immediate but the power value stakes a bit to go back to 0!

PedroLamas, 2019-11-21:
Here's the PR: https://github.com/Koenkk/zigbee-herdsman-converters/pull/767

Koenkk, 2019-11-22:
@PedroLamas that is indeed expected, the values have to change above a certain threshold before they are reported again (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L216)

closed 2019-11-22 by Koenkk

PedroLamas, 2019-11-22:
@Koenkk but shouldn't it update immediately to 0 when I turn it off? Because right now seems to take a couple of seconds before that actually happens!

Koenkk, 2019-11-22:
@PedroLamas you can try playing with the values (change `minimumReportInterval` to `0` e.g) and see if the behaviour changes. Note that after every change you have to bump the configureKey in order for the device to be reconfigured.

PedroLamas, 2019-11-22:
@Koenkk oh so that's what that key is about!!! 😂 I've been having some trouble understanding all the keys and values involved, but that is one less to worry!

I'll open a separate ticket If I find some better working values!

Thank you for all your work on this project (do expect a PayPal donation from me in the next few minutes!)

TheSkater, 2019-11-23:
I updates to the actual zigbee2mqtt DEV release. My HS2SK device does still not deliver any measurement data. Do I nedd to do somethings else? My orignal reqest is still open.

Koenkk, 2019-11-24:
It has been added to the latest dev yesterday, please update to that and try again

TheSkater, 2019-11-24:
Great, it's working now.

TheSkater, 2019-11-25:
But.......
When I stop the zigbee2mqtt service and start it it again I have to pair the device manally again!
All devices are present, only the HS2SK is missing.