Koenkk/zigbee2mqtt #2312: CC2531 Zigbee 3.0 router

lubbertkramer, 2019-11-11:
#What happened

After the release of 1.7 i wanted to try Zigbee2MQTT as Deconz replacement. I wanted to use my CC2531 sticks that i have as router in my network and was wondering why there are no 3.0 router files because there are zigbee 3.0 coordinator files

## What did you expect to happen
Router Zigbee 3.0 for CC2531

## Improvement
Will there be improvements when i'm using the CC2531 as zigbee 3.0 router instead of 1.2 or doesnt this matter?

Koenkk, 2019-11-11:
@ptvoinfo do you have any interest in providing a zigbee 3.0 router firmware?

> Will there be improvements when i'm using the CC2531 as zigbee 3.0 router instead of 1.2 or doesnt this matter?

In practice probably not.

ptvoinfo, 2019-11-11:
I'm still using v1.2 routers in my network without any problems. Therefore I didn't try to create a firmware for v3.

closed 2019-11-11 by lubbertkramer

David-Lor, 2021-10-04:
@lubbertkramer @ptvoinfo What version are you using in your coordinators? I was wondering if it is possible to use a cc2531 as router (thus with zigbee 1.2), having a 3.0 coordinator.

ptvoinfo, 2021-10-05:
@David-Lor I use Zigbee 1.2 for CC2530 and CC2531.