Koenkk/zigbee2mqtt #2314: (Suggestion/Tip) Zigbee 3.0 Devices

jasimancas, 2019-11-11:
Hi,

I have seen the clear benefits of the Zigbee 3.0 network versus 1.2.
Would it be possible that in the Supported Devices section (https://www.zigbee2mqtt.io/information/supported_devices.html) it would be indicated which version of Zigbee supports each one?

I understand that if I have all Zigbee 1.2 devices, I won't have the advantages of migrating to 3.0, will I?

A row with the Zigbee version of each device would be great (might help with some)

ciotlosm, 2019-11-11:
Would be nice for that page to include a few more filters:
- Is Router
- Powered by (batter vs power line)
- Manufacturer
- Type (lightbulb, relay, switch, motion, etc)

jasimancas, 2019-11-11:
And for example split data in few rows (description, command (on/off, brightness), etc)

Koenkk, 2019-11-11:
The manufacturer is already there (vendor), the other information can be added to the meta (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L3774). 

Something like:
- `deviceType`: `Router` or `EndDevice`
- `powerSource`: `Battery` or `AC Mains`
-  `type`: `bulb`, `relay`, etc.

Feel free to make PR's, I will make sure they end up in the docs.

jasimancas, 2019-11-11:
I can, for example, split the description in two rows, one for description and another for "events" (on/off, etc)

And add another row with the ZIGBEE version (for example Xiaomi Zigbee 1.2)

Do you see it well?

I do the PR directly editing the file?

Koenkk, 2019-11-12:
Desciption and supports (events) are already splitted.

Yes please directly edit the files

closed 2019-11-23 by jasimancas