Koenkk/zigbee2mqtt #2315: Does not start after Distribution upgrade

MCWeid, 2019-11-11:
Hi There,
I upgraded my Raspberry today to buster distribution and my zigbee2mqtt stick is not starting anymore.

See error below:
 node index.js

                Zigbee2mqtt requires node version 10, you are running v12.13.0!

zigbee2mqtt:info  2019-11-11T13:41:13: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-11-11.14-41-13'
zigbee2mqtt:info  2019-11-11T13:41:13: Starting zigbee2mqtt version 1.7.0 (commit #551a58a)
zigbee2mqtt:info  2019-11-11T13:41:13: Starting zigbee-herdsman...
zigbee2mqtt:error 2019-11-11T13:41:14: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-11-11T13:41:14: Failed to start zigbee
zigbee2mqtt:error 2019-11-11T13:41:14: Exiting...
zigbee2mqtt:error 2019-11-11T13:41:14: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:109:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:4:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:107:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/stream.js:197:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/stream.js:241:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.7.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.7.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /opt/loxberry/.npm/_logs/2019-11-11T13_41_14_476Z-debug.log

Do I need to downgrade node to version 10 to make it work again?

gericho, 2019-11-11:
same to me, commented out username and password coz not needed anymore and it started normally, I'm not sure it can help

MCWeid, 2019-11-11:
Does not help in my case unfortunately. Why is the login not needed anymore?

gericho, 2019-11-11:
is not needed only in my case since I removed mqtt auth.

BTW, I think is a YAML code error, I made a backup of the /opt/zigbee2mqtt/data directory, removed the /opt/zigbee2mqtt directory completely, reinstall using the guide here at point 2 [(https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html)] edited the configuration.yaml in /opt/zigbee2mqtt/data, commented out everything that came standard in the file, and paste the old configuration. I also pasted the database.db and state.json, then restarted the zigbee2mqtt service

Koenkk, 2019-11-11:
Your adapter is in use by another process:

```
Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
```

Could it be that modemmanager is installed? See https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for more info.

closed 2020-01-19 by stale[bot]