Koenkk/zigbee2mqtt #2316: transition time sent as float by home assistant is ignored

davidbb, 2019-11-11:
# Bug Report
## What happened
Home Assistant 0.101 changed the data type of the ``transition`` parameter for MQTT lights from int to float [27253](https://github.com/home-assistant/home-assistant/pull/27253). I believe this is causing zigbee2mqtt to ignore the parameter and use the default transition (or none if none is configured). 

## What did you expect to happen

## How to reproduce it (minimal and precise)
Here is what my MQTT broker sees when I send a light.turn_off event to the ``tradfri_light`` device with a transition of 60. 

``
{"state": "OFF", "transition": 60.0}
``

But the zigbee2mqtt logs show: 
``
MQTT publish: topic 'zigbee2mqtt/tradfri_hall', payload '{"state":"OFF","linkquality":34,"brightness":255}'
``

The light turns off immediately.

## Debug Info
zigbee2mqtt version: 1.7.0
CC253X firmware version: (not sure)

davidbb, 2019-11-11:
Upon further investigation, the float/int change might be a red herring. I updated HA to 0.101 at the same time as my zigbee2mqtt update to 1.7.0. If I manually send an ``int`` transition I get the same result (transition ignored). 

``
mosquitto_pub -h 192.168.XX.XX -t "zigbee2mqtt/tradfri_hall/set" -m '{"state": "ON", "transition": 60}'
``
Results in:
``
zigbee2mqtt/tradfri_hall {"state":"ON","linkquality":34,"brightness":255}
``

Perhaps this is related to [1810](https://github.com/Koenkk/zigbee2mqtt/issues/1810), but not sure since I'm not passing brightness or color, only transition.

Koenkk, 2019-11-11:
The transition for the state has been removed as this caused bugs (https://github.com/Koenkk/zigbee2mqtt/issues/2220).

davidbb, 2019-11-12:
thank you for the clarification. for anyone trying to turn off their Ikea Tradfri lights with a transition, my workaround is to do it in 2 steps:

1) set the lights to 1% brightness with a transition time
2) turn the lights off

closed 2019-11-12 by davidbb

flaektrem, 2021-11-28:
@Koenkk still no fix for this? 
Really miss that feature to turn off lights with transition in just one step without any work around.

Elkropac, 2022-03-10:
Hi, 
when using HA and nodered, should i use MQTT to drive the lights instead of HA light entities? 
I would like to do some short turn transition when motion is detected and long turn off transition after timer times out (no motion) - like i do with wifi bulbs - but all changes are instant.
I tried to put two commands - first "turn_on" with brightness = 1 and then "turn_on" with target brightness and transition

fliespl, 2023-01-28:
I can confirm it's still an issue. If i use two properties effect is unknown. Most of the times if skips transition time.

soldag, 2023-09-19:
I just bought my first tradfri lights and can confirm what @fliespl is reporting. Any chance this will be fixed @Koenkk ?