Koenkk/zigbee2mqtt #2317: Set group properties added with bridge/config/add_group

yamlCase, 2019-11-11:
It appears there is no way to set the following properties through the `/bridge/config/add_group` MQTT topic:

```
retain: false
transition: 2
optimistic: true
 ```

sjorge, 2019-11-11:
That was actually intentional from my part when i did the change, as there was no update_group either. So I kept the scope small to just allow the id to be specified.

If we implement this we should also implement a update_group so we can change the settings at a later time.

yamlCase, 2019-11-11:
@sjorge glad to know this is actively being worked on!  
n my case, I'm defining all my groups in node-red and assigning light bulbs to the zigbee2mqtt groups there.  I find it much easier to "attach" and "detach" bulb nodes to group nodes, and group nodes to device nodes (remote buttons, switches, etc) than editing the groups.yaml.  Also, I don't have to restart z2m going this route.

sjorge, 2019-11-11:
That how the initial PR I did to add the 'id' got started, I wanted to create a group from node-red... but with a specific ID.

Currently I am hitting #2265 and that's a bit beyond my skill level to fix. But once that is fixed, I will probably look at extending add_group and adding an update_group mqtt topic.

closed 2020-01-19 by stale[bot]