Koenkk/zigbee2mqtt #2318: Occasionally get UnhandledPromiseRejectionWarning from waitress.js

sjorge, 2019-11-11:
# Bug Report
## What happened
```
(node:88401) UnhandledPromiseRejectionWarning: Error: Timeout - 36534 - 11 - 21 - 1 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:425:11)
    at tryOnTimeout (timers.js:289:5)
    at listOnTimeout (timers.js:252:5)
    at Timer.processTimers (timers.js:212:10)
(node:88401) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:88401) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

I get this error every now and then after having paired 2 new E1743 remotes. Both got unbound from group with ID 99 and bound to 2 other groups, each group has 1 bulb in it, one has an IKEA bulb the other a Hue bulb.

This may be related to the 'emulated' reporting, but I also got it at least once after removing a group.

## What did you expect to happen
n/a

## How to reproduce it (minimal and precise)
Not sure, I thought it might be the emulated reporting stuff for the hue bulb but I drop and raised the brightness for it a few times without triggering it.

## Debug Info
zigbee2mqtt version: 1.6.0-dev @ 2fb4e20
CC253X firmware version: 20190425

Koenkk, 2019-11-11:
Thanks for the logs, should be fixed in latest dev.

closed 2019-11-11 by sjorge

loglux, 2019-12-06:
I've got the same problem today
```

Dec 06 23:57:21 littlehelp npm[20312]: (node:20339) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
Dec 06 23:57:21 littlehelp npm[20312]: (node:20339) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

Koenkk, 2019-12-07:
@loglux thanks, but in order to solve this we need much more context and steps to reproduce

jffz, 2020-01-08:
Bumping this issue as I do also have it randomly after boot.

zigbee2mqtt version: 1.8.0-dev
CC253X firmware version: 20190608

jd1900, 2020-04-20:
I'm having a similar issue since this evening.

When pressing a switch through the HA UI it gets very laggy, to the extent that it doesn't work sometimes.

Here some logs:

`zigbee2mqtt:info 2020-04-20 22:47:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'relay_aqara_2' failed: 'Error: Command 0x00158deviceUUID/2 genOnOff.off({}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"relay_aqara_2"}}'(node:233) UnhandledPromiseRejectionWarning: Error: Timeout - 42437 - 1 - 143 - 6 - 11 after 6000ms at Timeout._onTimeout (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24) at listOnTimeout (internal/timers.js:531:17) at processTimers (internal/timers.js:475:7)(node:233) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 71)zigbee2mqtt:error 2020-04-20 22:47:22: Publish 'set' 'state' to 'relay_aqara_2' failed: 'Error: Command 0x00158deviceUUID/1 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC channel access failure' (225))'zigbee2mqtt:info 2020-04-20 22:47:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'relay_aqara_2' failed: 'Error: Command 0x00158deviceUUID/1 genOnOff.off({}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"relay_aqara_2"}}'(node:233) UnhandledPromiseRejectionWarning: Error: Timeout - 42437 - 2 - 144 - 6 - 11 after 6000ms at Timeout._onTimeout (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24) at listOnTimeout (internal/timers.js:531:17) at processTimers (internal/timers.js:475:7)(node:233) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 72)zigbee2mqtt:error 2020-04-20 22:47:33: Publish 'set' 'state' to 'relay_aqara_2' failed: 'Error: Command 0x00158deviceUUID/2 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC channel access failure' (225))'`

Also getting this error:
zigbee2mqtt:error 2020-04-20 22:48:06: Publish 'set' 'state' to 'relay_aqara_2' failed: 'Error: Command 0x00158deviceUUID/1 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC channel access failure' (225))'

The device is the same commented in here (Aqara LLKZMK11LM):
https://github.com/Koenkk/zigbee2mqtt/issues/3375

**Debug Info**
Zigbee2mqtt version: 1.12.2
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20191106.zip

Koenkk, 2020-04-21:
@jd1900 

The error indicates that the RF channel is too busy

- Is this when controlling many devices at once?
- Could there be interference from other (wifi?) network? 
   - https://support.metageek.com/hc/en-us/articles/203845040-ZigBee-and-WiFi-Coexistence

jd1900, 2020-04-21:
> @jd1900
> 
> The error indicates that the RF channel is too busy
> 
> * Is this when controlling many devices at once?
> * Could there be interference from other (wifi?) network?
>   
>   * https://support.metageek.com/hc/en-us/articles/203845040-ZigBee-and-WiFi-Coexistence

I wouldn't say that's the case. I barely have 5 devices to test, and 2 are with battery. 
I've solved in an easy way. Remove the coordinator (CC26X2R1) USB from the Raspberry PI, restart, and reconnect the USB. 
If I just reboot the RPi it doesn't improve, but when removing the USB it improves. I don't know the reason..., but every few days with the coordinator ON (CC26X2R1) I need to unplug it and re-plug after a reboot. Just a reboot is not enough.

Koenkk, 2020-04-21:
@jd1900 can you try with the latest firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin ?

jd1900, 2020-04-21:
Thank you. I've just installed, and so far, so good. 

But it's been only some minutes. I'll report in the next days if I see errors.