Koenkk/zigbee2mqtt #2319: Device not paring after inital connection

arcanefoam, 2019-11-11:
# Bug Report
## What happened
I was trying to add support for a new device.
Working with master branch I was able to pair the device and saw all the data it was sending.
I checked out the dev branch as I wanted to contribute the device.
In the dev branch, it seems all the clusters for the device -- and probable the device (drayton radiator thermostat) or similar -- has been added so I could no longer see any "missing converter messages", but I didnt see any messages in the MQTT topics. So I had to figure the configuration. 

I thought I would go back to the master branch so I could get all the missing converters and then figure out what was needed in the configuration.
However, after going back to master and starting zigbee2mqtt again, I can not get the device to join the network. 

I have deleted the db every time I get a new version as I am just testing my setup.

I added debug for herdsman and I started seeing this messages:
```
zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":74,"manufacturerCode":4190,"commandIdentifier":10},"Payload":[{"attrId":57345,"dataType":33,"attrData":0}],"Cluster":{"ID":2821,"attributes":{"numberOfResets":{"ID":0,"type":33,"name":"numberOfResets"},"persistentMemoryWrites":{"ID":1,"type":33,"name":"persistentMemoryWrites"},"macRxBcast":{"ID":256,"type":35,"name":"macRxBcast"},"macTxBcast":{"ID":257,"type":35,"name":"macTxBcast"},"macRxUcast":{"ID":258,"type":35,"name":"macRxUcast"},"macTxUcast":{"ID":259,"type":35,"name":"macTxUcast"},"macTxUcastRetry":{"ID":260,"type":33,"name":"macTxUcastRetry"},"macTxUcastFail":{"ID":261,"type":33,"name":"macTxUcastFail"},"aPSRxBcast":{"ID":262,"type":33,"name":"aPSRxBcast"},"aPSTxBcast":{"ID":263,"type":33,"name":"aPSTxBcast"},"aPSRxUcast":{"ID":264,"type":33,"name":"aPSRxUcast"},"aPSTxUcastSuccess":{"ID":265,"type":33,"name":"aPSTxUcastSuccess"},"aPSTxUcastRetry":{"ID":266,"type":33,"name":"aPSTxUcastRetry"},"aPSTxUcastFail":{"ID":267,"type":33,"name":"aPSTxUcastFail"},"routeDiscInitiated":{"ID":268,"type":33,"name":"routeDiscInitiated"},"neighborAdded":{"ID":269,"type":33,"name":"neighborAdded"},"neighborRemoved":{"ID":270,"type":33,"name":"neighborRemoved"},"neighborStale":{"ID":271,"type":33,"name":"neighborStale"},"joinIndication":{"ID":272,"type":33,"name":"joinIndication"},"childMoved":{"ID":273,"type":33,"name":"childMoved"},"nwkFcFailure":{"ID":274,"type":33,"name":"nwkFcFailure"},"apsFcFailure":{"ID":275,"type":33,"name":"apsFcFailure"},"apsUnauthorizedKey":{"ID":276,"type":33,"name":"apsUnauthorizedKey"},"nwkDecryptFailures":{"ID":277,"type":33,"name":"nwkDecryptFailures"},"apsDecryptFailures":{"ID":278,"type":33,"name":"apsDecryptFailures"},"packetBufferAllocateFailures":{"ID":279,"type":33,"name":"packetBufferAllocateFailures"},"relayedUcast":{"ID":280,"type":33,"name":"relayedUcast"},"phyToMacQueueLimitReached":{"ID":281,"type":33,"name":"phyToMacQueueLimitReached"},"packetValidateDropCount":{"ID":282,"type":33,"name":"packetValidateDropCount"},"averageMacRetryPerApsMessageSent":{"ID":283,"type":33,"name":"averageMacRetryPerApsMessageSent"},"lastMessageLqi":{"ID":284,"type":32,"name":"lastMessageLqi"},"lastMessageRssi":{"ID":285,"type":40,"name":"lastMessageRssi"}},"name":"haDiagnostic","commands":{},"commandsResponse":{}}},"networkAddress":63898,"endpoint":1,"linkquality":84,"groupID":0}' +5ms
  zigbee-herdsman:controller:log 'zcl' data is from unknown device with network adress '63898', skipping... +3ms
```

## What did you expect to happen

Decive paired

## How to reproduce it (minimal and precise)

Hardware specific so hard to reproduce

arcanefoam, 2019-11-11:
It would seem the device remembers the network, but the herdsman does not remember the device (since I deleted the DB). Can changing the network ID work? As it will create a new network for the device to join?

Koenkk, 2019-11-12:
You can change the Zigbee channel and pair the device afterwards. Also make sure you are on zigbee2mqtt 1.7.0

arcanefoam, 2019-11-12:
Worked!

closed 2019-11-12 by arcanefoam