Koenkk/zigbee2mqtt #2320: availability docs are a bit vague about some things.

sjorge, 2019-11-11:
- no mention that devices are queried for values on DeviceAnnouncement
- `The availability_timeout defines the interval in seconds at which devices are pinged. Note that only non battery powered devices are checked for availability. availability_timeout: 0 disables the feature.` is a bit vague... is this the same as not setting `availability_timeout` ? Or does this mean that it will announce 'online' at zigbee2mqtt startup and 'offline' when zigbee2mqtt is stopped? (Does it still do the query on DeviceAccouncement?

Koenkk, 2019-11-12:
I've tried to improve the docs: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/information/availability.md.

If anything is still unclear please make a PR.

sjorge, 2019-11-12:
👍 so `0` is disable the feature

closed 2019-11-12 by sjorge