Koenkk/zigbee2mqtt #2321: Home Assistant: Last seen attribute for all types

sti0, 2019-11-11:
# Bug Report
## What happened
The new 1.7 feature "last seen"-attribute is not published for all types to Home Assistant.

## What did you expect to happen
Lights, switches and climate entities didn't get the last seen attribute while binary sensors and sensors got it.

## How to reproduce it (minimal and precise)
Set "last_seen" option in configuration.yaml

## Debug Info
zigbee2mqtt version: 1.7
CC253X firmware version: 20190425

Koenkk, 2019-11-12:
Currently last seen is only updated when messages are received. E.g. bulbs dont send messages so they wont have a last seen.

I guess we should also update the last seen on a successful command to a device.

sti0, 2019-11-12:
This would be great. 
You actually mean, that last seen is only published to HA if messages are received, right? In z2m I got a last seen attribute for my tradri bulbs 
`zigbee2mqtt:info  2019-11-12T18:46:49: MQTT publish: topic 'zigbee2mqtt/kuche_tresen', payload '{"state":"OFF","linkquality":86,"brightness":120,"color_temp":333,"color_mode":2,"color":{"x":0.4599,"y":0.4106},"last_seen":1573584409303}'`

I noticed that the climate entity won't get the last seen in HA but the battery sensor of the climate got it.

Koenkk, 2019-11-14:
If the battery sensor has it the climate sensor should also have it (they read from the same mqtt message). Can you post screenshots?

sti0, 2019-11-14:
Sure.
Climate Entity:
[image]
Battery Sensor Entity:

[image]

as I see now other attributes like `eurotronic_error_state`, `pi_heating_demand `also not published for the climate entity.

Koenkk, 2019-11-15:
If the battery sensors shows them, the climate one should also (because of https://www.home-assistant.io/integrations/climate.mqtt/#json_attributes_topic), 2 things could be going on:
- zigbee2mqtt doesn't publish a `json_attributes_topic` in the home assistant discovery payload (this is logged right after zigbee2mqtt startup)
- If zigbee2mqtt publishes a `json_attributes_topic`, the home assistant documentation is wrong (it doesn't support this) -> in this case create an issue in the home assistant issue tracker.

sti0, 2019-11-16:
On startup zigbee2mqtt publish the following discovery payload to HA:

```
{
    "min_temp": "5",
    "max_temp": "30",
    "modes": [
        "off",
        "auto",
        "heat"
    ],
    "mode_state_topic": "zigbee2mqtt/climate_buro",
    "mode_state_template": "{{ value_json.system_mode }}",
    "mode_command_topic": "zigbee2mqtt/climate_buro/set/system_mode",
    "current_temperature_topic": "zigbee2mqtt/climate_buro",
    "current_temperature_template": "{{ value_json.local_temperature }}",
    "temperature_state_topic": "zigbee2mqtt/climate_buro",
    "temperature_state_template": "{{ value_json.current_heating_setpoint }}",
    "temperature_command_topic": "zigbee2mqtt/climate_buro/set/current_heating_setpoint",
    "name": "climate_buro_climate",
    "unique_id": "0x00158d000192de9e_climate_zigbee2mqtt",
    "device": {
        "identifiers": [
            "zigbee2mqtt_0x00158d000192de9e"
        ],
        "name": "climate_buro",
        "sw_version": "Zigbee2mqtt 1.6.0",
        "model": "Spirit Zigbee wireless heater thermostat (SPZB0001)",
        "manufacturer": "Eurotronic"
    },
    "availability_topic": "zigbee2mqtt/climate_buro/availability"
}
```

`json_attributes_topic` is missing. 

But I have another question. You said

> E.g. bulbs dont send messages so they wont have a last seen.
> I guess we should also update the last seen on a successful command to a device.

But a Tradfri bulb updates last seen if it sends a linkquality update:
`zigbee2mqtt:info  2019-11-16T21:12:59: MQTT publish: topic 'zigbee2mqtt/kuche_spule', payload '{"state":"OFF","linkquality":52,"brightness":120,"color_temp":333,"color_mode":2,"color":{"x":0.4599,"y":0.4106},"last_seen":1573938779985}'`

...some minutes later....

`zigbee2mqtt:info  2019-11-16T21:15:08: MQTT publish: topic 'zigbee2mqtt/kuche_spule', payload '{"state":"OFF","linkquality":86,"brightness":120,"color_temp":333,"color_mode":2,"color":{"x":0.4599,"y":0.4106},"last_seen":1573938908514}'`

Another example pressing a Tradfri remote:

`zigbee2mqtt:info  2019-11-16T21:18:22: MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":39,"last_seen":1573939102377,"action":"toggle"}'`

... another press ...

`zigbee2mqtt:info  2019-11-16T21:18:24: MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":42,"last_seen":1573939104937,"action":"toggle"}'`

Last seen is updated in both cases. 
Therefore I think publishing `json_attributes_topic` for lights and switches makes sense, too?

Koenkk, 2019-11-18:
It seems that nowadays all MQTT integration of home assistant support the `json_attributes_topic`. I've updated zigbee2mqtt to discover this for all now. Your issue should be fixed in the latest dev.

sti0, 2019-11-18:
Thank you. Works for now `lights` and `switches` with last dev branch. `climate` entitys still won't get the attribute?!

```
{
    "min_temp": "5",
    "max_temp": "30",
    "modes": [
        "off",
        "auto",
        "heat"
    ],
    "mode_state_topic": "zigbee2mqtt/climate_buro",
    "mode_state_template": "{{ value_json.system_mode }}",
    "mode_command_topic": "zigbee2mqtt/climate_buro/set/system_mode",
    "current_temperature_topic": "zigbee2mqtt/climate_buro",
    "current_temperature_template": "{{ value_json.local_temperature }}",
    "temperature_state_topic": "zigbee2mqtt/climate_buro",
    "temperature_state_template": "{{ value_json.current_heating_setpoint }}",
    "temperature_command_topic": "zigbee2mqtt/climate_buro/set/current_heating_setpoint",
    "name": "climate_buro_climate",
    "unique_id": "0x00158d000192de9e_climate_zigbee2mqtt",
    "device": {
        "identifiers": [
            "zigbee2mqtt_0x00158d000192de9e"
        ],
        "name": "climate_buro",
        "sw_version": "Zigbee2mqtt 1.7.1+dev",
        "model": "Spirit Zigbee wireless heater thermostat (SPZB0001)",
        "manufacturer": "Eurotronic"
    },
    "availability_topic": "zigbee2mqtt/climate_buro/availability"
}
```

They should have received it before your change.
https://github.com/Koenkk/zigbee2mqtt/commit/661b2f1cc072717b4ae294cd30eefa429f4c2dff#diff-0c056233b68fd3653afb296f98947c8fL1066

Koenkk, 2019-11-19:
Found the issue, should be fixed now.

sti0, 2019-11-20:
Works as expected. Thank you.

closed 2019-11-20 by sti0