Koenkk/zigbee2mqtt #2323: Trying to open or close Keen Smart Vents fails with 'TypeError: Cannot read property 'convertSet' of undefined'

alexanv1, 2019-11-12:
# Bug Report
## What happened
After upgrading zigbee2mqtt hassio addon to 1.7 I can no longer change the state (open/close) of any of my Keen Smart Vents.
The log snippet is below:
debug 2019-11-12T09:24:40: Received MQTT message on 'zigbee2mqtt/Office Smart Vent/set' with data 'OPEN'
debug 2019-11-12T09:24:40: Publishing 'set' 'state' to 'Office Smart Vent'
error 2019-11-12T09:24:40: Publish 'set' 'state' to 'Office Smart Vent' failed: 'TypeError: Cannot read property 'convertSet' of undefined'

Tried the dev version of the addon as well with the same results.
## What did you expect to happen
No error
## How to reproduce it (minimal and precise)
## Debug Info
zigbee2mqtt version: version 1.7.0+dev (commit #a9211abd24680338c8f070532f0b83bd1a8b79f0)
CC253X firmware version: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190223}}'

Koenkk, 2019-11-12:
Do you by any chance have a custom zigbee-shepherd-converters mapped? (When added support for a new device) if so please remove this.

Koenkk, 2019-11-12:
What is your setup? bare-metal linux, docker or hassio?

alexanv1, 2019-11-12:
My setup is Hassio. I used a custom devices.js months ago but have removed it since. That file is not present in share\zigbee2mqtt. And I haven't added custom converters via any other mechanism.

mccoy88f, 2019-11-12:
> What is your setup? bare-metal linux, docker or hassio?

Hassio

Koenkk, 2019-11-12:
Seems it was a bug, sorry!

Should be fixed in latest dev, for hassio use `-edge` (will be available in +- 2 hours from now).

mccoy88f, 2019-11-12:
Hi... i install and reinstall NOW zigbee2mqtt-edge and now have zigbee2mqtt version 1.7.1+dev (commit #ba18c5f773c160b54977908ed241cd40547a2fff)

but no solution... this is the log (2 different devices, lupus and blaukpunt)
zigbee2mqtt:error 2019-11-12T23:32:59: Publish 'set' 'state' to '0x00124b00144b48ab' failed: 'Error: Cluster 'genLevelCtrl' has no command 'currentLevel''
zigbee2mqtt:info  2019-11-12T23:32:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x00124b00144b48ab' failed: 'Error: Cluster 'genLevelCtrl' has no command 'currentLevel''","meta":{"friendly_name":"0x00124b00144b48ab"}}'
zigbee2mqtt:error 2019-11-12T23:33:00: Publish 'set' 'state' to '0x00124b0008b7d45d' failed: 'Error: Cluster 'genLevelCtrl' has no command 'currentLevel''
zigbee2mqtt:info  2019-11-12T23:33:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x00124b0008b7d45d' failed: 'Error: Cluster 'genLevelCtrl' has no command 'currentLevel''","meta":{"friendly_name":"0x00124b0008b7d45d"}}'

alexanv1, 2019-11-14:
Same error for me unfortunately (using zigbee2mqtt-edge):

error 2019-11-14T11:54:48: Publish 'set' 'state' to 'Office Smart Vent' failed: 'Error: Cluster 'genLevelCtrl' has no command 'currentLevel''
info  2019-11-14T11:54:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Office Smart Vent' failed: 'Error: Cluster 'genLevelCtrl' has no command 'currentLevel''","meta":{"friendly_name":"Office Smart Vent"}}'

Koenkk, 2019-11-14:
Should also be fixed in latest dev, please try again over 2 hours with latest hassio edge.

mccoy88f, 2019-11-14:
Perfect... Seems to works now! Thanks

Koenkk, 2019-11-15:
great, thanks!

closed 2019-11-15 by Koenkk