Koenkk/zigbee2mqtt #2324: Xiaomi Aqara Motion Sensor RTCGQ11LM not respecting "occupancy_timeout" settings

rsalunga29, 2019-11-12:
# Bug Report
The mentioned device is not respecting `occupancy_timeout` setting for `device_options` which can be found in the configuration file, in my case `devices.yaml`. I've set my `occupancy_timeout` to `60` seconds, however it seems that it still takes `90` seconds (which is the default) for timeout.

I believe this may have cause the issue but I'm not really sure. https://github.com/Koenkk/zigbee-herdsman-converters/blob/81c995224c683112551202c9d8a9215f8874a04c/converters/fromZigbee.js#L12

## What did you expect to happen
Timeout should be `60` seconds and not `90` seconds

## Debug Info
zigbee2mqtt version: 1.6.0
CC253X firmware version: CC2531_DEFAULT_20190608

Koenkk, 2019-11-12:
Please provide debug logs and your `configuration.yaml`.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

rsalunga29, 2019-11-13:
@Koenkk sure! Here's my `configuration.yaml`
`
homeassistant: false
permit_join: true
advanced:
  log_level: debug
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
queue:
  delay: 250
serial:
  port: /dev/ttyACM0
devices: devices.yaml
`
and here's my `devices.yaml`
`
'0x00158d00036bbdf7':
  friendly_name: Xiaomi Aqara Motion Sensor
  retain: false
  options:
    no_occupancy_since: [10, 60]
    occupancy_timeout: 60
`
and here's my debug logs (from my index.js)
`
topic - time: zigbee2mqtt/Xiaomi Aqara Motion Sensor 12:49:22 PM
message: {"illuminance":268,"linkquality":92,"occupancy":true,"battery":100,"voltage":3055}
topic - time: zigbee2mqtt/Xiaomi Aqara Motion Sensor 12:50:14 PM
message: {"illuminance":950,"linkquality":99,"occupancy":true,"battery":100,"voltage":3055}
topic - time: zigbee2mqtt/Xiaomi Aqara Motion Sensor 12:51:44 PM
message: {"illuminance":950,"linkquality":99,"occupancy":false,"battery":100,"voltage":3055}

### none `after occupancy: false`
`

Koenkk, 2019-11-13:
The configuration for the device should be

```yaml
'0x00158d00036bbdf7':
  friendly_name: Xiaomi Aqara Motion Sensor
  retain: false
  no_occupancy_since: [10, 60]
  occupancy_timeout: 60
```

rsalunga29, 2019-11-14:
Okay! I'll try that now

rsalunga29, 2019-11-14:
@Koenkk it still takes 90 seconds before an `occupancy: false` is returned

rsalunga29, 2019-11-14:
@Koenkk I just found out that `devices.yaml` automatically formats my device configuration to
```yaml
options:
    no_occupancy_since:
      - 10
      - 60
    occupancy_timeout: 60
```

Koenkk, 2019-11-14:
Can you share your complete config now?

pitoganzado, 2019-11-14:
I'm getting same problem with [IKEA motion detection device ](https://www.zigbee2mqtt.io/devices/E1525.html)

sti0, 2019-11-16:
@pitoganzado  this is made by design. Ikea motion sensor reports after 3 minutes.

@rsalunga29 the Aqara motion sensor is designed to report after 90 sec. You need to "hardware hack" the Aqara motion sensor that it reports faster. Did you do this as described here for example: https://livebywant.tistory.com/13?category=703455

Koenkk, 2019-11-16:
@sti0 an occupancy timeout of 60 should be valid, occupancy false is generated by zigbee2mqtt itself so it should work, But I still see `options` in his configuration which is not valid.

sti0, 2019-11-16:
@Koenkk ok. but the device itself would report another occupancy true after 3 minutes (ikea) oder 90 sec (aqara). right?

Koenkk, 2019-11-17:
For aqara it is 60 seconds but yes that is indeed the case.

rsalunga29, 2019-11-18:
@Koenkk here's my full device.yaml now, nothing has changed so far except for additional devices:
```yaml
'0x00158d000396a7f7':
  friendly_name: Xiaomi Aqara Smart LED Bulb Right
'0x00158d00036bbdf7':
  friendly_name: Xiaomi Aqara Motion Sensor
  options:
    no_occupancy_since:
      - 10
      - 60
    occupancy_timeout: 60
'0x00158d000325f8a0':
  friendly_name: Xiaomi Mijia Door/Window Sensor
'0x00158d0002b8057c':
  friendly_name: Xiaomi Aqara Smart LED Bulb Left
'0x00158d0002c65c0f':
  friendly_name: Xiaomi Aqara Smart LED Bulb Center
```

Koenkk, 2019-11-18:
So it should be

```
'0x00158d000396a7f7':
  friendly_name: Xiaomi Aqara Smart LED Bulb Right
'0x00158d00036bbdf7':
  friendly_name: Xiaomi Aqara Motion Sensor
  no_occupancy_since:
    - 10
    - 60
  occupancy_timeout: 60
'0x00158d000325f8a0':
  friendly_name: Xiaomi Mijia Door/Window Sensor
'0x00158d0002b8057c':
  friendly_name: Xiaomi Aqara Smart LED Bulb Left
'0x00158d0002c65c0f':
  friendly_name: Xiaomi Aqara Smart LED Bulb Center
```

rsalunga29, 2019-11-18:
@Koenkk Thanks! It's working now

closed 2019-11-18 by rsalunga29

romainpaul, 2021-02-10:
Hello, I have the same issue. I had occupancy_timeout: 60, but any change, it remains 90 s :

Add-on => Zigbee2mqtt => Configuration :

```
data_path: /share/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://-----'
  user:-----
  password: -----
serial:
  port: >-
    /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001CD4B234-if00
```
...

/config/devices.yaml :

'```
0x00158d00066954a1':
  friendly_name: Xiaomi Aqara Motion Sensor
  no_occupancy_since :
   - 10
   - 60
  occupancy_timeout: 60

```
What I see in Zigbee2mqtt settings (only "friendly_name": "Xiaomi Aqara Motion Sensor")

"devices": {
            "0x00124b002267ebe0": {
                "friendly_name": "0x00124b002267ebe0"
            },
            "0x00124b00226a9778": {
                "friendly_name": "0x00124b00226a9778"
            },
            "0x00124b00226aa462": {
                "friendly_name": "0x00124b00226aa462"
            },
            "0x00124b00226aa65e": {
                "friendly_name": "0x00124b00226aa65e"
            },
            "0x00158d00066954a1": {
                "friendly_name": "Xiaomi Aqara Motion Sensor"
            },
            "0x00158d00067be206": {
                "friendly_name": "Xiaomi Mijia Door/Window Sensor"
            },
            "0x04cf8cdf3c7cb556": {
                "friendly_name": "0x04cf8cdf3c7cb556"
            }

What is missing ?

Thx