Koenkk/zigbee2mqtt #2325: IKEA E1524/E1810 remote control / new behaviour after firmware update

didiht, 2019-11-12:
The remote control has got a new firmware update (version 2.3.014) which includes Zigbee 3.0 in their description.

After pairing I saw the Ikea remote sending **groupID 0**
`debug 2019-11-12T13:43:37: Received Zigbee message from '0x000d6ffffe052461', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20190708","swBuildId":"2.3.014"}' from endpoint 1 with groupID 0`

Then I tried to bind the remote to a new group `{"friendly_name": "g_Fernbedienung-3", "id": 93}` **with success!**
`debug 2019-11-12T13:56:31: Received Zigbee message from 'Fernbedienung-3', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 93`

So in my eyes the IKEA remote (firmware >= 2.3.014) can be handled in the same way as the Ikea switch 1743.
**This should be mentioned** in the documentation https://www.zigbee2mqtt.io/devices/E1524_E1810.html

Another point:
I saw the **no converter available** message in the log, maybe an issue?
`debug 2019-11-12T13:43:34: No converter available for 'E1524/E1810' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"TRADFRI remote control","manufacturerName":"IKEA of Sweden","powerSource":3}'`

Koenkk, 2019-11-12:
- No converter available isn't an issue, it doesn't contain any relevant information so this can be ignored.
- Great! Please provide a pr by editing the following file: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/E1524_E1810.md

didiht, 2019-11-13:
I created pr https://github.com/Koenkk/zigbee2mqtt.io/pull/157

Koenkk, 2019-11-14:
Merged, thanks!

closed 2019-11-14 by Koenkk