Koenkk/zigbee2mqtt #2326: fail: Zigbee2MqttAssistant.Services.MqttConnectionService[0]

Freubelaar, 2019-11-12:
Hi Guys,

Need some help, I got a fail message in my log. see info below, I hope its complete

**zigbee2mqtt version 1.7
HA version 101.3**

zigbee2mqttassistant config:
**{
  "settings": {
    "base_topic": "zigbee2mqtt",
    "homeassistant": true,
    "mqttserver": "core-mosquitto",
    "MqttPort": 1883,
    "MqttSecure": false,
    "mqttusername": " ",
    "mqttpassword": " ",
    "LowBatteryThreshold": 25
  }
}**

**Log zigbee2mqttassistant:**
Starting Zigbee2MqttAssistant v0.3.88+Branch.master.Sha.a34fe059880e14ed9862286d6aa30d3853b87009...
Hosting environment: Production
Content root path: /app
Now listening on: http://[::]:80
Application started. Press Ctrl+C to shut down.
fail: Zigbee2MqttAssistant.Services.MqttConnectionService[0]
      Error processing MQTT message on topic 'zigbee2mqtt/bridge/config/devices'
System.ArgumentOutOfRangeException: Index and length must refer to a location within the string.
Parameter name: length
   at System.String.Substring(Int32 startIndex, Int32 length)
   at Zigbee2MqttAssistant.Services.BridgeStateService.UpdateDevices(String payload) in d:\a\1\s\Zigbee2MqttAssistant\Services\BridgeStateService.cs:line 322
   at Zigbee2MqttAssistant.Services.MqttConnectionService.DispatchDevicesMessage(MqttApplicationMessage msg) in d:\a\1\s\Zigbee2MqttAssistant\Services\MqttConnectionService.cs:line 459
   at Zigbee2MqttAssistant.Services.MqttConnectionService.HandleApplicationMessageReceivedAsync(MqttApplicationMessageReceivedEventArgs eventArgs) in d:\a\1\s\Zigbee2MqttAssistant\Services\MqttConnectionService.cs:line 315
fail: Zigbee2MqttAssistant.Services.MqttConnectionService[0]
      Error processing MQTT message on topic 'zigbee2mqtt/bridge/config/devices'
System.ArgumentOutOfRangeException: Index and length must refer to a location within the string.
Parameter name: length
   at System.String.Substring(Int32 startIndex, Int32 length)
   at Zigbee2MqttAssistant.Services.BridgeStateService.UpdateDevices(String payload) in d:\a\1\s\Zigbee2MqttAssistant\Services\BridgeStateService.cs:line 322
   at Zigbee2MqttAssistant.Services.MqttConnectionService.DispatchDevicesMessage(MqttApplicationMessage msg) in d:\a\1\s\Zigbee2MqttAssistant\Services\MqttConnectionService.cs:line 459
   at Zigbee2MqttAssistant.Services.MqttConnectionService.HandleApplicationMessageReceivedAsync(MqttApplicationMessageReceivedEventArgs eventArgs) in d:\a\1\s\Zigbee2MqttAssistant\Services\MqttConnectionService.cs:line 315

Koenkk, 2019-11-12:
Looks like a zigbee2mqtt assistant issue, please create an issue at their issue tracker.

closed 2019-11-12 by Koenkk

Freubelaar, 2019-11-12:
Yep, sorry