Koenkk/zigbee2mqtt #2327: Cant remove Xiaomi motion sensor

mpuff, 2019-11-12:
Hello,

i have a Xiaomi Motion Sensor that is not reachable an i get no status from this.
When i try to remove the device i get these error message:

Nov 12 17:37:22 zigbee2mqttpi01 npm[8324]: zigbee2mqtt:error 2019-11-12T16:37:22: Failed to remove Xiaomi_motion_sensor_FF_Bath (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)

i am on the latest 1.7.0 version.

mpuff, 2019-11-12:
it is in the documentation, sry

closed 2019-11-12 by mpuff

Sonsi79, 2019-12-02:
Hey, what's the rusult for this problem.
Have the same problem by removing an Xiaomi temp sensor

mpuff, 2019-12-02:
Hello,

https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html

read the zigbee2mqtt/bridge/config/remove part.