Koenkk/zigbee2mqtt #2329: Question regarding newly listed supported device

uncle-fed, 2019-11-12:
Apologies for opening issue for a simple question but I did not find any other way to communicate directly.

A few days ago this device was added to a supported list. 

TS0201 - Aqara temperature & humidity sensor with display.

https://www.zigbee2mqtt.io/devices/TS0201.html

I would really like to use something like this but googling for it produces zero results.

Where is the information about the device coming from? Any links would be appreciated!

Koenkk, 2019-11-12:
From https://github.com/Koenkk/zigbee-herdsman-converters/pull/716#issuecomment-546676426, but it seems it isn't available under that link anymore.

uncle-fed, 2019-11-12:
Thanks for the information!
I guess we'll just have to be a bit more patient.
Must be brand new device that hasn't reached market fully yet, if it it's not even listed on the usual Chinese web sites...

uncle-fed, 2019-11-12:
Oh I see... Must be confusion coming from the fact that this is not actually a device made by Xiaomi for Aquara / Mi Home. This seems to be just some generic Chinese brand selling these...

https://www.aliexpress.com/item/33040664471.html?spm=a2g0o.detail.1000014.7.421e590ep2IJDT&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.13338.132444.000000000000000&scm_id=1007.13338.132444.000000000000000&scm-url=1007.13338.132444.000000000000000&pvid=7bab299f-7d68-47bc-a461-cee8dd7dbbdc

kirovilya, 2019-11-12:
it is Tuya!
https://github.com/pipiche38/Domoticz-Zigate/issues/481
https://fr.aliexpress.com/item/4000277514396.html

uncle-fed, 2019-11-12:
Honestly, looks like it neither Tuya nor Aqara. I found some more places now that sell it. Chinese sellers just slap any logo/brand of the day on top, it seems. The main point is that it's Zigbee compatible and works with zigbee2mqtt :-)). Ordered mine just now!  ;-) Will update if/once I get it and disassemble it to have a look inside.

closed 2019-11-12 by uncle-fed