Koenkk/zigbee2mqtt #2330: Better range ?

Jeoffreybauvin, 2019-11-12:
Hi there,

I'm using the CC2531 for my Zigbee network. But, I have some range  issues...

Is there any way to improve the range of the network ? I have a temperature probe in my garden shed and it's out of range :(.

Thanks,

copystring, 2019-11-12:
https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html

Jeoffreybauvin, 2019-11-12:
Wow, I feel bad, I didn't check the documentation first.

Thanks @copystring , I'll check this out.

copystring, 2019-11-12:
You're welcome.

Jeoffreybauvin, 2019-11-12:
One more question @copystring : what range can be expected with CC2531 ? I just tried with an USB extension cable, but, no luck... 

Wifi channel (11 by default) seems to be "free" also.

james-fry, 2019-11-12:
Jfyi WiFi channels are not the same as zigbee

james-fry, 2019-11-12:
[image]

Jeoffreybauvin, 2019-11-13:
So I don't understand why my signal is so poor.  I even tried to shutdown the wifi, my tradfri gateway, and nothing changes...

james-fry, 2019-11-13:
In general the range of the cc2532 is quite poor.
The best way to increase range is routers - in my network this role is provided by light bulbs and power socket switches, but you can also add dedicated routers (ikea make one, or you can flash router firmware onto a cc2531).
You might get a bit more range with an antenna or with one of the dongles with an antenna such as [@h4nc sells.](https://zigbee2mqtt.discourse.group/t/ad-buy-ready2use-zigbee2mqtt-stick-flashed-antenna-mod-and-printed-case/22)
Or consider one of the adapters with more power - either with RF frontend or the more advanced devices that just got supported in 1.7:
https://www.zigbee2mqtt.io/information/supported_adapters.html

vampik, 2019-11-13:
External antenna drastically increases link quality and network range. After changing CC2531 + PCB antenna to CC2531 + external antenna I got link quality increase from 0~10 to 40~60. My sensors are 5-7 meters and 1 wall away from the coordinator, so CC2531 + PCB antenna range is extremely poor.
CC2531 with external antenna can be found on Aliexpress for $7-8.

james-fry, 2019-11-13:
Aliexpress is one option if budget is a big concern, but there are a lot of positive feedback for the hand modified devices h4nc is selling.

Jeoffreybauvin, 2019-11-13:
Great, I sent an email to h4nc :).

skandragon, 2019-11-25:
I have hand-modified a USB dongle after the PCB antenna seemed poor.  The signal quality increased but not as much as I expected using a well-tuned antenna.

I now have several CC2530 devices with external antennas, and once again, so far...  poor signal quality from the coordinator to all devices.  My routers all have excellent quality to the end devices connected to them, but anything -- including a router 5 feet from the coordinator -- indicate very poor link quality.

I'm not sure why...  I've tried three different devices, two with PCB antennas, and two with external.

Jeoffreybauvin, 2019-12-10:
I have now a stick from @h4nc :). But I still have some range issues :/... 

For example, I'm using a Xiaomi temperature sensor (WSDCGQ11LM), and when I'm moving it 1 meter away from the stick, the link quality fall from 97 to 55... 

Anothe example : my study (with my raspberry and my stick) is right above my garage. If I put my xiaomi sensor in the garage, the link quality fall to 0...   I still have some value but,  I'm wondering about the link quality to 0... Maybe I don't understand the link quality value in Zigbee2mqtt's logs...

h4nc, 2019-12-10:
I also also have a sensor which did not work before the mod and works fine now but still shows a low Linkquality. Besides that one the Linkquality of most of my sensors went up.

To be honest I also did not understand the value of Linkquality yet, because for me it makes absolutely no difference in terms of reliability if a sensor shows 100 or 0 Linkquality. For me my network started to work a 100% reliable after the mod.

Things you could also have a look at:
- channel (mind that WiFi and Zigbee channels are not the same, so wifi 11 is not equal to Zigbee 11
- try different positions of the antenna, maybe also try an usb cable between the stick and your pi
- if nothing helps you might need to add routers. You could try to flash your old CC2531 with a router firmware.

Jeoffreybauvin, 2019-12-16:
In my case, quality to 0 means no result :D. 

[image]

My sensor is not sending any information since I moved it to the garage :(. 

I think I'll move the antenna on the ground floor...   I'll see if it's better.

h4nc, 2019-12-16:
> I think I'll move the antenna on the ground floor... I'll see if it's better.

Probably a good idea to place it somewhere else.

Jeoffreybauvin, 2019-12-22:
I just placed it on the ground floor, and it's seems better.

But, now, I think I have an issue with Zigbee2mqtt. 

When I'm starting zigbee2mqtt, I do have some values. But they don't seem to be refreshed after.

In Zigbee2mqtt's logs, I have this :

```
déc. 22 14:26:42 domotic npm[5894]: zigbee2mqtt:info  2019-12-22T13:26:42: MQTT publish: topic 'zigbee2mqtt/doorsensor_abri_jardin', payload '{"contact":true,"linkquality":0,"battery":100,"voltage":3015}'
déc. 22 14:26:42 domotic npm[5894]: zigbee2mqtt:info  2019-12-22T13:26:42: MQTT publish: topic 'zigbee2mqtt/doorsensor_abri_jardin', payload '{"contact":true,"linkquality":0,"battery":100,"voltage":3015}'
déc. 22 14:26:42 domotic npm[5894]: zigbee2mqtt:info  2019-12-22T13:26:42: MQTT publish: topic 'zigbee2mqtt/doorsensor_abri_jardin', payload '{"contact":true,"linkquality":0,"battery":100,"voltage":3015}'
déc. 22 14:26:43 domotic npm[5894]: zigbee2mqtt:info  2019-12-22T13:26:43: MQTT publish: topic 'zigbee2mqtt/doorsensor_abri_jardin', payload '{"contact":true,"linkquality":0,"battery":100,"voltage":3015}'
déc. 22 14:26:43 domotic npm[5894]: zigbee2mqtt:info  2019-12-22T13:26:43: MQTT publish: topic 'zigbee2mqtt/doorsensor_abri_jardin', payload '{"contact":true,"linkquality":15,"battery":100,"voltage":3015}'
déc. 22 14:27:14 domotic npm[5894]: zigbee2mqtt:info  2019-12-22T13:27:14: Starting network scan (includeRoutes 'false')
déc. 22 14:27:14 domotic npm[5894]: zigbee2mqtt:info  2019-12-22T13:27:14: Network scan finished
déc. 22 14:27:14 domotic npm[5894]: zigbee2mqtt:info  2019-12-22T13:27:14: MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/graphviz', payload 'digraph G {
déc. 22 14:27:14 domotic npm[5894]: node[shape=record];
déc. 22 14:27:14 domotic npm[5894]:   "0x00124b001936a41f" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{Coordinator|0x00124b001936a41f (0)|2019-12-22T14:27:14+01:00}"];
déc. 22 14:27:14 domotic npm[5894]:   "0x00158d000322bba4" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{tempsensor_abri_jardin|0x00158d000322bba4 (28507)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|unknown}"];
déc. 22 14:27:14 domotic npm[5894]:   "0x00158d0003130670" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{doorsensor_porte_entree|0x00158d0003130670 (47598)|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|unknown}"];
déc. 22 14:27:14 domotic npm[5894]:   "0x00158d0003219520" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{doorsensor_abri_jardin|0x00158d0003219520 (55075)|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|2019-12-22T14:26:43+01:00}"];
déc. 22 14:27:14 domotic npm[5894]:   "0x00158d00036b594f" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{tempsensor_garage|0x00158d00036b594f (43430)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|unknown}"];
déc. 22 14:27:14 domotic npm[5894]:   "0x00158d00036b594f" -> "0x00124b001936a41f" [penwidth=1, weight=0, color="#994444", label="170"]
déc. 22 14:27:14 domotic npm[5894]: }'
déc. 22 14:30:00 domotic npm[5894]: zigbee2mqtt:info  2019-12-22T13:30:00: Starting network scan (includeRoutes 'false')
déc. 22 14:30:00 domotic npm[5894]: zigbee2mqtt:info  2019-12-22T13:30:00: Network scan finished
déc. 22 14:30:00 domotic npm[5894]: zigbee2mqtt:info  2019-12-22T13:30:00: MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/graphviz', payload 'digraph G {
déc. 22 14:30:00 domotic npm[5894]: node[shape=record];
déc. 22 14:30:00 domotic npm[5894]:   "0x00124b001936a41f" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{Coordinator|0x00124b001936a41f (0)|2019-12-22T14:30:00+01:00}"];
déc. 22 14:30:00 domotic npm[5894]:   "0x00158d000322bba4" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{tempsensor_abri_jardin|0x00158d000322bba4 (28507)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|unknown}"];
déc. 22 14:30:00 domotic npm[5894]:   "0x00158d0003130670" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{doorsensor_porte_entree|0x00158d0003130670 (47598)|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|unknown}"];
déc. 22 14:30:00 domotic npm[5894]:   "0x00158d0003219520" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{doorsensor_abri_jardin|0x00158d0003219520 (55075)|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|2019-12-22T14:26:43+01:00}"];
déc. 22 14:30:00 domotic npm[5894]:   "0x00158d00036b594f" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{tempsensor_garage|0x00158d00036b594f (43430)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|unknown}"];
déc. 22 14:30:00 domotic npm[5894]:   "0x00158d00036b594f" -> "0x00124b001936a41f" [penwidth=1, weight=0, color="#994444", label="170"]
déc. 22 14:30:00 domotic npm[5894]: }'
déc. 22 14:40:00 domotic npm[5894]: zigbee2mqtt:info  2019-12-22T13:40:00: Starting network scan (includeRoutes 'false')
déc. 22 14:40:00 domotic npm[5894]: zigbee2mqtt:info  2019-12-22T13:40:00: Network scan finished
déc. 22 14:40:00 domotic npm[5894]: zigbee2mqtt:info  2019-12-22T13:40:00: MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/graphviz', payload 'digraph G {
déc. 22 14:40:00 domotic npm[5894]: node[shape=record];
[... cut ...] ;
déc. 22 14:50:00 domotic npm[5894]:   "0x00158d00036b594f" -> "0x00124b001936a41f" [penwidth=1, weight=0, color="#994444", label="170"]
déc. 22 14:50:00 domotic npm[5894]: }'
```

I see only updates from my Graphviz network... 

In home-assistant, the status is "unknown" :

[image]

Last seen : 23 minutes ago :( :

[image]

And now, if I restart Zigbee2mqtt : 

[image]

Devices are now refreshed, and link quality is very good.

[image]

I'm probably missing something here, but can't find it...

h4nc, 2019-12-22:
Devices are now refreshed, and link quality is very good.

But the picture says Linkquality 0.
I don’t know the issue that you need to restart z2M to get the values again.
Maybe you should try to re-pair a sensor to see if that testsensor works stable after pairing again.

Jeoffreybauvin, 2020-01-02:
@h4nc  Yes, I waited a few days to confirm this, but indeed, devices are now refreshed... until one stops being refreshed... 

One temp sensor stopped refreshing itself 2 days ago. I waited long enough to see if it came back to life, but it didn't.  I haven't moved it... 

So today I restarted zigbee2mqtt .. and my device showed up again...  Strange.

Is there anything I can do ? Maybe tweak some configuration in zigbee2mqtt to "force" refresh ?

h4nc, 2020-01-02:
That's very strange. If it comes back after a restart, it looks like it is indeed a software issue (but I honestly don't know).
You could also try if moving the sensor nearer to your coordinator would help. If that's the case a router might help to retain the connection.

closed 2020-03-10 by stale[bot]