Koenkk/zigbee2mqtt #2331: Use deviceId AND friendly name

jpg0, 2019-11-12:
I would like to:
- Reference devices by their device ID (mainly because it's immutable), and have changes reported by device ID (in MQTT). I believe this is the best device key for machine interpretation.
- See devices in UIs by their 'friendly name'. When I say UI, I include logs, network map, etc - anywhere that is designed for a human to be reading, rather than a machine.

Right now, it seems that I must choose one or the other - if I don't set friendly names, then looking at logs and network maps are really hard because I need to x-reference them with a list of device IDs to names. If I do set them, then I need to maintain mappings in two places - both zigbee2mqtt, as well as they system which interprets the MQTT messages (Openhab in my case). I do relocate (and hence rename) devices, which would mean keeping two lists in sync.

So my request I believe reduces to:
- MQTT state messages emitted by zigbee2mqtt should always contain the device ID.
- It should be possible to send commands to devices using the device ID (even if a friendly name has been set).

Please let me know if this is already the case! The docs suggest that once you use friendly names, you must also uses these over MQTT.

sjorge, 2019-11-12:
That would probably break a whole lot of peoples current flows though, if you switch to always using the device ID for MQTT topics.

I'm not 100% sure but I think you can already send message to either the friendly_name or device_id when doing a /set but it's been ages since I tried and my memory is not the best.

jpg0, 2019-11-12:
I completely agree that it's a bad idea to break existing APIs; it would be a better approach to maintain compatibility yet allow using either key. 

BUT, I just looked at the code and you're right, setting values accepts the id OR the friendly name (and TBH checks the device ID first, which is also indexed), so this part would work for me. (My use case is that I want friendly names in the logs & networkmap - e.g. the human interfaces, but device IDs over MQTT - the machine interface.)

So writing state to devices is all good because it accepts both. However, reading device state will NOT work because it does not publish via the ID, only the friendly name. (So in my case Openhab, which is working via the ID, will not be able to map the messages.)

Koenkk, 2019-11-13:
- Setting by `device_id` is indeed already supported (see it as an "undocumented feature")
- The other way around is not. Perhaps you could create an intermediate python script, which receives the message based on the friendly_name and broadcasts them again with the device ID, I think you can retrieve all required information via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigdevicesget

sjorge, 2019-11-13:
It might be easier to not configure friendly_name's in zigbee2mqtt, and then have a small script that replaces the device_id in the logs with the friendly_name, maybe that can be queried from OpenHAB.

jpg0, 2019-11-13:
I agree that both of those approaches would work, although neither are ideal. Would you be open to a setting to choose whether `name` or `ieeeAddress` is sent for device status? That seems a very simple option and I would be happy do it myself and submit a PR?

Koenkk, 2019-11-14:
@jpg0 if we are implementing something like that I feel it should be more generic. Something I'm thinking of
```yaml
'0x1234':
  friendly_name: my_name
  topic: 'zigbee2mqtt/my_different_topic
```

When `topic` is set, it will use this instead of the default `zigbee2mqtt/friendly_name`.

Some time ago there was already some discussion about this: https://github.com/Koenkk/zigbee2mqtt/issues/179

sjorge, 2019-11-14:
Maybe with some simple variable substitution so you can use things like
- ${prefix} (the mqtt prefix, usually zigbee2mqtt but configurable)
- ${friendly_name}
- ${device_id}

And maybe some other ones too:
- ${device_model}
- ${device_manufacturare}

jpg0, 2019-11-15:
Sounds reasonable. I'd be happy to add that, I just need to get onto 3.x first. Where is best for me to look to check that I haven't got things that are known not to work on 3.x?

carldebilly, 2019-11-15:
@Koenkk If you add a _custom topic per device_, please add this information to the `bridge/config/devices/get` response so monitoring tools like Z2MA would be able to find their topics.

Koenkk, 2019-11-15:
> Sounds reasonable. I'd be happy to add that, I just need to get onto 3.x first. Where is best for me to look to check that I haven't got things that are known not to work on 3.x?

What do you mean with 3.x?

sjorge, 2019-11-15:
I’m guessing they are talking about the new 3.x firmware

jpg0, 2019-11-21:
Given that I can't do this right now, I'll close this issue for now and hopefully submit a PR at some point.

closed 2019-11-21 by jpg0