Koenkk/zigbee2mqtt #2332: PM-C140-ZB - No power attribute

rootsnet, 2019-11-13:
# Bug Report

## What happened
I moved my PM-C140-ZB devices from smartthings to zigbee2mqtt. the zigbee2mqtt is working with mosquitto 1.6.7 and home assistant 0.101.3.
But I was not able to get the power information on the home assistant. 
So I checked the log of the zigbee2mqtt, and I found below.

$ grep computer_room_wall_outlet_with_meter log.txt 
info  2019-11-12T03:10:35: computer_room_wall_outlet_with_meter (0x****************): PM-C140-ZB - Dawon DNS IOT remote control smart buried-type outlet (Router)
info  2019-11-12T03:10:36: MQTT publish: topic 'zigbee2mqtt/computer_room_wall_outlet_with_meter', payload '{"state":"ON","linkquality":76}'
info  2019-11-12T03:10:36: MQTT publish: topic 'homeassistant/sensor/0x****************/power/config', payload '{"unit_of_measurement":"W","icon":"mdi:factory","value_template":"{{ value_json.power }}","state_topic":"zigbee2mqtt/computer_room_wall_outlet_with_meter","json_attributes_topic":"zigbee2mqtt/computer_room_wall_outlet_with_meter","name":"computer_room_wall_outlet_with_meter_power","unique_id":"0x****************_power_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x****************"],"name":"computer_room_wall_outlet_with_meter","sw_version":"Zigbee2mqtt 1.7.0","model":"IOT remote control smart buried-type outlet (PM-C140-ZB)","manufacturer":"Dawon DNS"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2019-11-12T03:10:36: MQTT publish: topic 'homeassistant/switch/0x****************/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/computer_room_wall_outlet_with_meter/set","state_topic":"zigbee2mqtt/computer_room_wall_outlet_with_meter","name":"computer_room_wall_outlet_with_meter_switch","unique_id":"0x****************_switch_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x****************"],"name":"computer_room_wall_outlet_with_meter","sw_version":"Zigbee2mqtt 1.7.0","model":"IOT remote control smart buried-type outlet (PM-C140-ZB)","manufacturer":"Dawon DNS"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2019-11-12T03:10:36: MQTT publish: topic 'homeassistant/sensor/0x****************/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqttcomputer_room_wall_outlet_with_meter","json_attributes_topic":"zigbee2mqtt/computer_room_wall_outlet_with_meter","name":"computer_room_wall_outlet_with_meter_linkquality","unique_id":"0x****************_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x****************"],"name":"computer_room_wall_outlet_with_meter","sw_version":"Zigbee2mqtt 1.7.0","model":"IOT remote control smart buried-type outlet (PM-C140-ZB)","manufacturer":"Dawon DNS"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2019-11-12T03:11:21: MQTT publish: topic 'zigbee2mqtt/computer_room_wall_outlet_with_meter', payload '{"state":"ON","linkquality":76}'
...

There is no 'power' attribute.

## What did you expect to happen
The power information of PM-C140-ZB

## How to reproduce it (minimal and precise)
It is happening now.

## Debug Info
zigbee2mqtt version: 1.7
CC2531 firmware version: CC2531_DEFAULT_20190608

Koenkk, 2019-11-14:
@stkang does power work for you?

rootsnet, 2019-11-14:
I downgraded the zigbee2mqtt to 1.6. and I was able to read the 'power' attribute from the log as below. 

# grep computer_room_wall_outlet_with_meter log.txt
11/14/2019, 10:07:28 PM - debug: Configuring computer_room_wall_outlet_with_meter (0x****************) ...
11/14/2019, 10:07:28 PM - info: MQTT publish: topic 'homeassistant/sensor/0x****************/power/config', payload '{"unit_of_measurement":"W","icon":"mdi:factory","value_template":"{{ value_json.power }}","state_topic":"zigbee2mqtt/computer_room_wall_outlet_with_meter","json_attributes_topic":"zigbee2mqtt/computer_room_wall_outlet_with_meter","name":"computer_room_wall_outlet_with_meter_power","unique_id":"0x****************_power_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x****************"],"name":"computer_room_wall_outlet_with_meter","sw_version":"Zigbee2mqtt 1.6.0","model":"IOT remote control smart buried-type outlet (PM-C140-ZB)","manufacturer":"Dawon DNS"},"availability_topic":"zigbee2mqtt/bridge/state"}'
11/14/2019, 10:07:28 PM - info: MQTT publish: topic 'homeassistant/switch/0x****************/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/computer_room_wall_outlet_with_meter/set","state_topic":"zigbee2mqtt/computer_room_wall_outlet_with_meter","name":"computer_room_wall_outlet_with_meter_switch","unique_id":"0x****************_switch_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x****************"],"name":"computer_room_wall_outlet_with_meter","sw_version":"Zigbee2mqtt 1.6.0","model":"IOT remote control smart buried-type outlet (PM-C140-ZB)","manufacturer":"Dawon DNS"},"availability_topic":"zigbee2mqtt/bridge/state"}'
11/14/2019, 10:07:28 PM - info: MQTT publish: topic 'homeassistant/sensor/0x****************/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/computer_room_wall_outlet_with_meter","json_attributes_topic":"zigbee2mqtt/computer_room_wall_outlet_with_meter","name":"computer_room_wall_outlet_with_meter_linkquality","unique_id":"0x****************_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x****************"],"name":"computer_room_wall_outlet_with_meter","sw_version":"Zigbee2mqtt 1.6.0","model":"IOT remote control smart buried-type outlet (PM-C140-ZB)","manufacturer":"Dawon DNS"},"availability_topic":"zigbee2mqtt/bridge/state"}'
11/14/2019, 10:07:29 PM - info: Successfully configured computer_room_wall_outlet_with_meter (0x****************)
11/14/2019, 10:07:29 PM - info: MQTT publish: topic 'zigbee2mqtt/computer_room_wall_outlet_with_meter', payload '{"state":"ON","linkquality":78,"last_seen":1573736849417,"power":0}'

Also I was able to read the 'power' value on home assistant.
In addition, I used the 'database.db' file of zigbee2mqtt 1.7 during downgrade. and I removed the persistence_file(mosquitto.db) during downgrade.
I hope this information be helpful.

Koenkk, 2019-11-14:
On zigbee2mqtt 1.7.1, can you try changing the device definition in `devices.js` to

```js
{
    zigbeeModel: ['PM-C140-ZB'],
    model: 'PM-C140-ZB',
    vendor: 'Dawon DNS',
    description: 'IOT remote control smart buried-type outlet',
    supports: 'on/off, power and energy measurement',
    fromZigbee: [fz.on_off, fz.generic_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 3},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        
        const payload = [{
            attribute: 'instantaneousDemand',
            minimumReportInterval: 10,
            maximumReportInterval: 60,
            reportableChange: 1,
        }];
        await endpoint.configureReporting('seMetering', payload);

        await endpoint.read('seMetering', ['multiplier', 'divisor']);
    },
},
```

Instructions on how to modify this file locally can be found here: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

stkang, 2019-11-15:
@Koenkk 

I am still using version 1.6.0.

I'll upgrade and test this weekend.

Thank you for reporting the issue.

rootsnet, 2019-11-18:
I was able to read the 'power' value without 'devices.js' editing when I upgrade the zigbee2mqtt from 1.6 to 1.7.1.

And I found that 2 versions are have different 'database.db' file.

This is the 'database.db' file from first 1.7 installation.
```
{"id":1,"type":"Coordinator","ieeeAddr":"0x0012***********5","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,110],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{},"binds":[]},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{}}
{"id":2,"type":"Router","ieeeAddr":"0x001************c","nwkAddr":45148,"manufId":4602,"manufName":"DAWON_DNS","powerSource":"Mains (single phase)","modelId":"PM-C140-ZB","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,2,3,4,6,25,1794,2820,8,9],"outClusterList":[0,2,3,4,6,25,1794,2820,8,9],"clusters":{"genBasic":{"attributes":{"modelId":"PM-C140-ZB","manufacturerName":"DAWON_DNS","powerSource":1,"zclVersion":1,"appVersion":6,"stackVersion":2,"hwVersion":2,"dateCode":"2019-07-11","swBuildId":"10064168"}},"seMetering":{"attributes":{"multiplier":1,"divisor":1000}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x0012***********5","endpointID":1}]}},"appVersion":6,"stackVersion":2,"hwVersion":2,"dateCode":"2019-07-11","swBuildId":"10064168","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2}}
{"id":3,"type":"Router","ieeeAddr":"0x001************8","nwkAddr":32315,"manufId":4602,"manufName":"DAWON_DNS","powerSource":"Mains (single phase)","modelId":"PM-C140-ZB","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,2,3,4,6,25,1794,2820,8,9],"outClusterList":[0,2,3,4,6,25,1794,2820,8,9],"clusters":{"genBasic":{"attributes":{"modelId":"PM-C140-ZB","manufacturerName":"DAWON_DNS","powerSource":1,"zclVersion":1,"appVersion":6,"stackVersion":2,"hwVersion":2,"dateCode":"2019-07-11","swBuildId":"10064168"}},"seMetering":{"attributes":{"multiplier":1,"divisor":1000}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x0012***********5","endpointID":1}]}},"appVersion":6,"stackVersion":2,"hwVersion":2,"dateCode":"2019-07-11","swBuildId":"10064168","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2}}
{"id":4,"type":"Router","ieeeAddr":"0x001************7","nwkAddr":28427,"manufId":4602,"manufName":"DAWON_DNS","powerSource":"Mains (single phase)","modelId":"PM-C140-ZB","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,2,3,4,6,25,1794,2820,8,9],"outClusterList":[0,2,3,4,6,25,1794,2820,8,9],"clusters":{"genBasic":{"attributes":{"modelId":"PM-C140-ZB","manufacturerName":"DAWON_DNS","powerSource":1,"zclVersion":1,"appVersion":6,"stackVersion":2,"hwVersion":2,"dateCode":"2019-07-11","swBuildId":"10064168"}},"seMetering":{"attributes":{"multiplier":1,"divisor":1000}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x0012***********5","endpointID":1}]}},"appVersion":6,"stackVersion":2,"hwVersion":2,"dateCode":"2019-07-11","swBuildId":"10064168","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2}}
{"id":5,"type":"Router","ieeeAddr":"0x001************5","nwkAddr":51547,"manufId":4602,"manufName":"DAWON_DNS","powerSource":"Mains (single phase)","modelId":"PM-C140-ZB","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,2,3,4,6,25,1794,2820,8,9],"outClusterList":[0,2,3,4,6,25,1794,2820,8,9],"clusters":{"genBasic":{"attributes":{"modelId":"PM-C140-ZB","manufacturerName":"DAWON_DNS","powerSource":1,"zclVersion":1,"appVersion":6,"stackVersion":2,"hwVersion":2,"dateCode":"2019-07-11","swBuildId":"10064168"}},"seMetering":{"attributes":{"multiplier":1,"divisor":1000}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x0012***********5","endpointID":1}]}},"appVersion":6,"stackVersion":2,"hwVersion":2,"dateCode":"2019-07-11","swBuildId":"10064168","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2}}
{"id":6,"type":"Router","ieeeAddr":"0x001************6","nwkAddr":65263,"manufId":4602,"manufName":"DAWON_DNS","powerSource":"Mains (single phase)","modelId":"PM-C140-ZB","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,2,3,4,6,25,1794,2820,8,9],"outClusterList":[0,2,3,4,6,25,1794,2820,8,9],"clusters":{"genBasic":{"attributes":{"modelId":"PM-C140-ZB","manufacturerName":"DAWON_DNS","powerSource":1,"zclVersion":1,"appVersion":6,"stackVersion":2,"hwVersion":2,"dateCode":"2019-07-11","swBuildId":"10064168"}} [... cut ...] ":251,"meteringDeviceType":0,"temperatureUnitOfMeas":0,"temperatureFormatting":0,"instantaneousDemand":5,"demandLimit":0}},"haElectricalMeasurement":{"dir":{"value":3},"attributes":{"measurementType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x0012***********5","endpointID":1}]}},"appVersion":6,"stackVersion":2,"hwVersion":2,"dateCode":"2019-07-11","swBuildId":"10064168","zclVersion":1,"meta":{"configured":2}}
```

Please let me know if I should test 'devices.js' editing.

stkang, 2019-11-18:
@rootsnet 

It does not work in 1.7.1.
You will need to modify the device.js file recommended by @Koenkk .

rootsnet, 2019-11-18:
@stkang 

Thank you for your comment.

I will try to clean the configuration up(clean install) and re-pair devices in 1.7.1 with above 'devices.js' editing on this weekend.
Also I will share the result on this page.

rootsnet, 2019-11-24:
@Koenkk 
I did to clean install as master version and applied your commented code.
Unfortunately, the commented code did not work.

Koenkk, 2019-11-25:
@rootsnet can you try the solution from: https://github.com/Koenkk/zigbee-herdsman/issues/89#issuecomment-558246497

rootsnet, 2019-11-26:
I think I can test latest dev branch this next weekend or the weekend after next.
I will share the result on this page if I finish to test.

rootsnet, 2020-01-24:
@Koenkk

I am very sorry for late comment.

I tested a PM-C140-ZB device on current dev version(1.9.0-dev).
I re-paired the device, and it works well with power attribute.

I will test 1.9.0 version soon. and I will update the result.

rootsnet, 2020-01-24:
@Koenkk

I tested 1.9.0 version. 
It works well with power attribute.
Thank you so much for your help.

Koenkk, 2020-01-24:
Cool!

closed 2020-01-24 by Koenkk