Koenkk/zigbee2mqtt #2333: npm ERR! zigbee2mqtt@1.7.1 start: `node index.js`

Glenn-Dandy, 2019-11-13:
Hello i have installed the Moduls
now when i try npm start. i got this messages

`> zigbee2mqtt@1.7.1 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2019-11-13T07:11:34: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-11-13.08-11-33'
zigbee2mqtt:info  2019-11-13T07:11:34: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
zigbee2mqtt:info  2019-11-13T07:11:34: Starting zigbee-herdsman...
zigbee2mqtt:error 2019-11-13T07:11:34: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-11-13T07:11:34: Failed to start zigbee
zigbee2mqtt:error 2019-11-13T07:11:34: Exiting...
zigbee2mqtt:error 2019-11-13T07:11:34: Error: /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node: undefined symbol: _ZN2v88internal35IsolateFromNeverReadOnlySpaceObjectEj
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:730:18)
    at Module.load (internal/modules/cjs/loader.js:600:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
    at Function.Module._load (internal/modules/cjs/loader.js:531:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at bindings (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/linux.js:2:36)
    at Module._compile (internal/modules/cjs/loader.js:701:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.7.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.7.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/wozise/.npm/_logs/2019-11-13T07_11_34_420Z-debug.log`

the debug log says

`0 info it worked if it ends with ok
1 verbose cli [ '/home/wozise/.nvm/versions/node/v10.15.3/bin/node',
1 verbose cli   '/home/wozise/.nvm/versions/node/v10.15.3/bin/npm',
1 verbose cli   'start' ]
2 info using npm@6.4.1
3 info using node@v10.15.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.7.1~prestart: zigbee2mqtt@1.7.1
6 info lifecycle zigbee2mqtt@1.7.1~start: zigbee2mqtt@1.7.1
7 verbose lifecycle zigbee2mqtt@1.7.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.7.1~start: PATH: /home/wozise/.nvm/versions/node/v10.15.3/lib/nod$
9 verbose lifecycle zigbee2mqtt@1.7.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.7.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.7.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.7.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.7.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/home/wozise/.nvm/versions/node/v10.15.3/lib/node$
13 verbose stack     at EventEmitter.emit (events.js:189:13)
13 verbose stack     at ChildProcess.<anonymous> (/home/wozise/.nvm/versions/node/v10.15.3/lib/node$
13 verbose stack     at ChildProcess.emit (events.js:189:13)
13 verbose stack     at maybeClose (internal/child_process.js:970:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.7.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.15.0-51-generic
17 verbose argv "/home/wozise/.nvm/versions/node/v10.15.3/bin/node" "/home/wozise/.nvm/versions/nod$
18 verbose node v10.15.3
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.7.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.7.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]`

hm i dont know why. I cant start zigbee2mqtt.

JohanPotgieter, 2019-11-15:
I have the same issue with this version on Raspberry PI 3 running dietPI. Also tried with Raspian Stretch.

Glenn-Dandy, 2019-11-15:
i have found the problem. 

when i do  
sudo chown (myuser):dialout /dev/ttyACM0

after then the npm stark works

when u use a pi check pi:pi  @/dev/ttyACM0 <- the port where the stick is.
i dont know why the script cant open when the port is root:root

Koenkk, 2019-11-15:
@Glenn-Dandy in that case you also need to run zigbee2mqtt as root, otherwise the process does not have access. This is also explained here: https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start. Good that you solved your issue.

closed 2019-11-15 by Koenkk

ntkoopman, 2021-12-04:
I had this issue today and solved it be rebuilding with `npm rebuild`