Koenkk/zigbee2mqtt #2334: double mqtt message

kapiorr, 2019-11-13:
Zigbee2mqtt version latest.
Xiaomi Mija wireless switch WXKG01LM

On click zigbee2mqtt send double message:

Log:
```
Nov 13 18:38:50 raspberrypi npm[10970]: zigbee2mqtt:info  2019-11-13T18:38:50: MQTT publish: topic 'zigbee2mqtt/Przycisk', payload '{"linkquality":131,"click":"single"}'
Nov 13 18:38:50 raspberrypi npm[10970]: zigbee2mqtt:info  2019-11-13T18:38:50: MQTT publish: topic 'zigbee2mqtt/Przycisk', payload '{"linkquality":131,"click":""}'
Nov 13 18:38:52 raspberrypi npm[10970]: zigbee2mqtt:info  2019-11-13T18:38:52: MQTT publish: topic 'zigbee2mqtt/Przycisk', payload '{"linkquality":131,"click":"double"}'
Nov 13 18:38:52 raspberrypi npm[10970]: zigbee2mqtt:info  2019-11-13T18:38:52: MQTT publish: topic 'zigbee2mqtt/Przycisk', payload '{"linkquality":131,"click":""}'
Nov 13 18:38:54 raspberrypi npm[10970]: zigbee2mqtt:info  2019-11-13T18:38:54: MQTT publish: topic 'zigbee2mqtt/Przycisk', payload '{"linkquality":136,"click":"double"}'
Nov 13 18:38:54 raspberrypi npm[10970]: zigbee2mqtt:info  2019-11-13T18:38:54: MQTT publish: topic 'zigbee2mqtt/Przycisk', payload '{"linkquality":136,"click":""}
```

Koenkk, 2019-11-14:
This is expected, see https://github.com/Koenkk/zigbee2mqtt/issues/1155#issuecomment-467094619

Duplicate of #1155 and #1212

closed 2019-11-14 by Koenkk

kapiorr, 2019-11-14:
Maybe expected but now showing last action on button in ha.

Koenkk, 2019-11-14:
That is true, but this is needed to trigger a state change which can be used in automations.

kapiorr, 2019-11-14:
My automations worked without doble messages (in my case) on older versions zigbe2mqtt.