Koenkk/zigbee2mqtt #2335: Unable to adjust RGB and WHITE simultaneously.

igorpuchkovcom, 2019-11-14:
Unable to adjust RGB and WHITE simultaneously.

If white_value is present RGB controller just switching off.
Same behaviour is for RGB. When there is no white_value value present WHITE controller switching off.

No way to have both controlles switched on and managed at the same time.

There is three independent parameters:
- Color
- Brightness (applied only for RGB controller)
- White value (applied only for WHITE controller)

ON/OFF state should be shared between controllers.

_Originally posted by @igorpuchkovcom in https://github.com/Koenkk/zigbee2mqtt/issues/1711#issuecomment-553284153_

Koenkk, 2019-11-14:
If this is about a gledopto controller, this is indeed true this matches the behaviour when comnected to the hue hub. Previously people were having troubles turning of the white leds.

igorpuchkovcom, 2019-11-14:
Yes, it is about Gledopto GL-C-007 controller. It is not about hue hub.

toZegbee converter just not accept RGB parameters if white_value key is present and switch RGB off.
If white_value key is NOT present then it only accept RGB parameters and switch white leds off.
Also brightness and white value parameters are linked.
Brightness stay for RGB only.

With actual implementation there is no way to manage white and RGB at the same time.

Any argument about this behaviour?

Why just not allow no manage white (white_value) and rgb (color and brightness) independently?

Koenkk, 2019-11-14:
There was quite some discussion about this here: https://github.com/Koenkk/zigbee2mqtt/issues/1731

From what I remember of it: having both channels on (RGB and white) causes other buggy side effects.

Eventually Gledopto has send me their controllers, I've compared the behavior of them when paired with the Hue bridge and implemented it in the same way in zigbee2mqtt.

Oggymator, 2019-12-17:
In Hue App it is possible to turn on RGB and White.

Koenkk, 2019-12-17:
In the latest dev you should now be able to do this.

In order to enable it, set `separate_control: true` for the device in the zigbee2mqtt `configuration.yaml`. Example

```yaml
devices:
  '0x00124b001a13b3b3':
    friendly_name: 'gledopto_controller'
    separate_control: true
```

stoinov, 2019-12-22:
How exactly do you set and get separate controls? I tied using `white_value` but it does not seem to have any effect. Using the `get` topic does not return any different information than what is on the stable branch.

rcarmo, 2019-12-25:
Merry Xmas!

I'd really like to see some sample MQTT payloads for this for those of us who don't use Home Assistant. I'm trying to convert HSL values to RGBW, and in between #1731, #1711 and #1738, I'm having a little trouble figuring out the right payloads for setting HSL color in RGB mode and color temperature  in "white" mode...

stoinov, 2019-12-27:
@rcarmo [This link](https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttfriendly_nameset) should give you an idea how to use all possible options for setting RGB light (you can directly set RGB color) and also just use `color_temp` for white mode.
The current request we have in this thread, is option to allow for option of using both RGB and White/CCT concurently, as the current version only allows for one of them at a time.

dasoho, 2019-12-27:
Many thanks @Koenkk and @andreasbrett for addressing this issue which I also raised with #2540 
but there is still one issue left which I also mentioned in #2540:
I have a GL-C-008 which actually has two white channels (warm white & cold white (CCT)) in addition to the rgb channels. With e.g. {"white_value":100} I just can control the cold white LEDs, the warm white LED's stay off. In an earlier release, the white LED's were controlled with a separate endpoint for white and there I was able to set the color_temp on the white endpoint.
To get this functionality back with the new concept (no seperate white endpoint), I guess it would be necessary to have an additional property like "white_color_temp" along with "white_value". 
Would this make sense?

rcarmo, 2019-12-29:
Thanks @stoinov, I hadn’t found that page in the docs (I did search, but somehow it just never came up).

nixent, 2020-01-12:
Experience same behavior with GL-C-008 2IDs as described by @dasoho in the [comment](https://github.com/Koenkk/zigbee2mqtt/issues/2335#issuecomment-569330563): only warm white can be controlled on CCT channel.
My device reports its model as GL-C-007 and has endpoints [11,13,15], where 11 is for RGB and 15 is for CCT.

I have tried to switch from white_value to color_temp and change [converters.light_brightness](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js#L1154) to converters.light_colortemp. That didn't change anything. I still can control brightness of CCT channel with brightness payload but color_temp doesn't affect CCT channel.

Another observation is that color_temp payload is actually changing color values on RGB channel. It looks like color_temp payload works with RGB channel instead of CCT. Cannot find in the code to confirm. 

@Koenkk could you please point if my assumption about payload for CCT that goes to RGB is correct and where to look for reasons?

rickkdotnet, 2020-01-13:
I just wanted to chime in that I'm affected by this issue too. I have 2 2ID controllers which each control 2 sets of lights (so a set of RGB and a set of white lights per controller) and I can only turn on either white or RGB from z2m.

FYI this change also causes problems with the Home Assistant scene functionality, because a white_value: -1 is somehow reported back to HA. Home Assistant sees this as invalid and won't save the scene from the editor, and throws errors when you want to apply a scene. Unfortunately I didn't have time to debug this further.

To be honest, I think the ideal behavior with a 2ID controller is that white and and RGB lights are presented as separate entities.

nixent, 2020-01-13:
@rickkdotnet just use [separate_control: true](https://github.com/Koenkk/zigbee2mqtt/issues/2335#issuecomment-566701129) to control RGB and CCT simultaneously, however it will allow to change brightness of warm white only, cold white will be off, color_temp will change color of RGB.

nixent, 2020-01-13:
Regarding handling of RGB+CCT devices, it would be best to handle RGB and CCT endpoints as separate devices, regardless if it is 1ID or 2ID. 
RGB device should accept `brightness` and `color`, while CCT take own `brightness` and `color_temp`.

I'll check how it is done for hue devices, Gledopto implementation should be similar.

Koenkk, 2020-01-13:
@nixent I disagree, when you then want to turn off a physical single light device, you have to turn of 2 devices, this is very inconvenient. But perhaps it should be activated when `separate_control: true` is used.

stoinov, 2020-01-13:
@Koenkk I agree that this should be setting if possible. But I wonder is there a physical difference of the two types of controller 1ID and 2ID in how Z2M sees them? Is there a way for a user to identify it easily? I have two, but there are no markings on them.
If there is a hardware difference that will add significant overhead, then maybe we should identify them as distinct devices and have them manifest 1 or 2 lights respectively, instead of trying to hack and fix some unforeseen problems.
Given what I've read in all of the issues and the explanation [posted here](https://www.reddit.com/r/Gledopto/comments/8spz43/which_gledopto_controller_do_i_need/) (toward the end of the post), I think this would be the optimal solution.

nixent, 2020-01-14:
Current code for GL-C-007/GL-C-008 implements logic to control RGB and CCT/W channels **simultaneously**. White channel can be controlled by `white_value`. CCT control doesn't work at the moment due to small bug, I have fixed it, will make a pull request after testing.

@Koenkk has started implementation of independent control of RGB and CCT/W channels with introduction of `separate_control` flag, however there are few unresolved issues. 
Both, RGB and CCT endpoints accept: `state`, `brightness`, `color_temp`. To control them separately there must be sort of flag to indicate destination endpoint. One of the obvious solutions is to keep `state`, `brightness`, `color_temp` to control both endpoints simultaneously and add `RGB_*` and `White_*` commands for independent control. The other option is to register separate devices and use kind of proxy, similar to `via_device` in Home assistant. Any good ideas?

stoinov, 2020-01-14:
I agree with @nixent suggestions but still I am not sure if those solutions depend on the type of the GL-C-008 contrller - 1ID ot 2ID. If not, then I think by default a `RGB_*` and `white_*` options should be implemented, and if `separate_control` is set, then two distinct devices should be presented to the user. This I think will give the broadest options for people to choose from.
One concern can be how HA integration should be handled and if used, a error can be thrown if incompatible mode is selected.

NOTE: I have GL-MC-001 and this seems to have identical functionality (RGBCCT) but not sure if it's presented as 1 device or two distinct ones. Any links to how-to's on sniffing up those details?

IVI053, 2020-02-02:
With `seperate_control` set for my GL-C-007	I'm unable to figure out how to turn on all LEDs together. It seems that the flag did not really change anything. Could someone please provide a payload that should turn RGBW fully on?

BMicraft, 2020-02-07:
@IVI053 Recent devices should have a handwritten number on the qc sticker on the back of the device.

My GL-C-008 (rgb+cct) 2id controller which should advertise itself as GL-C-006 and GL-C-007 simultaneously (according to gledopto) gets misidentified as GL-C-007 only which leaves me completely unable to control color temperature in 99dc214fbe7251b5bd65897dea44df6c0002493a (did not work before either).

Could this be related to #2872?

closed 2020-04-15 by stale[bot]