Koenkk/zigbee2mqtt #2337: MQTT message storm on Tradfri dimmer

vdkeybus, 2019-11-14:
# Bug Report
## What happened
When turning the Ikea Tradfri dimmer, a lot of messages is issued. In fact too much for HA, to the point that the web page is no longer served ('connection lost'). But that's another issue.

Occasionally (but in my case approx. immediately) zigbee2mqtt continues to send these messages at high rate. The only way to stop this is restart zigbee2mqtt. Obviously HA chokes completely.

## What did you expect to happen
No messages should be sent.

## How to reproduce it (minimal and precise)
Turn a Tradfri dimmer and observe.

## Debug Info
On dev branch
zigbee2mqtt/bridge/config {"version":"1.6.0","commit":"6f92a41","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}

Koenkk, 2019-11-14:
Duplicate of https://github.com/Koenkk/zigbee-herdsman/issues/88#issuecomment-552531131 (solution is also there)

vdkeybus, 2019-11-14:
@Koenkk:

The issue is not the amount of messages (for which your rate-limit fix would work), but the fact they are repeated indefinitely (for hours; I have 6.5 million of them so far). I'm not sure if they are actual transmissions. I've seen a minutes-long streak stop when a sensor in the same area made a transmission, but this could still be coincidence.

When you turn the dimmer and you get, say, 250 messages, HA indeed gets slow, but eventually recovers.

The case I want to bring up is that in my system, the same messages were repeated indefinitely. Normally, the Tradfri dimmers issue many, but different messages,

'normal' behaviour:
`Nov 07 18:16:12 zacate npm[2571]: zigbee2mqtt:info  2019-11-07T17:16:12: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe11ab90', payload '{"brightness":0,"linkquality":42,"rate":195,"level":0,"battery":60,"action":"rotate_right"}'
Nov 07 18:16:12 zacate npm[2571]: zigbee2mqtt:info  2019-11-07T17:16:12: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe11ab90', payload '{"brightness":0,"linkquality":42,"rate":195,"level":0,"battery":60,"action":""}'
Nov 07 18:16:12 zacate npm[2571]: zigbee2mqtt:info  2019-11-07T17:16:12: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe11ab90', payload '{"brightness":0,"linkquality":42,"rate":70,"level":0,"battery":60,"action":"rotate_right"}'
Nov 07 18:16:12 zacate npm[2571]: zigbee2mqtt:info  2019-11-07T17:16:12: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe11ab90', payload '{"brightness":0,"linkquality":42,"rate":70,"level":0,"battery":60,"action":""}'
Nov 07 18:16:12 zacate npm[2571]: zigbee2mqtt:info  2019-11-07T17:16:12: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe11ab90', payload '{"brightness":1,"linkquality":42,"rate":70,"level":0,"battery":60,"action":"rotate_right"}'
Nov 07 18:16:12 zacate npm[2571]: zigbee2mqtt:info  2019-11-07T17:16:12: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe11ab90', payload '{"brightness":1,"linkquality":42,"rate":70,"level":0,"battery":60,"action":""}'`

And the new, problematic behviour:

`Nov 13 12:35:53 zacate npm[24452]: zigbee2mqtt:info  2019-11-13T11:35:53: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe11ab90', payload '{"brightness":255,"linkquality":31,"rate":70,"level":0,"battery":74,"action":"rotate_right"}'
Nov 13 12:35:53 zacate npm[24452]: zigbee2mqtt:info  2019-11-13T11:35:53: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe11ab90', payload '{"brightness":255,"linkquality":31,"rate":70,"level":0,"battery":74,"action":""}'
Nov 13 12:35:53 zacate npm[24452]: zigbee2mqtt:info  2019-11-13T11:35:53: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe11ab90', payload '{"brightness":255,"linkquality":31,"rate":70,"level":0,"battery":74,"action":"rotate_right"}'
Nov 13 12:35:53 zacate npm[24452]: zigbee2mqtt:info  2019-11-13T11:35:53: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe11ab90', payload '{"brightness":255,"linkquality":31,"rate":70,"level":0,"battery":74,"action":""}'
Nov 13 12:35:53 zacate npm[24452]: zigbee2mqtt:info  2019-11-13T11:35:53: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe11ab90', payload '{"brightness":255,"linkquality":31,"rate":70,"level":0,"battery":74,"action":"rotate_right"}'
Nov 13 12:35:53 zacate npm[24452]: zigbee2mqtt:info  2019-11-13T11:35:53: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe11ab90', payload '{"brightness":255,"linkquality":31,"rate":70,"level":0,"battery":74,"action":""}'`

Koenkk, 2019-11-18:
The problem is probably that your remote doesn't send the stop event (https://github.com/Koenkk/zigbee-herdsman-converters/commit/ad29b2b6bcdb8ae561d21f2c8baf5c82fd681e51#diff-63bec0da09650ada978527cc004902e2R131) therefore the interval timer keeps continuing. Does this always happen?

vdkeybus, 2020-01-19:
It happens quite a lot. Is this zigbee2mqtt sending 'continuation' messages ? Is there actually nothing received ?

It may be happening when a message is lost, but z2m shouldn't be behaving like that, no ?

Koenkk, 2020-01-20:
@vdkeybus I agree, the tradfri dimmer integration should be rewritten, see https://github.com/Koenkk/zigbee2mqtt/issues/2627#issuecomment-570646883 (therefore closing this).

closed 2020-01-20 by Koenkk