Koenkk/zigbee2mqtt #2338: 1.7.1+dev -startup problem

Kryzek, 2019-11-14:
I updated to latest dev and Z2M does not start. Using CC2652R.
```
zigbee2mqtt:info  2019-11-14T09:21:14: Starting zigbee2mqtt version 1.7.1+dev (commit #e33b49e)
zigbee2mqtt:info  2019-11-14T09:21:14: Starting zigbee-herdsman...
zigbee2mqtt:debug 2019-11-14T09:21:14: Using zigbee-herdsman with settings: '{"network":{"panID":6755,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":null}}'
zigbee2mqtt:error 2019-11-14T09:22:15: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-11-14T09:22:15: Failed to start zigbee
zigbee2mqtt:error 2019-11-14T09:22:15: Exiting...
zigbee2mqtt:error 2019-11-14T09:22:15: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
```

Does not start with my backup -version either.

With DEBUG=*
```
  serialport/stream binding.read finished { bytesRead: 3 } +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [8,0,143] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,103,0,0,100,105,161,28,0,75,18,0,0,0,7,8,0,143] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 7 - 0 - [di�] - 143 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- UTIL - getDeviceInfo - {"status":0,"ieeeaddr":"0x00124b001ca16964","shortaddr":0,"devicetype":7,"devicestate":8,"numassocdevices":0,"assocdeviceslist":[]} +12ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:startZnp Start ZNP as coordinator... +12ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - startupFromApp - {"startdelay":100} +11ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,37,64,100,0,3] +11ms
  serialport/stream _write 7 bytes of data +1ms
  serialport/binding-abstract write 7 bytes +1ms
  serialport/bindings/unixWrite Starting write 7 bytes offset 0 bytesToWrite 7 +10ms
  serialport/stream _read reading { start: 120, toRead: 65416 } +1ms
  serialport/binding-abstract read +1ms
  serialport/bindings/unixRead Starting read +2ms
  serialport/bindings/unixWrite write returned: wrote 7 bytes +1ms
  serialport/bindings/unixWrite Finished writing 7 bytes +0ms
  serialport/stream binding.write write finished +0ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +4ms
  serialport/bindings/poller received "readable" +2ms
  serialport/binding-abstract read +2ms
  serialport/bindings/unixRead Starting read +2ms
  serialport/bindings/unixRead Finished read { bytesRead: 6,
  buffer:
   <Buffer fe 0e 61 02 02 01 02 07 01 82 17 34 01 00 ff ff ff ff ca fe 03 61 08 00 01 55 3e fe 03 61 08 00 01 55 3e fe 06 61 08 00 04 00 08 00 00 63 fe 03 61 08 ... > } bytes +0ms
  serialport/stream binding.read finished { bytesRead: 6 } +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,64,0,36] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,64,0,36] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 64 - [] - 36 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - startupFromApp - {"status":0} +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  serialport/stream _read reading { start: 126, toRead: 65410 } +0ms
  serialport/binding-abstract read +0ms
  serialport/bindings/unixRead Starting read +1ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +1ms
zigbee2mqtt:error 2019-11-14T09:31:44: Error while starting zigbee-herdsman
winston:file written true false
zigbee2mqtt:error 2019-11-14T09:31:44: Failed to start zigbee
winston:file written true false
zigbee2mqtt:error 2019-11-14T09:31:44: Exiting...
winston:file written true false
zigbee2mqtt:error 2019-11-14T09:31:44: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
winston:file written true false
```

Kryzek, 2019-11-14:
I tested with clean install of dev-branch and it works...
``` 
zigbee2mqtt:info  2019-11-14T10:47:30: Logging to directory: '/opt/zigbee2mqtt-clean/data/log/2019-11-14.12-47-30'
zigbee2mqtt:debug 2019-11-14T10:47:30: Can't load state from file /opt/zigbee2mqtt-clean/data/state.json (doesn't exist)
zigbee2mqtt:info  2019-11-14T10:47:30: Starting zigbee2mqtt version 1.7.1+dev (commit #e33b49e)
zigbee2mqtt:info  2019-11-14T10:47:30: Starting zigbee-herdsman...
zigbee2mqtt:debug 2019-11-14T10:47:30: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt-clean/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt-clean/data/database.db.backup","backupPath":"/opt/zigbee2mqtt-clean/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":null}}'
zigbee2mqtt:info  2019-11-14T10:47:32: zigbee-herdsman started
zigbee2mqtt:info  2019-11-14T10:47:32: Coordinator firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'
zigbee2mqtt:debug 2019-11-14T10:47:32: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
zigbee2mqtt:info  2019-11-14T10:47:32: Currently 0 devices are joined:
zigbee2mqtt:info  2019-11-14T10:47:32: Zigbee: disabling joining new devices.
```

Kryzek, 2019-11-14:
I just could not get my network back running with "panID":6755 (which was in configured in configuration with ```pan_id: 0x1a63```) for some reason. I tried shutting down my routers by de-powering them. Finally I just decided to change  ```pan_id: 0x1a62```  and re-pair everything.

yo-han, 2019-11-15:
I had the same error yesterday. Unplugged the sniffer, restarted zigbee2mqtt and everything worked as aspected again. Maybe it helps.

XRyu, 2019-11-15:
sadly no, did not work. Tried even reflashing the CC1352P2 board, restarted my docker-server but nothing helped. i really hope i wont need to re-pair everything again :/

Koenkk, 2019-11-15:
This shouldn't happen just after upgrading, did you do anything regarding changing channels or panid?

XRyu, 2019-11-16:
No, nothing changed, just stopped working after update and restart on wed 18:something (last status in HA)

Kryzek, 2019-11-16:
I experienced this after doing following steps:
Stop Z2M with ctrl-c as I run it in screen on my Linux-box.
Backup whole Z2M directory (/opt/z2m)another folder and create backup from data-directory also.
Update.
Restore data-contents from data-backup.
And npm start.

It did not start. And after I noticed that I could not get it running with original install I did new one with started on another panID (logs also above). I tried re-flashing on another workstation also. It did not change anything. 

So I cleared my db, and devices etc and changed panID and re-paired everything. Works again and thank god I have only one hard to pair device (Gledopto). Others I can reset using Hue dimmer.

Oh. And binds and groups are cool since they work even if Z2M is down as long as devices are not losing power. Maybe we could have some ”examples” page on documentation how to setup those with a bit more steps in explanation?  Such as grouping bulb with Ikea motion sensor or binding Hue dimmer to bulb etc...

str3000, 2019-11-16:
Hello,
I have a similar problem.
I bought a Texas Instruments CC1352P-2 and wan't to connect it. last hex file flashed.
stick has "ttyAMC1", my CC2531 has ttyAMC0.
After a lot of trouble shooting I installed z2m release new.
Using my running configuration (data directory)

Behavior with 1.7.1:
- CC2531 running without problem
-  CC1352P-2 not running

Log:
error 2019-11-15T21:22:12: Error while starting zigbee-herdsman

npm log:
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.13.0
3 info using node@v10.17.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.7.1~prestart: zigbee2mqtt@1.7.1
6 info lifecycle zigbee2mqtt@1.7.1~start: zigbee2mqtt@1.7.1
7 verbose lifecycle zigbee2mqtt@1.7.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.7.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
9 verbose lifecycle zigbee2mqtt@1.7.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.7.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.7.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.7.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.7.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.7.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.15.0-70-generic
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.17.0
19 verbose npm  v6.13.0
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.7.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.7.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

XRyu, 2019-11-17:
Had no chance, tried re-pairing, new clean config folder and yaml (i am using z2m in docker on synology)
Tried reflashing several times but nothing works.

right now i have no clue whats going wrong and where to look what might fix it..
it looks the device-path is correctly available on ttyACM0 and now i dont know anymore.
@Koenkk any idea?

These are the logs of the last few failed starts:
```
Using '/app/data' as data directory,
,
> zigbee2mqtt@1.7.1 start /app,
> node index.js,
,
zigbee2mqtt:info  2019-11-17T02:07:46: Logging to directory: 'data/log/2019-11-17.03-07-45',
zigbee2mqtt:debug 2019-11-17T02:07:46: Removing old log directory 'data/log/2019-11-17.02-57-22',
zigbee2mqtt:debug 2019-11-17T02:07:46: Loaded state from file /app/data/state.json,
zigbee2mqtt:info  2019-11-17T02:07:46: Starting zigbee2mqtt version 1.7.1+dev (commit #31246ef),
zigbee2mqtt:info  2019-11-17T02:07:46: Starting zigbee-herdsman...,
zigbee2mqtt:debug 2019-11-17T02:07:46: Using zigbee-herdsman with settings: '{"network":{"panID":6755,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[12],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}',
2019-11-17T02:07:46.592Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[57,139,99,167,56,46,16,240,106,183,156,189,144,1,100,138],"panID":6755,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[12]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json"}',
2019-11-17T02:07:46.594Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":true,"autoOpen":false},
2019-11-17T02:07:46.600Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened,
2019-11-17T02:07:46.600Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload,
2019-11-17T02:07:46.601Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239],
2019-11-17T02:07:47.606Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {},
2019-11-17T02:07:47.607Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35],
2019-11-17T02:07:47.611Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,1,2,7],
2019-11-17T02:07:47.612Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,1,2,7],
2019-11-17T02:07:47.612Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [1,130,23,52,1,0,255,255,255,255,202],
2019-11-17T02:07:47.613Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,1,2,7,1,130,23,52,1,0,255,255,255,255,202],
2019-11-17T02:07:47.614Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [�4����] - 202,
2019-11-17T02:07:47.615Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106},
2019-11-17T02:07:47.617Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [],
2019-11-17T02:07:47.618Z zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack3x0' ({"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}),
2019-11-17T02:07:47.619Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":96,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}},
2019-11-17T02:07:47.620Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,96,0,0,74],
2019-11-17T02:07:47.628Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,8,2,0,105],
2019-11-17T02:07:47.628Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,8,2,0,105],
2019-11-17T02:07:47.628Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 8 - [] - 105,
2019-11-17T02:07:47.629Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":2,"len":0,"value":{"type":"Buffer","data":[]}},
2019-11-17T02:07:47.629Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [],
2019-11-17T02:07:47.630Z zigbee-herdsman:adapter:zStack:startZnp Restoring coordinator from backup,
2019-11-17T02:07:47.632Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {},
2019-11-17T02:07:47.632Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35],
2019-11-17T02:07:47.635Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,1,2,7],
2019-11-17T02:07:47.635Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,1,2,7],
2019-11-17T02:07:47.635Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [1,130,23,52,1,0,255,255],
2019-11-17T02:07:47.635Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,1,2,7,1,130,23,52,1,0,255,255],
2019-11-17T02:07:47.636Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [255,255,202],
2019-11-17T02:07:47.636Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,1,2,7,1,130,23,52,1,0,255,255,255,255,202],
2019-11-17T02:07:47.636Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [�4����] - 202,
2019-11-17T02:07:47.636Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106},
[... cut ...]
    at tryOnTimeout (timers.js:300:5),
    at listOnTimeout (timers.js:263:5),
    at Timer.processTimers (timers.js:223:10),
npm ERR! code ELIFECYCLE,
npm ERR! errno 1,
npm ERR! zigbee2mqtt@1.7.1 start: `node index.js`,
npm ERR! Exit status 1,,
 ,
npm ERR! Failed at the zigbee2mqtt@1.7.1 start script.,
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.,
,
npm ERR! A complete log of this run can be found in:,
npm ERR!     /root/.npm/_logs/2019-11-17T02_11_04_945Z-debug.log,
```

Kryzek, 2019-11-17:
You seem to have same panID that I could not get my instance to run with ("panID":6755”). Do you have other zigbee networks?

For me it helped when I added under
advanced:
  pan_id: 0x1a62

To configuration file.

Kryzek, 2019-11-17:
Every log with problems here seem to have "panID":6755.

wes1993, 2019-11-17:
Hello to Everyone.
I have the same problem, there is a way to downgrade to the previus version?
I have Hassio.
Thanks a Lot
Stefano

XRyu, 2019-11-17:
I now also changed the pan_id to 6754 (0x1a62) and re-paired everything, now i hope i wont have to do this more often XD

dextraa9791, 2019-12-08:
https://hastebin.com/igasoxineb.coffeescript

Also having the same isseu with the 1.7.1 build. My pan_id is set to default 6754.
What can i do to fix this?

Koenkk, 2019-12-09:
@dextraa9791 did you try a different panid?

dextraa9791, 2019-12-09:
> @dextraa9791 did you try a different panid?

What panid should i try?

Koenkk, 2019-12-09:
`6755`

dextraa9791, 2019-12-15:
> `6755`

@Koenkk still get the same error, it looked like the led on my zigbee usb started flashing red while trying to start zigbee2mqtt

dextraa9791, 2019-12-15:
> > `6755`
> 
> @Koenkk still get the same error, it looked like the led on my zigbee usb started flashing red while trying to start zigbee2mqtt

replugged the usb device and restarted the addon, now it seems to work, error isn't showing up anymore. Changing pan_id to 6755 worked.

closed 2020-02-20 by stale[bot]