Koenkk/zigbee2mqtt #2339: Danalock on 1.7.x

martonbtoth, 2019-11-14:
# Bug Report
## What happened
Commands to danalock time out, state does not update. Like this:

```
zigbee2mqtt:info  2019-11-14T19:51:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'lock' failed: 'Error: Timeout - 28358 - 1 - 5 - 257 - 11 after 15000ms'","meta":{"friendly_name":"lock"}}'
```

I could not even repair the device on 1.7.0 and 1.7.1:

```
zigbee2mqtt:info  2019-11-14T19:58:23: Starting interview of '0x000b57fffe59cf4a'
zigbee2mqtt:info  2019-11-14T19:58:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000b57fffe59cf4a"}}'
zigbee2mqtt:error 2019-11-14T19:58:45: Failed to interview '0x000b57fffe59cf4a', device has not successfully been paired
zigbee2mqtt:info  2019-11-14T19:58:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x000b57fffe59cf4a"}}'
```

Reflashing the stick/removing zigbee2mqtt state/config/db files and starting over did not work either. 

I'm going to debug this further during the weekend.

## What did you expect to happen

Commands do not time out, state updates. Just like on 1.6

## How to reproduce it (minimal and precise)

1. Pair danalock v3 on 1.6.0
2. Update to 1.7.0
3. Try to change the state of the lock

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: CC2531_DEFAULT_20190608

Koenkk, 2019-11-15:
When executing the commands, does the lock physically change state?

martonbtoth, 2019-11-16:
If I pair it on 1.6.0 and upgrade to 1.7 then yes. I am able to open/close the lock that way.

nanosonde, 2019-11-16:
Sams problem here.
CC2531 with 
```
Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}
```
Danalock V3 used to work before.

Scratch installation of 1.7.1 with no devices paired at the beginning.
```
Nov 16 12:17:40 rpi3-knx npm[26085]: zigbee2mqtt:info  2019-11-16T11:17:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000b57fffe59d134"}}'
Nov 16 12:17:40 rpi3-knx npm[26085]: zigbee2mqtt:info  2019-11-16T11:17:40: Starting interview of '0x000b57fffe59d134'
Nov 16 12:17:40 rpi3-knx npm[26085]: zigbee2mqtt:info  2019-11-16T11:17:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000b57fffe59d134"}}'
Nov 16 12:18:00 rpi3-knx npm[26085]: zigbee2mqtt:error 2019-11-16T11:18:00: Failed to interview '0x000b57fffe59d134', device has not successfully been paired
Nov 16 12:18:00 rpi3-knx npm[26085]: zigbee2mqtt:info  2019-11-16T11:18:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x000b57fffe59d134"}}'
```

nanosonde, 2019-11-16:
Herdsman debug logs: 

```
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +19ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,43,200,52,209,89,254,255,87,11,0,0,0,129] +29s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,43,200,52,209,89,254,255,87,11,0,0,0,129] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [+�4�Y��W
                                                                                    ] - 129 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":51243,"extaddr":"0x000b57fffe59d134","parentaddr":0} +29s
  zigbee-herdsman:controller:log Device '0x000b57fffe59d134' joined +29s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +6ms
  zigbee-herdsman:controller:log Device '0x000b57fffe59d134' accepted by handler +3ms
  zigbee-herdsman:controller:log New device '0x000b57fffe59d134' joined +0ms
  zigbee-herdsman:controller:log Creating device '0x000b57fffe59d134' +0ms
zigbee2mqtt:info  2019-11-16T11:24:43: Device '0x000b57fffe59d134' joined
zigbee2mqtt:info  2019-11-16T11:24:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000b57fffe59d134"}}'
  zigbee-herdsman:controller:log Interview '0x000b57fffe59d134' start +17ms
zigbee2mqtt:info  2019-11-16T11:24:43: Starting interview of '0x000b57fffe59d134'
zigbee2mqtt:info  2019-11-16T11:24:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000b57fffe59d134"}}'
  zigbee-herdsman:controller:device Interview - start device '0x000b57fffe59d134' +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":51243,"nwkaddrofinterest":51243} +57s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,43,200,43,200,35] +57s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +46ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [] - 102 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +57s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,43,200,52,209,89,254,255,87,11,0,0,0,129] +4s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,43,200,52,209,89,254,255,87,11,0,0,0,129] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [+�4�Y��W
                                                                                    ] - 129 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":51243,"extaddr":"0x000b57fffe59d134","parentaddr":0} +4s
  zigbee-herdsman:controller:log Device '0x000b57fffe59d134' joined +4s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:controller:log Device '0x000b57fffe59d134' accepted by handler +1ms
  zigbee-herdsman:controller:log Not interviewing '0x000b57fffe59d134', completed 'false', in progress 'true' +1ms
  zigbee-herdsman:controller:device Interview - first node descriptor request failed for '0x000b57fffe59d134', retrying... +10s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":51243,"nwkaddrofinterest":51243} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,43,200,43,200,35] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +6s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,43,200,52,209,89,254,255,87,11,0,0,0,129] +4s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,43,200,52,209,89,254,255,87,11,0,0,0,129] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [+�4�Y��W
                                                                                    ] - 129 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":51243,"extaddr":"0x000b57fffe59d134","parentaddr":0} +10s
  zigbee-herdsman:controller:log Device '0x000b57fffe59d134' joined +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:controller:log Device '0x000b57fffe59d134' accepted by handler +1ms
  zigbee-herdsman:controller:log Not interviewing '0x000b57fffe59d134', completed 'false', in progress 'true' +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,43,200,43,200,52,209,89,254,255,87,11,0,128,232] +44ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,43,200,43,200,52,209,89,254,255,87,11,0,128,232] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [+�+�4�Y��W
                                                                                      �] - 232 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":51243,"nwkaddr":51243,"ieeeaddr":"0x000b57fffe59d134","capabilities":128} +47ms
  zigbee-herdsman:controller:log Device announce '0x000b57fffe59d134' +46ms
[... cut ...]
  zigbee-herdsman:controller:device     at Timer.processTimers (timers.js:223:10)' +10s
  zigbee-herdsman:controller:error Interview failed for '0x000b57fffe59d134 with error 'Error: AREQ - ZDO - nodeDescRsp after 10000ms' +0ms
zigbee2mqtt:error 2019-11-16T11:25:03: Failed to interview '0x000b57fffe59d134', device has not successfully been paired
zigbee2mqtt:info  2019-11-16T11:25:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x000b57fffe59d134"}}'
```

Koenkk, 2019-11-17:
As I dont have this device myself I need some help of you.

Can you do the following for both Zigbee2mqtt 1.6 and 1.7.1 when pairing the device:
- Run with `DEBUG=* npm start`
- Sniff the traffic https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

martonbtoth, 2019-11-22:
I have attached the logs you requested.
Also I was able to pair the lock with 1.7.1 once and I could also record that :)

[danalock-pairing-logs.zip](https://github.com/Koenkk/zigbee2mqtt/files/3881282/danalock-pairing-logs.zip)

nanosonde, 2019-11-23:
@martonbtoth 
Sorry, the content in the ZIP file is not of much use as you have not startet Z2M as @Koenkk requested: `DEBUG=* npm start`. At least the log files do not contain any DEBUG level messages.

Furthermore, the archive does not include any captured Zigbee packets from Wireshark for example.
You need sniffing HW for this: e.g. another CC2531 stick.

martonbtoth, 2019-11-23:
Ahh sorry, I zipped the wrong folder. I'm going to upload the captured packets and the correct logs tomorrow when I get home.

martonbtoth, 2019-11-25:
[danalock-zigbee2mqtt.zip](https://github.com/Koenkk/zigbee2mqtt/files/3887635/danalock-zigbee2mqtt.zip)
This should do it, thanks for taking a look guys! :+1:

Koenkk, 2019-11-26:
Could you try replacing the content of `node_modules/zigbee-herdsman/dist/controller/model/devices.js` with https://gist.github.com/Koenkk/3aeb003a91415f2b65049a1b59aec3e5 and pair the device again.

Please provide the log when running with `DEBUG=zigbee-herdsman:controller* npm start`

martonbtoth, 2019-11-26:
I just had to RMA my Danalock today (its motor has failed 🙄).
@nanosonde could you please take a look?

nanosonde, 2019-11-27:
Ok, it worked. When hitting the pairing button first time, the device sent a "leave network" message.
After pressing it again, it successfully paired.
However, I saw "node descriptor request" failed two times before it was able to interview the device properly.

```
> zigbee2mqtt@1.7.1 start /opt/zigbee2mqtt-1.7.1
> node index.js

zigbee2mqtt:info  2019-11-27T09:53:53: Logging to directory: '/opt/zigbee2mqtt-1.7.1/data/log/2019-11-27.10-53-52'
zigbee2mqtt:debug 2019-11-27T09:53:53: Loaded state from file /opt/zigbee2mqtt-1.7.1/data/state.json
zigbee2mqtt:info  2019-11-27T09:53:53: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
zigbee2mqtt:info  2019-11-27T09:53:53: Starting zigbee-herdsman...
zigbee2mqtt:debug 2019-11-27T09:53:53: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt-1.7.1/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt-1.7.1/data/database.db.backup","backupPath":"/opt/zigbee2mqtt-1.7.1/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt-1.7.1/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt-1.7.1/data/database.db.backup","backupPath":"/opt/zigbee2mqtt-1.7.1/data/coordinator_backup.json"}' +0ms
  zigbee-herdsman:controller:log Started with result 'resumed' +1s
  zigbee-herdsman:controller:log Injected database: true, adapter: true +0ms
  zigbee-herdsman:controller:log No coordinator in database, querying... +2ms
zigbee2mqtt:info  2019-11-27T09:53:55: zigbee-herdsman started
zigbee2mqtt:info  2019-11-27T09:53:55: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
zigbee2mqtt:debug 2019-11-27T09:53:55: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
zigbee2mqtt:info  2019-11-27T09:53:55: Currently 0 devices are joined:
zigbee2mqtt:warn  2019-11-27T09:53:55: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2019-11-27T09:53:55: Allowing new devices to join.
zigbee2mqtt:warn  2019-11-27T09:53:55: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2019-11-27T09:53:55: Zigbee: allowing new devices to join.
  zigbee-herdsman:controller:log Permit joining +187ms
zigbee2mqtt:info  2019-11-27T09:53:55: Connecting to MQTT server at mqtt://192.168.10.2
zigbee2mqtt:info  2019-11-27T09:53:55: Connected to MQTT server
zigbee2mqtt:info  2019-11-27T09:53:55: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2019-11-27T09:53:55: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.7.1","commit":"b459c35","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
  zigbee-herdsman:controller:log Device leave '0x000b57fffe59d134' +34s
zigbee2mqtt:warn  2019-11-27T09:54:28: Device '0x000b57fffe59d134' left the network
zigbee2mqtt:info  2019-11-27T09:54:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x000b57fffe59d134"}}'
  zigbee-herdsman:controller:log Device '0x000b57fffe59d134' joined +29s
  zigbee-herdsman:controller:log Device '0x000b57fffe59d134' accepted by handler +2ms
  zigbee-herdsman:controller:log New device '0x000b57fffe59d134' joined +1ms
  zigbee-herdsman:controller:log Creating device '0x000b57fffe59d134' +0ms
zigbee2mqtt:info  2019-11-27T09:54:58: Device '0x000b57fffe59d134' joined
zigbee2mqtt:info  2019-11-27T09:54:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000b57fffe59d134"}}'
  zigbee-herdsman:controller:log Interview '0x000b57fffe59d134' start +52ms
zigbee2mqtt:info  2019-11-27T09:54:58: Starting interview of '0x000b57fffe59d134'
zigbee2mqtt:info  2019-11-27T09:54:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000b57fffe59d134"}}'
  zigbee-herdsman:controller:device Interview - start device '0x000b57fffe59d134' +0ms
  zigbee-herdsman:controller:log Device '0x000b57fffe59d134' joined +9s
  zigbee-herdsman:controller:log Device '0x000b57fffe59d134' accepted by handler +1ms
  zigbee-herdsman:controller:log Not interviewing '0x000b57fffe59d134', completed 'false', in progress 'true' +1ms
  zigbee-herdsman:controller:device Interview - 1 node descriptor request failed for '0x000b57fffe59d134', retrying... +10s
  zigbee-herdsman:controller:log Device '0x000b57fffe59d134' joined +4s
  zigbee-herdsman:controller:log Device '0x000b57fffe59d134' accepted by handler +1ms
  zigbee-herdsman:controller:log Not interviewing '0x000b57fffe59d134', completed 'false', in progress 'true' +0ms
  zigbee-herdsman:controller:log Device announce '0x000b57fffe59d134' +251ms
zigbee2mqtt:debug 2019-11-27T09:55:11: Device '0x000b57fffe59d134' announced itself
[... cut ...] atteryPercentThres1":{"ID":59,"type":32,"name":"batteryPercentThres1"},"batteryPercentThres2":{"ID":60,"type":32,"name":"batteryPercentThres2"},"batteryPercentThres3":{"ID":61,"type":32,"name":"batteryPercentThres3"},"batteryAlarmState":{"ID":62,"type":27,"name":"batteryAlarmState"}},"name":"genPowerCfg","commands":{},"commandsResponse":{}}},"networkAddress":6263,"endpoint":1,"linkquality":44,"groupID":0}' +2s
zigbee2mqtt:info  2019-11-27T09:55:24: Succesfully configured '0x000b57fffe59d134'
```

nanosonde, 2019-11-27:
It is paired, but I do not get any messages when I lock/unlock the danalock manually as it used to be with in older versions. Could it be that the reporting feature is not working properly anymore?

I have set "reporting" to true.

Koenkk, 2019-11-27:
@nanosonde I think this is a different issue in zigbee2mqtt 1.7.1 which should have been fixed in the latest dev branch. The latest dev branch also contains the changes needed to successfully interview this device. So the changes to `devices.js` I provided are not needed anymore. For now:

- Update to latest dev
- Remove the device from the network. (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove)
- Pair the device

nanosonde, 2019-11-27:
@Koenkk I did the steps in the order you described.

```
Nov 27 20:30:25 rpi3-knx systemd[1]: Started zigbee2mqtt.
Nov 27 20:30:26 rpi3-knx npm[12000]: > zigbee2mqtt@1.7.1 start /opt/zigbee2mqtt-1.7.1
Nov 27 20:30:26 rpi3-knx npm[12000]: > node index.js
Nov 27 20:30:28 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:30:28: Logging to directory: '/opt/zigbee2mqtt-1.7.1/data/log/2019-11-27.20-30-28'
Nov 27 20:30:28 rpi3-knx npm[12000]: zigbee2mqtt:debug 2019-11-27 20:30:28: Loaded state from file /opt/zigbee2mqtt-1.7.1/data/state.json
Nov 27 20:30:28 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:30:28: Starting zigbee2mqtt version 1.7.1+dev (commit #c2be957)
Nov 27 20:30:28 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:30:28: Starting zigbee-herdsman...
Nov 27 20:30:28 rpi3-knx npm[12000]: zigbee2mqtt:debug 2019-11-27 20:30:28: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt-1.7.1/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt-1.7.1/data/database.db.backup","backupPath":"/opt/zigbee2mqtt-1.7.1/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
Nov 27 20:30:30 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:30:30: zigbee-herdsman started
Nov 27 20:30:30 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:30:30: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
Nov 27 20:30:30 rpi3-knx npm[12000]: zigbee2mqtt:debug 2019-11-27 20:30:30: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
Nov 27 20:30:30 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:30:30: Currently 0 devices are joined:
Nov 27 20:30:30 rpi3-knx npm[12000]: zigbee2mqtt:warn  2019-11-27 20:30:30: `permit_join` set to  `true` in configuration.yaml.
Nov 27 20:30:30 rpi3-knx npm[12000]: zigbee2mqtt:warn  2019-11-27 20:30:30: Allowing new devices to join.
Nov 27 20:30:30 rpi3-knx npm[12000]: zigbee2mqtt:warn  2019-11-27 20:30:30: Set `permit_join` to `false` once you joined all devices.
Nov 27 20:30:30 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:30:30: Zigbee: allowing new devices to join.
Nov 27 20:30:30 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:30:30: Connecting to MQTT server at mqtt://192.168.10.2
Nov 27 20:30:30 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:30:30: Connected to MQTT server
Nov 27 20:30:30 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:30:30: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Nov 27 20:30:30 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:30:30: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.7.1+dev","commit":"c2be957","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
Nov 27 20:31:00 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:31:00: Device 'danalock' joined
Nov 27 20:31:00 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:31:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"danalock"}}'
Nov 27 20:31:00 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:31:00: Starting interview of 'danalock'
Nov 27 20:31:00 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:31:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"danalock"}}'
Nov 27 20:31:13 rpi3-knx npm[12000]: zigbee2mqtt:debug 2019-11-27 20:31:13: Device 'danalock' announced itself
Nov 27 20:31:23 rpi3-knx npm[12000]: zigbee2mqtt:debug 2019-11-27 20:31:23: Received Zigbee message from 'danalock', type 'readResponse', cluster 'genBasic', data '{"modelId":"V3-BTZB","manufacturerName":"Danalock","powerSource":3}' from endpoint 1 with groupID 0
Nov 27 20:31:23 rpi3-knx npm[12000]: zigbee2mqtt:debug 2019-11-27 20:31:23: No converter available for 'V3-BTZB' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"V3-BTZB","manufacturerName":"Danalock","powerSource":3}'
Nov 27 20:31:24 rpi3-knx npm[12000]: zigbee2mqtt:debug 2019-11-27 20:31:24: Received Zigbee message from 'danalock', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2,"appVersion":1,"stackVersion":1}' from endpoint 1 with groupID 0
Nov 27 20:31:24 rpi3-knx npm[12000]: zigbee2mqtt:debug 2019-11-27 20:31:24: No converter available for 'V3-BTZB' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":2,"appVersion":1,"stackVersion":1}'
Nov 27 20:31:25 rpi3-knx npm[12000]: zigbee2mqtt:debug 2019-11-27 20:31:25: Received Zigbee message from 'danalock', type 'readResponse', cluster 'genBasic', data '{"hwVersion":4,"dateCode":"19042019\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}' from endpoint 1 with groupID 0
Nov 27 20:31:25 rpi3-knx npm[12000]: zigbee2mqtt:debug 2019-11-27 20:31:25: No converter available for 'V3-BTZB' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":4,"dateCode":"19042019\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}'
Nov 27 20:31:25 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:31:25: Successfully interviewed 'danalock', device has successfully been paired
[... cut ...]
Nov 27 20:31:25 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:31:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"danalock","model":"V3-BTZB","vendor":"Danalock","description":"BT/ZB smartlock","supported":true}}'
Nov 27 20:31:25 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:31:25: Configuring 'danalock'
Nov 27 20:31:27 rpi3-knx npm[12000]: zigbee2mqtt:info  2019-11-27 20:31:27: Succesfully configured 'danalock'
```

It paired successfully, but I do not receive any LOCK/UNLOCK status updates.

Koenkk, 2019-11-27:
Strange, could you sniff the traffic while turning on/off the lock?  (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

nanosonde, 2019-11-27:
[danalock_pairing_configure_reporting_lock_unlock.zip](https://github.com/Koenkk/zigbee2mqtt/files/3898637/danalock_pairing_configure_reporting_lock_unlock.zip)

@Koenkk 
From the traffic I can see that configure reporting was successful.
But when locking/unlocking the lock, it seems that it wants to send data ("data request"), but nobody cares about it.

The capture file also contains the complete pairing.

Koenkk, 2019-11-28:
Thanks, I cannot find the problem yet, could you provide the same log when pairing it with zigbee2mqtt 1.6 (also lock/unlock the lock after this). In this way we can compare.

jysaloma, 2019-11-28:
I do have a similar problems with the lock. I was able to pair the device on second try.

1st try:
`debug 2019-11-27 23:00:05: Received Zigbee message from '0x000b57fffe59d827', type 'readResponse', cluster 'genBasic', data '{"modelId":"V3-BTZB","manufacturerName":"Danalock","powerSource":3}' from endpoint 1 with groupID 0
debug 2019-11-27sa 23:00:05: No converter available for 'V3-BTZB' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"V3-BTZB","manufacturerName":"Danalock","powerSource":3}'
info  2019-11-27 23:00:05: MQTT publish: topic 'homeassistant/lock/0x000b57fffe59d827/lock/config', payload '{"command_topic":"zigbee2mqtt/0x000b57fffe59d827/set","value_template":"{{ value_json.state }}","state_topic":"zigbee2mqtt/0x000b57fffe59d827","json_attributes_topic":"zigbee2mqtt/0x000b57fffe59d827","name":"0x000b57fffe59d827_lock","unique_id":"0x000b57fffe59d827_lock_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000b57fffe59d827"],"name":"0x000b57fffe59d827","sw_version":"Zigbee2mqtt 1.7.1+dev","model":"BT/ZB smartlock (V3-BTZB)","manufacturer":"Danalock"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2019-11-27 23:00:05: MQTT publish: topic 'homeassistant/sensor/0x000b57fffe59d827/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x000b57fffe59d827","json_attributes_topic":"zigbee2mqtt/0x000b57fffe59d827","name":"0x000b57fffe59d827_linkquality","unique_id":"0x000b57fffe59d827_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000b57fffe59d827"],"name":"0x000b57fffe59d827","sw_version":"Zigbee2mqtt 1.7.1+dev","model":"BT/ZB smartlock (V3-BTZB)","manufacturer":"Danalock"},"availability_topic":"zigbee2mqtt/bridge/state"}'
debug 2019-11-27 23:00:05: Received Zigbee message from '0x000b57fffe59d827', type 'readResponse', cluster 'genBasic', data '{"modelId":"V3-BTZB","manufacturerName":"Danalock","powerSource":3}' from endpoint 1 with groupID 0
debug 2019-11-27 23:00:05: No converter available for 'V3-BTZB' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"V3-BTZB","manufacturerName":"Danalock","powerSource":3}'
debug 2019-11-27 23:00:06: Received Zigbee message from '0x000b57fffe59d827', type 'readResponse', cluster 'genBasic', data '{"modelId":"V3-BTZB","manufacturerName":"Danalock","powerSource":3}' from endpoint 1 with groupID 0
debug 2019-11-27 23:00:06: No converter available for 'V3-BTZB' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"V3-BTZB","manufacturerName":"Danalock","powerSource":3}'
error 2019-11-27 23:00:06: Failed to interview '0x000b57fffe59d827', device has not successfully been paired
info  2019-11-27 23:00:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x000b57fffe59d827"}}'
info  2019-11-27 23:00:06: Configuring '0x000b57fffe59d827'
error 2019-11-27 23:00:07: Failed to configure '0x000b57fffe59d827', attempt 1 (Error: Data request failed with error: 'MAC transaction expired' (240)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:571:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:5:58))`

2nd try:
`info  2019-11-27 23:26:27: Device '0x000b57fffe59d827' joined
info  2019-11-27 23:26:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000b57fffe59d827"}}'
info  2019-11-27 23:26:27: Starting interview of '0x000b57fffe59d827'
info  2019-11-27 23:26:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000b57fffe59d827"}}'
debug 2019-11-27 23:26:40: Device '0x000b57fffe59d827' announced itself
debug 2019-11-27 23:26:49: Received Zigbee message from '0x000b57fffe59d827', type 'readResponse', cluster 'genBasic', data '{"modelId":"V3-BTZB","manufacturerName":"Danalock","powerSource":3}' from endpoint 1 with groupID 0
debug 2019-11-27 23:26:49: No converter available for 'V3-BTZB' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"V3-BTZB","manufacturerName":"Danalock","powerSource":3}'
debug 2019-11-27 23:26:51: Received Zigbee message from '0x000b57fffe59d827', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2,"appVersion":1,"stackVersion":1}' from endpoint 1 with groupID 0
debug 2019-11-27 23:26:51: No converter available for 'V3-BTZB' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":2,"appVersion":1,"stackVersion":1}'
debug 2019-11-27 23:26:51: Received Zigbee message from '0x000b57fffe59d827', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2,"appVersion":1,"stackVersion":1}' from endpoint 1 with groupID 0
debug 2019-11-27 23:26:51: No converter available for 'V3-BTZB' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":2,"appVersion":1,"stackVersion":1}'
debug 2019-11-27 23:26:52: Received Zigbee message from '0x000b57fffe59d827', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2,"appVersion":1,"stackVersion":1}' from endpoint 1 with groupID 0
debug 2019-11-27 23:26:52: No converter available for 'V3-BTZB' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":2,"appVersion":1,"stackVersion":1}'
[... cut ...]
zigbee2mqtt:error 2019-11-28 23:41:57: Publish 'set' 'state' to '0x000b57fffe59d827' failed: 'Error: Timeout - 18250 - 1 - 53 - 257 - 11 after 15000ms'
zigbee2mqtt:info  2019-11-28 23:41:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x000b57fffe59d827' failed: 'Error: Timeout - 18250 - 1 - 53 - 257 - 11 after 15000ms'","meta":{"friendly_name":"0x000b57fffe59d827"}}'
zigbee2mqtt:debug 2019-11-28 23:42:51: Saving state to file /share/zigbee2mqtt/state.json`

[image]

nanosonde, 2019-12-02:
@Koenkk I did not find the time so far for capturing the Zigbee traffic.

However, I found this [document](https://forum.iobroker.net/assets/uploads/files/1553118067506-td003zigbeemanual.pdf).

On page 3 it says: "Reporting The lock supports reporting of attributes when the recipient is in the binding table."

Do I explicitly bind it to the coordinator?

jysaloma, 2019-12-03:
@Koenkk that above issue that I had with timeouts seems to be fixed by changing the similar [line](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L4950) than you suggested [here](https://github.com/Koenkk/zigbee2mqtt/issues/2456#issuecomment-561308245) for Danalock.

After the change I could see the messages fine in the log but the lock didn't turn so I had to also remove the batteries from the lock to make it functioning.

here is the latest log for working lock:

debug 2019-12-03 23:59:41: Received MQTT message on 'zigbee2mqtt/0x000b57fffe59d827/set' with data 'UNLOCK'
debug 2019-12-03 23:59:41: Publishing 'set' 'state' to '0x000b57fffe59d827'
info  2019-12-03 23:59:45: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe59d827', payload '{"state":"UNLOCK"}'
debug 2019-12-03 23:59:47: Received MQTT message on 'zigbee2mqtt/0x000b57fffe59d827/set' with data 'LOCK'
debug 2019-12-03 23:59:47: Publishing 'set' 'state' to '0x000b57fffe59d827'
info  2019-12-03 23:59:49: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe59d827', payload '{"state":"LOCK"}'
debug 2019-12-04 00:00:00: Received MQTT message on 'zigbee2mqtt/bridge/config/devices/get' with data ''
info  2019-12-04 00:00:00: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"ieeeAddr":"0x00124b0014b885a6","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1575410400305},{"ieeeAddr":"0x00158d00041d0f0f","type":"EndDevice","networkAddress":63512,"model":"MCCGQ11LM","friendly_name":"0x00158d00041d0f0f","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_magnet.aq2","hardwareVersion":2,"softwareBuildID":"3000-0001","dateCode":"20161128","lastSeen":null},{"ieeeAddr":"0x00158d00042340a1","type":"EndDevice","networkAddress":46188,"model":"MCCGQ11LM","friendly_name":"0x00158d00042340a1","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_magnet.aq2","hardwareVersion":2,"softwareBuildID":"3000-0001","dateCode":"20161128","lastSeen":null},{"ieeeAddr":"0x000b57fffe59d827","type":"EndDevice","networkAddress":18250,"model":"V3-BTZB","friendly_name":"0x000b57fffe59d827","manufacturerID":4444,"manufacturerName":"Danalock","powerSource":"Battery","modelID":"V3-BTZB","hardwareVersion":4,"dateCode":"03092018\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","lastSeen":1575410389098}]'
debug 2019-12-04 00:00:00: Received MQTT message on 'zigbee2mqtt/bridge/config/devices' with data '[{"ieeeAddr":"0x00124b0014b885a6","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1575410400305},{"ieeeAddr":"0x00158d00041d0f0f","type":"EndDevice","networkAddress":63512,"model":"MCCGQ11LM","friendly_name":"0x00158d00041d0f0f","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_magnet.aq2","hardwareVersion":2,"softwareBuildID":"3000-0001","dateCode":"20161128","lastSeen":null},{"ieeeAddr":"0x00158d00042340a1","type":"EndDevice","networkAddress":46188,"model":"MCCGQ11LM","friendly_name":"0x00158d00042340a1","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_magnet.aq2","hardwareVersion":2,"softwareBuildID":"3000-0001","dateCode":"20161128","lastSeen":null},{"ieeeAddr":"0x000b57fffe59d827","type":"EndDevice","networkAddress":18250,"model":"V3-BTZB","friendly_name":"0x000b57fffe59d827","manufacturerID":4444,"manufacturerName":"Danalock","powerSource":"Battery","modelID":"V3-BTZB","hardwareVersion":4,"dateCode":"03092018\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","lastSeen":1575410389098}]'
info  2019-12-04 00:00:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b0014b885a6","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1575410400328},{"ieeeAddr":"0x00158d00041d0f0f","type":"EndDevice","networkAddress":63512,"model":"MCCGQ11LM","friendly_name":"0x00158d00041d0f0f","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_magnet.aq2","hardwareVersion":2,"softwareBuildID":"3000-0001","dateCode":"20161128","lastSeen":null},{"ieeeAddr":"0x00158d00042340a1","type":"EndDevice","networkAddress":46188,"model":"MCCGQ11LM","friendly_name":"0x00158d00042340a1","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_magnet.aq2","hardwareVersion":2,"softwareBuildID":"3000-0001","dateCode":"20161128","lastSeen":null},{"ieeeAddr":"0x000b57fffe59d827","type":"EndDevice","networkAddress":18250,"model":"V3-BTZB","friendly_name":"0x000b57fffe59d827","manufacturerID":4444,"manufacturerName":"Danalock","powerSource":"Battery","modelID":"V3-BTZB","hardwareVersion":4,"dateCode":"03092018\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","lastSeen":1575410389098}]}'
debug 2019-12-04 00:00:00: Received MQTT message on 'zigbee2mqtt/bridge/networkmap' with data 'raw'
info  2019-12-04 00:00:00: Starting network scan (includeRoutes 'false')
debug 2019-12-04 00:00:00: LQI succeeded for 'Coordinator'
info  2019-12-04 00:00:00: Network scan finished
[... cut ...]
debug 2019-12-04 00:00:33: Publishing 'set' 'state' to '0x000b57fffe59d827'
info  2019-12-04 00:00:39: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe59d827', payload '{"state":"UNLOCK"}'
debug 2019-12-04 00:02:23: Received MQTT message on 'zigbee2mqtt/0x000b57fffe59d827/set' with data 'UNLOCK'
debug 2019-12-04 00:02:23: Publishing 'set' 'state' to '0x000b57fffe59d827'
info  2019-12-04 00:02:25: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe59d827', payload '{"state":"UNLOCK"}'

Is that line that I changed something that should be added to code too for Danalock?

Koenkk, 2019-12-04:
@nanosonde nice catch, it was indeed not bound, please try with:

```js
{
    zigbeeModel: ['V3-BTZB'],
    model: 'V3-BTZB',
    vendor: 'Danalock',
    description: 'BT/ZB smartlock',
    supports: 'lock/unlock, battery',
    fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery_200],
    toZigbee: [tz.generic_lock],
    meta: {options: {disableDefaultResponse: true}, configureKey: 5},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
        await configureReporting.lockState(endpoint);
        await configureReporting.batteryPercentageRemaining(endpoint);
    },
},
```

@jysaloma I've added it to the section above

Let me know if this definition work properly then I will update it.

nanosonde, 2019-12-05:
> @nanosonde nice catch, it was indeed not bound, please try with:
>
> ```js
> {
>     zigbeeModel: ['V3-BTZB'],
>     model: 'V3-BTZB',
>     vendor: 'Danalock',
>     description: 'BT/ZB smartlock', [...]

Yes! That was the reason! 
It is working now as expected. :-)

Koenkk, 2019-12-06:
@nanosonde so with that config (https://github.com/Koenkk/zigbee2mqtt/issues/2339#issuecomment-561812531) the device full works?

nanosonde, 2019-12-06:
> @nanosonde so with that config ([#2339 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/2339#issuecomment-561812531)) the device fully works?

Yes, as far as I can tell.

Koenkk, 2019-12-07:
Thanks, will be available in next zigbee2mqtt release.

closed 2019-12-07 by Koenkk