Koenkk/zigbee2mqtt #2341: Timezone Error in 1.7.1

jasimancas, 2019-11-15:
Hi, i have a problem with datetime in docker, Im living in Europe/Madrid, But the time zone of the container shows one hour less.

Bug? Bad config?

I have: "-TZ=Europe/Madrid

I have tried with /etc/locatime:etc/localtime:ro and nothing...

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: latest sourcerouting

Koenkk, 2019-11-15:
Should be fixed in latest dev! thanks

iStitch07, 2019-11-23:
Hello. 
I can't to set log timestamp to local time too
Have in docker-compose.yml
   environment:
      - TZ=Europe/Moscow
    volumes:
      - ./zigbee2mqtt:/app/data
      - /etc/localtime:/etc/localtime:ro
      - /etc/timezone:/etc/timezone:ro

and if I attach console, command data show me correct my local time. But in log I anyway have timestamp GMT as I think

Koenkk, 2019-11-25:
@iStitch07 I cannot reproduce this with the latest dev, can you?

iStitch07, 2019-11-25:
@Koenkk latest-dev work perfect in this plane
zigbee2mqtt:info  2019-11-25 21:54:38: Logging to directory: '/app/data/log/2019-11-25.21-54-38'
zigbee2mqtt:info  2019-11-25 21:54:39: Starting zigbee2mqtt version 1.7.1+dev (commit #7d1635a)
zigbee2mqtt:info  2019-11-25 21:54:39: Starting zigbee-herdsman...

I have this error on just latest, sorry )

Koenkk, 2019-11-26:
Cool, thanks

closed 2019-11-26 by Koenkk