Koenkk/zigbee2mqtt #2342: Failing trying to add unsupported device EcoDim Eco.Dim-007 Zigbee & Z-wave dimmer

iFloris, 2019-11-15:
# Bug Report
## What happened
I am trying to add an unsupported device, the [EcoDim Eco.Dim-007 Zigbee & Z-wave dimmer](https://www.ecodim.nl/eco-dim07-zigbee-z-wave-smart-led-dimmer.html) to z2m on Home Assistant (HASSIO in linux).
Following the [how-to](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html), I added the device to `node_modules/zigbee-herdsman-converters/devices.js`

## What did you expect to happen
I expected the z2m log to tell me about the device capabilities as described in the how-to, but the device is never mentioned in the log after adding the device. It does show up in the z2m network map like this:
`"0xccccccfffe3a1537" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{0xccccccfffe3a1537|0xccccccfffe3a1537 (843)|EcoDim EcoDim Eco.Dim-007 Zigbee & Z-wave dimmer  (Dimmer-Switch-ZB3.0)|unknown}"];`

## How to reproduce it (minimal and precise)
This is what I added to devices.js
`{
    zigbeeModel: ['Dimmer-Switch-ZB3.0'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'Dimmer-Switch-ZB3.0', // Vendor model number, look on the device for a model number
    vendor: 'EcoDim B.V.', // Vendor of the device (only used for documentation and startup logging)
    description: 'EcoDim Eco.Dim-007 Zigbee & Z-wave dimmer ', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'Dimming lights', // Actions this device supports (only used for documentation)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
},`

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: CC2531_SOURCE_ROUTING_20190619

When first joining, the device was detected like this:
`zigbee2mqtt:info  2019-11-15T12:37:00: Device '0xccccccfffe3a1537' joined
zigbee2mqtt:info  2019-11-15T12:37:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0xccccccfffe3a1537"}}'
zigbee2mqtt:info  2019-11-15T12:37:00: Starting interview of '0xccccccfffe3a1537'
zigbee2mqtt:info  2019-11-15T12:37:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xccccccfffe3a1537"}}'
zigbee2mqtt:warn  2019-11-15T12:37:10: Received message from unsupported device with Zigbee model 'Dimmer-Switch-ZB3.0'
zigbee2mqtt:warn  2019-11-15T12:37:10: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2019-11-15T12:37:10: Received message from unsupported device with Zigbee model 'Dimmer-Switch-ZB3.0'
zigbee2mqtt:warn  2019-11-15T12:37:10: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2019-11-15T12:37:10: Received message from unsupported device with Zigbee model 'Dimmer-Switch-ZB3.0'
zigbee2mqtt:warn  2019-11-15T12:37:10: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:info  2019-11-15T12:37:10: Successfully interviewed '0xccccccfffe3a1537', device has successfully been paired
zigbee2mqtt:warn  2019-11-15T12:37:10: Device '0xccccccfffe3a1537' with Zigbee model 'Dimmer-Switch-ZB3.0' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt:info  2019-11-15T12:37:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xccccccfffe3a1537","supported":false}}'`

Koenkk, 2019-11-15:
Probably 
```js
{
    zigbeeModel: ['Dimmer-Switch-ZB3.0'],
    model: 'Dimmer-Switch-ZB3.0',
    vendor: 'EcoDim B.V.',
    description: 'EcoDim Eco.Dim-007 Zigbee & Z-wave dimmer ',
    extend: generic.light_onoff_brightness,
},
``` 
is enough to make it work.

iFloris, 2019-11-15:
thanks a lot @Koenkk I will try it tomorrow!

iFloris, 2019-11-16:
@Koenkk yes, that was what was needed! Thank you very much.
The dimmer now shows up and I am able to control it. However, manual controlling the physical dimmer does not yet show up in HA. For example, turning the dimmer off does not show as off in HA. Also link quality is not reported. If you do not mind hand-holding me a bit more: How should I go about finding out how the dimmer reports its status? 

This is how z2m sees the dimmer in HA now:
```
zigbee2mqtt:info  2019-11-16T10:33:48: MQTT publish: topic 'homeassistant/light/0xccccccfffe3a1537/light/config', payload '{"brightness":true,"schema":"json","command_topic":"zigbee2mqtt/0xccccccfffe3a1537/set","state_topic":"zigbee2mqtt/0xccccccfffe3a1537","name":"0xccccccfffe3a1537_light","unique_id":"0xccccccfffe3a1537_light_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0xccccccfffe3a1537"],"name":"0xccccccfffe3a1537","sw_version":"Zigbee2mqtt 1.7.1","model":"EcoDim Eco.Dim-007 Zigbee & Z-wave dimmer  (Dimmer-Switch-ZB3.0)","manufacturer":"EcoDim B.V."},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-11-16T10:33:48: MQTT publish: topic 'homeassistant/sensor/0xccccccfffe3a1537/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0xccccccfffe3a1537","json_attributes_topic":"zigbee2mqtt/0xccccccfffe3a1537","name":"0xccccccfffe3a1537_linkquality","unique_id":"0xccccccfffe3a1537_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0xccccccfffe3a1537"],"name":"0xccccccfffe3a1537","sw_version":"Zigbee2mqtt 1.7.1","model":"EcoDim Eco.Dim-007 Zigbee & Z-wave dimmer  (Dimmer-Switch-ZB3.0)","manufacturer":"EcoDim B.V."},"availability_topic":"zigbee2mqtt/bridge/state"}'
```

Koenkk, 2019-11-16:
Try:

```js
{
    zigbeeModel: ['Dimmer-Switch-ZB3.0'],
    model: 'Dimmer-Switch-ZB3.0',
    vendor: 'EcoDim B.V.',
    description: 'EcoDim Eco.Dim-007 Zigbee & Z-wave dimmer ',
    extend: generic.light_onoff_brightness,
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await configureReporting.onOff(endpoint);
        await configureReporting.brightness(endpoint);
    },
},
```

iFloris, 2019-11-16:
@Koenkk yes, that did it, the dimmer now completely works. It is quite satisfying to see the dimmer slider move as the physical button is turned. How should I go about adding this to the supported devices here? 

Again, I really appreciate the help. Your Z2M is such an amazing tool!

Koenkk, 2019-11-17:
Thanks, it will be supported in the zigbee2mqtt dev branch in a few hours.

closed 2019-11-17 by Koenkk

Miesvanderlippe, 2021-01-24:
@Koenkk As far as I can tell, the basic version of the dimmer has pretty much the same functionality but it isn't recognized by HASS. Could you add support for it without the above logs? I'm having a hard time getting the buttons off. 

https://www.ecobright.nl/led-dimmer/zigbee-led-dimmer-draai-0-200w-universeel-basic/

MDSDM, 2021-02-01:
Hi,

A bit weird, but all my ECODIM 10 devices are identified as ECODIM 07.
They are similar but the 07 is a knob based dimmer and the 10 is a push based dimmer.

Practically, I can control the 10 from ZigBee2MQTT - however the physical input is not registered (not in debug log what I can see).

Is there a way I can avoid the Z2M hi-hacking the 10 as an 07?