Koenkk/zigbee2mqtt #2343: No messages from Tradfi Dimmer E1743

vzorglub, 2019-11-15:
I successfully pair the device but there are some messages in the log saying that there are no converters available

The device is paired and appears in the configuration yaml

The log:
```
info  2019-11-15T16:02:35: Device '0x14b457fffed4bb1c' joined
info  2019-11-15T16:02:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x14b457fffed4bb1c"}}'
info  2019-11-15T16:02:35: Starting interview of '0x14b457fffed4bb1c'
info  2019-11-15T16:02:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x14b457fffed4bb1c"}}'
debug 2019-11-15T16:02:37: Device '0x14b457fffed4bb1c' announced itself
debug 2019-11-15T16:02:53: Received Zigbee message from '0x14b457fffed4bb1c', type 'readResponse', cluster 'genBasic', data '{"modelId":"TRADFRI on/off switch","manufacturerName":"IKEA of Sweden","powerSource":3}' from endpoint 1 with groupID 0
debug 2019-11-15T16:02:53: No converter available for 'E1743' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"TRADFRI on/off switch","manufacturerName":"IKEA of Sweden","powerSource":3}'
debug 2019-11-15T16:02:55: Received Zigbee message from '0x14b457fffed4bb1c', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3,"appVersion":34,"stackVersion":98}' from endpoint 1 with groupID 0
debug 2019-11-15T16:02:55: No converter available for 'E1743' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3,"appVersion":34,"stackVersion":98}'
debug 2019-11-15T16:02:57: Received Zigbee message from '0x14b457fffed4bb1c', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20190410","swBuildId":"2.2.008"}' from endpoint 1 with groupID 0
debug 2019-11-15T16:02:57: No converter available for 'E1743' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1,"dateCode":"20190410","swBuildId":"2.2.008"}'
info  2019-11-15T16:02:57: Successfully interviewed '0x14b457fffed4bb1c', device has successfully been paired
info  2019-11-15T16:02:57: Device '0x14b457fffed4bb1c' is supported, identified as: IKEA TRADFRI ON/OFF switch (E1743)
info  2019-11-15T16:02:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x14b457fffed4bb1c","model":"E1743","vendor":"IKEA","description":"TRADFRI ON/OFF switch","supported":true}}'
```

The issue is that no messages are received from the device when I press the buttons ON or OFF or Long press either

Thanks

vzorglub, 2019-11-15:
Start up log:
```
info  2019-11-15T15:56:23: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-11-15.15-56-21'
info  2019-11-15T15:56:24: Starting zigbee2mqtt version 1.7.0 (commit #551a58a)
info  2019-11-15T15:56:24: Starting zigbee-herdsman...
info  2019-11-15T15:56:26: zigbee-herdsman started
info  2019-11-15T15:56:26: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel"$info  2019-11-15T15:56:26: Currently 6 devices are joined:
info  2019-11-15T15:56:26: 0x0017880103b969ed (0x0017880103b969ed): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
info  2019-11-15T15:56:26: SmallBedroomWindow (0x00158d00034d81cd): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
info  2019-11-15T15:56:26: 0x14b457fffe4efac2 (0x14b457fffe4efac2): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
info  2019-11-15T15:56:26: TVLightLeft (0x086bd7fffe292b65): LED1649C5 - IKEA TRADFRI LED bulb E12/E14/E17 400 lumen, dimmable warm white, chandelier $info  2019-11-15T15:56:26: 0x14b457fffed4bb1c (0x14b457fffed4bb1c): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
info  2019-11-15T15:56:26: TVLightRight (0x000b3cfffefcde09): LED1649C5 - IKEA TRADFRI LED bulb E12/E14/E17 400 lumen, dimmable warm white, chandelier$warn  2019-11-15T15:56:26: `permit_join` set to  `true` in configuration.yaml.
warn  2019-11-15T15:56:26: Allowing new devices to join.
warn  2019-11-15T15:56:26: Set `permit_join` to `false` once you joined all devices.
info  2019-11-15T15:56:26: Zigbee: allowing new devices to join.
info  2019-11-15T15:56:26: Connecting to MQTT server at mqtt://192.168.0.34
error 2019-11-15T15:56:26: Failed to call 'DeviceConfigure' 'onZigbeeStarted' (TypeError: Cannot read property 'friendlyName' of null
    at DeviceConfigure.configure (/opt/zigbee2mqtt/lib/extension/deviceConfigure.js:64:52)
    at DeviceConfigure.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/deviceConfigure.js:41:28)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:320:44))
info  2019-11-15T15:56:27: Connected to MQTT server
```

sjorge, 2019-11-16:
Looks like the device config never applied, as the remote tends to sleep. 

Try to restart zigbee2mqtt and while doing so... click a button on the remote. That should hopefully allow it to be configured.

Koenkk, 2019-11-18:
Please try again with latest dev. Also make sure that the device is in `configuration.yaml`

pcb1962, 2019-11-28:
I've just bought a bunch of these and can confirm that they work perfectly in the latest dev build.
Only slightly odd thing is the leading spaces on the click values:
Nov 28 22:28:28 manatee npm[17276]: #033[32mzigbee2mqtt:info #033[39m 2019-11-28 22:28:28: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe98821e', payload '{"linkquality":68,"last_seen":"2019-11-28T22:28:28.464Z","click":"             **on**","elapsed":1919}'
Nov 28 22:28:30 manatee npm[17276]: #033[32mzigbee2mqtt:info #033[39m 2019-11-28 22:28:30: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe98821e', payload '{"linkquality":70,"last_seen":"2019-11-28T22:28:30.179Z","click":"             **off**","elapsed":1714}'
Nov 28 22:28:32 manatee npm[17276]: #033[32mzigbee2mqtt:info #033[39m 2019-11-28 22:28:32: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe98821e', payload '{"linkquality":65,"last_seen":"2019-11-28T22:28:32.537Z","click":"             **brightness_down**","elapsed":2358}'
Nov 28 22:28:35 manatee npm[17276]: #033[32mzigbee2mqtt:info #033[39m 2019-11-28 22:28:35: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe98821e', payload '{"linkquality":70,"last_seen":"2019-11-28T22:28:35.309Z","click":"             **brightness_stop**","elapsed":2771}'
Nov 28 22:28:39 manatee npm[17276]: #033[32mzigbee2mqtt:info #033[39m 2019-11-28 22:28:39: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe98821e', payload '{"linkquality":70,"last_seen":"2019-11-28T22:28:39.139Z","click":"             **brightness_up**","elapsed":3829}'
Nov 28 22:28:40 manatee npm[17276]: #033[32mzigbee2mqtt:info #033[39m 2019-11-28 22:28:40: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe98821e', payload '{"linkquality":68,"last_seen":"2019-11-28T22:28:40.141Z","click":"             **brightness_stop**","elapsed":1001}'

Koenkk, 2019-11-29:
@pcb1962 thanks for confirming!

closed 2019-11-29 by Koenkk

czikus, 2019-12-05:
I have the exact same problem. I tried the remedies suggested above but to no effect. I'm running the latest dev.

```
zigbee2mqtt:info  2019-12-05T08:54:52: Device 'ikea_dimmer1' joined
zigbee2mqtt:info  2019-12-05T08:54:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"ikea_dimmer1"}}'
zigbee2mqtt:info  2019-12-05T08:54:52: Starting interview of 'ikea_dimmer1'
zigbee2mqtt:info  2019-12-05T08:54:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"ikea_dimmer1"}}'
zigbee2mqtt:debug 2019-12-05T08:54:56: Device 'ikea_dimmer1' announced itself
zigbee2mqtt:debug 2019-12-05T08:55:06: Received Zigbee message from 'ikea_dimmer1', type 'readResponse', cluster 'genBasic', data '{"modelId":"TRADFRI remote control","manufacturerName":"IKEA of Sweden","powerSource":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-12-05T08:55:06: No converter available for 'E1524/E1810' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"TRADFRI remote control","manufacturerName":"IKEA of Sweden","powerSource":3}'
zigbee2mqtt:debug 2019-12-05T08:55:08: Received Zigbee message from 'ikea_dimmer1', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1,"appVersion":32,"stackVersion":87}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-12-05T08:55:08: No converter available for 'E1524/E1810' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1,"appVersion":32,"stackVersion":87}'
zigbee2mqtt:debug 2019-12-05T08:55:10: Received Zigbee message from 'ikea_dimmer1', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20190211","swBuildId":"1.2.223"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-12-05T08:55:10: No converter available for 'E1524/E1810' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1,"dateCode":"20190211","swBuildId":"1.2.223"}'
zigbee2mqtt:info  2019-12-05T08:55:10: Successfully interviewed 'ikea_dimmer1', device has successfully been paired
zigbee2mqtt:info  2019-12-05T08:55:10: Device 'ikea_dimmer1' is supported, identified as: IKEA TRADFRI remote control (E1524/E1810)
zigbee2mqtt:info  2019-12-05T08:55:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"ikea_dimmer1","model":"E1524/E1810","vendor":"IKEA","description":"TRADFRI remote control","supported":true}}'
zigbee2mqtt:info  2019-12-05T08:55:10: Configuring 'ikea_dimmer1'
zigbee2mqtt:info  2019-12-05T08:55:14: Succesfully configured 'ikea_dimmer1'
```

After that, no messages from the dimmer.

vzorglub, 2019-12-05:
I agree with above, I get exactly the same messages as before
I cleaned the db
I updated to dev branch
The device shows as `succesfully configured` but nothing on the logs or mqtt when I press a button on the device, long press, short press.... nothing

czikus, 2019-12-05:
I just bought got yet another dimmer for testing and the result is the same. So, now two dimmers are not sending any messages after successful configuration for me.

czikus, 2019-12-07:
Can we re-open the issue? The problem is clearly not solved.

Koenkk, 2019-12-07:
@czikus did you try the latest dev branch? After that remove it and repair it.

czikus, 2019-12-08:
Like I said in my comment, I did try the dev branch as of 3 days ago. I
removed and repaired multiple times. Did anything change since 3 days ago
that would warrant trying that again?

Koenkk, 2019-12-08:
Can you share your `database.db`?

vzorglub, 2019-12-08:
This is mine 2 lights and an E1743 at id 4

```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018e1a5ff","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,110],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList"
:[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"de
vId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"pr
ofId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{
},"binds":[]},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outC
lusterList":[1280],"clusters":{},"binds":[]},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"me
ta":{}}
{"id":2,"type":"Router","ieeeAddr":"0x086bd7fffe292b65","nwkAddr":55371,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E14 W
 op/ch 400lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attr
ibutes":{"modelId":"TRADFRI bulb E14 W op/ch 400lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"201
70302","swBuildId":"1.2.214"}}},"binds":[]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214","zclVersion":1,"interviewCompleted":true,"met
a":{}}
{"id":3,"type":"Router","ieeeAddr":"0x000b3cfffefcde09","nwkAddr":41660,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E14 W
 op/ch 400lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attr
ibutes":{"modelId":"TRADFRI bulb E14 W op/ch 400lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"201
70302","swBuildId":"1.2.214"}}},"binds":[]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214","zclVersion":1,"interviewCompleted":true,"met
a":{}}
{"id":4,"type":"EndDevice","ieeeAddr":"0x14b457fffed4bb1c","nwkAddr":37310,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI on/off switch","epLi
st":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2080,"inClusterList":[0,1,3,9,32,4096,64636],"outClusterList":[3,4,6,8,25,258,4096],"clusters":{"genBasic":{"attributes":{"mod
elId":"TRADFRI on/off switch","manufacturerName":"IKEA of Sweden","powerSource":3,"zclVersion":3,"appVersion":34,"stackVersion":98,"hwVersion":1,"dateCode":"20190410","swBuildId":"2.
2.008"}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e1a5ff","endpointID":1}]}},"appVersion":34,"stackVersion":98,"hwVersion":1,"dateCode":"20190410","s
wBuildId":"2.2.008","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1}}

```

vzorglub, 2019-12-08:
Ok there is progress:
`zigbee2mqtt/0x14b457fffed4bb1c`
```json
{"battery":74,"linkquality":60}
```
Received on MQTT when I pressed the switch
But NO messages about the state on the button
On both sides O and I, short press and long press

czikus, 2019-12-09:
Here's my entry for the dimmer:

```
{"id":16,"type":"EndDevice","ieeeAddr":"0x14b457fffe3e05bd","nwkAddr":23475,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI remote control","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":2096,"inClusterList":[0,1,3,9,2821,4096],"outClusterList":[3,4,5,6,8,25,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI remote control","manufacturerName":"IKEA of Sweden","powerSource":3,"zclVersion":1,"appVersion":32,"stackVersion":87,"hwVersion":1,"dateCode":"20190211","swBuildId":"1.2.223"}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00194a15b9","endpointID":1}]}},"appVersion":32,"stackVersion":87,"hwVersion":1,"dateCode":"20190211","swBuildId":"1.2.223","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1}}
```

vzorglub, 2019-12-10:
@Koenkk 
Shouldn't this issue be re-opened?
Thanks

pcb1962, 2019-12-10:
> Ok there is progress:
> `zigbee2mqtt/0x14b457fffed4bb1c`
> 
> ```json
> {"battery":74,"linkquality":60}

Off topic, sorry, but this is interesting, as for some time I have had an Aqara button which does something similar - I see a message when I click it but the "click" field is missing from the message:
2019-12-10 21:10:42: MQTT publish: topic 'zigbee2mqtt/button/spare', payload '{"linkquality":42,"last_seen":"2019-12-10T21:10:42.262Z","battery":100,"voltage":3105,"elapsed":15620}'
I haven't seen any other report of this behaviour until now.

vzorglub, 2019-12-10:
I see a message when I click it but only once in a while. Normally when I press it nothing happens at all.
But like you the "click" field is missing

czikus, 2019-12-11:
I see no messages at all.

Koenkk, 2019-12-12:
This is a known bug in zigbee2mqtt 1.7.1, I just confirmed that it works fine with the latest dev.

To fix:
- Switch to the latest dev (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)
- Start zigbee2qmtt 
- Force remove the device (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove)
- Pair the device
- Make sure you see `Succesfully configured ....` for the device
- Should work now.

vzorglub, 2019-12-13:
Nope, same as before
Will try to flash the Stick with latest version today
I will need to re-pair all devices
How do I clear all devices?
Can I just delete the database file and clear the conf?

Koenkk, 2019-12-13:
You don't need to repair when reflashing, see FAQ: https://www.zigbee2mqtt.io/information/FAQ.html

pcb1962, 2019-12-14:
I can confirm that my issue with the Aqara button having no "click" field in the message is fixed in the latest build, thanks.

czikus, 2019-12-14:
I have now confirmed twice that updating to the latest dev (1.7.1+dev (commit #1972eb4)) makes absolutely no difference. Everything pairs and configures fine, but no messages from the dimmer afterwards. Not even without the "click" field, just nothing.

Do you have any other suggestions? Can I do anything to help debug the issue? Shouldn't it be re-opened btw?

> This is a known bug in zigbee2mqtt 1.7.1, I just confirmed that it works fine with the latest dev.
>
> To fix:
>
> * Switch to the latest dev [...]

czikus, 2019-12-14:
Of course, I re-paired. Here's the log:

```
info  2019-12-14 11:06:29: Device 'ikea_dimmer1' joined
info  2019-12-14 11:06:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"ikea_dimmer1"}}'
info  2019-12-14 11:06:29: Starting interview of 'ikea_dimmer1'
info  2019-12-14 11:06:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"ikea_dimmer1"}}'
info  2019-12-14 11:06:29: MQTT publish: topic 'zigbee2mqtt/ikea_dimmer1/availability', payload 'online'
debug 2019-12-14 11:06:33: Device 'ikea_dimmer1' announced itself
debug 2019-12-14 11:06:47: Received Zigbee message from 'ikea_dimmer1', type 'readResponse', cluster 'genBasic', data '{"modelId":"TRADFRI remote control","manufacturerName":"IKEA of Sweden","powerSource":3}' from endpoint 1 with gr
debug 2019-12-14 11:06:47: No converter available for 'E1524/E1810' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"TRADFRI remote control","manufacturerName":"IKEA of Sweden","powerSource":3}'
debug 2019-12-14 11:06:49: Received Zigbee message from 'ikea_dimmer1', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1,"appVersion":32,"stackVersion":87}' from endpoint 1 with groupID 0
debug 2019-12-14 11:06:49: No converter available for 'E1524/E1810' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1,"appVersion":32,"stackVersion":87}'
debug 2019-12-14 11:06:51: Received Zigbee message from 'ikea_dimmer1', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20190211","swBuildId":"1.2.223"}' from endpoint 1 with groupID 0
debug 2019-12-14 11:06:51: No converter available for 'E1524/E1810' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1,"dateCode":"20190211","swBuildId":"1.2.223"}'
info  2019-12-14 11:06:51: Successfully interviewed 'ikea_dimmer1', device has successfully been paired
info  2019-12-14 11:06:51: Device 'ikea_dimmer1' is supported, identified as: IKEA TRADFRI remote control (E1524/E1810)
info  2019-12-14 11:06:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"ikea_dimmer1","model":"E1524/E1810","vendor":"IKEA","description":"TRADFRI
info  2019-12-14 11:06:51: Configuring 'ikea_dimmer1'
info  2019-12-14 11:06:56: Succesfully configured 'ikea_dimmer1'
```

After that, nothing.

Koenkk, 2019-12-14:
@czikus indeed very strange, could you sniff the traffic while rotating the button? (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

czikus, 2019-12-14:
@Koenkk is there a way to do that without flashing the stick? I don't have a programmer to re-flash the stick.

Koenkk, 2019-12-14:
No, but you also need a second stick.

czikus, 2019-12-14:
@vzorglub, you said that you will flash the stick. Any luck? Any chance you might do the sniffing since you have the same problem and a programmer?

vzorglub, 2019-12-14:
I haven't flashed the stick with the latest firmware yet

vzorglub, 2019-12-14:
> No, but you also need to do this on a separate (second) stick.

I only have one stick so it won't be me solving this.
I'll flash my stick tomorrow and let you know
Thanks

vzorglub, 2019-12-15:
After reflashing the stick to the newest firware, I had to re-pair all my devices BUT the E1743 is now showing click events
YEAH!!

Thanks

czikus, 2019-12-15:
Seems that I need to get a programmer anyways then.

szappanyos, 2019-12-16:
@vzorglub  did you flash this firmware: https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20190608.zip ?
Because I am having the same issues and I can't solve it.

vzorglub, 2019-12-16:
Yes I flashed this firmware, force-removed all devices, cleared the database, cleared the conf file and restated. Re-paired and now it works.

szappanyos, 2019-12-17:
Thanks @vzorglub 
It works, but unfortunately, it required a lot of work.

flatsiedatsie, 2021-02-01:
I seem to have this issue too.

```
No converter available for 'E1743' with cluster 'genBasic' and type 'readResponse' and data '{\"manufacturerName\":\"IKEA of Sweden\"}'"
```

I get `Successfully interviewed '0xec1bbdfffaaaaaaaa', device has successfully been paired`

However, the only incoming messages I ever get are:

`{"linkquality":110,"update":{"state":"idle"},"update_available":false}`

Here's the full debug log:

```

{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{\"data\":{\"value\":true},\"status\":\"ok\"}'"}
{"data":{"value":true},"status":"ok"}
{"level":"debug","message":"Received MQTT message on 'zigbee2mqtt/bridge/config/devices' with data '[{\"dateCode\":\"20180507\",\"friendly_name\":\"Coordinator\",\"ieeeAddr\":\"0x00124b0018e2ab77\",\"lastSeen\":1612193151477,\"networkAddress\":0,\"softwareBuildID\":\"zStack12\",\"type\":\"Coordinator\"},{\"dateCode\":\"20190308\",\"description\":\"TRADFRI motion sensor\",\"friendly_name\":\"0x680ae2fffe39cc48\",\"hardwareVersion\":1,\"ieeeAddr\":\"0x680ae2fffe39cc48\",\"lastSeen\":1612192517896,\"manufacturerID\":4476,\"manufacturerName\":\"IKEA of Sweden\",\"model\":\"E1525/E1745\",\"modelID\":\"TRADFRI motion sensor\",\"networkAddress\":56188,\"powerSource\":\"Battery\",\"softwareBuildID\":\"2.0.022\",\"type\":\"EndDevice\",\"vendor\":\"IKEA\"},{\"description\":\"-\",\"friendly_name\":\"0x680ae2fffe1614a6\",\"ieeeAddr\":\"0x680ae2fffe1614a6\",\"lastSeen\":1612193087833,\"networkAddress\":48241,\"type\":\"Unknown\",\"vendor\":\"-\"}]'"}
{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{\"message\":[{\"dateCode\":\"20180507\",\"friendly_name\":\"Coordinator\",\"ieeeAddr\":\"0x00124b0018e2ab77\",\"lastSeen\":1612193151520,\"networkAddress\":0,\"softwareBuildID\":\"zStack12\",\"type\":\"Coordinator\"},{\"dateCode\":\"20190308\",\"description\":\"TRADFRI motion sensor\",\"friendly_name\":\"0x680ae2fffe39cc48\",\"hardwareVersion\":1,\"ieeeAddr\":\"0x680ae2fffe39cc48\",\"lastSeen\":1612192517896,\"manufacturerID\":4476,\"manufacturerName\":\"IKEA of Sweden\",\"model\":\"E1525/E1745\",\"modelID\":\"TRADFRI motion sensor\",\"networkAddress\":56188,\"powerSource\":\"Battery\",\"softwareBuildID\":\"2.0.022\",\"type\":\"EndDevice\",\"vendor\":\"IKEA\"},{\"description\":\"-\",\"friendly_name\":\"0x680ae2fffe1614a6\",\"ieeeAddr\":\"0x680ae2fffe1614a6\",\"lastSeen\":1612193087833,\"networkAddress\":48241,\"type\":\"Unknown\",\"vendor\":\"-\"}],\"type\":\"devices\"}'"}
{"message":[{"dateCode":"20180507","friendly_name":"Coordinator","ieeeAddr":"0x00124b0018e2ab77","lastSeen":1612193151520,"networkAddress":0,"softwareBuildID":"zStack12","type":"Coordinator"},{"dateCode":"20190308","description":"TRADFRI motion sensor","friendly_name":"0x680ae2fffe39cc48","hardwareVersion":1,"ieeeAddr":"0x680ae2fffe39cc48","lastSeen":1612192517896,"manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1525/E1745","modelID":"TRADFRI motion sensor","networkAddress":56188,"powerSource":"Battery","softwareBuildID":"2.0.022","type":"EndDevice","vendor":"IKEA"},{"description":"-","friendly_name":"0x680ae2fffe1614a6","ieeeAddr":"0x680ae2fffe1614a6","lastSeen":1612193087833,"networkAddress":48241,"type":"Unknown","vendor":"-"}],"type":"devices"}
{"level":"info","message":"Device '0xec1bbdfffaaaaaaaa' joined"}
{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{\"data\":{\"friendly_name\":\"0xec1bbdfffaaaaaaaa\",\"ieee_address\":\"0xec1bbdfffaaaaaaaa\"},\"type\":\"device_joined\"}'"}
{"data":{"friendly_name":"0xec1bbdfffaaaaaaaa","ieee_address":"0xec1bbdfffaaaaaaaa"},"type":"device_joined"}
{"level":"info","message":"Starting interview of '0xec1bbdfffaaaaaaaa'"}
{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{\"data\":{\"friendly_name\":\"0xec1bbdfffaaaaaaaa\",\"ieee_address\":\"0xec1bbdfffaaaaaaaa\",\"status\":\"started\"},\"type\":\"device_interview\"}'"}
{"data":{"friendly_name":"0xec1bbdfffaaaaaaaa","ieee_address":"0xec1bbdfffaaaaaaaa","status":"started"},"type":"device_interview"}
{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{\"message\":{\"friendly_name\":\"0xec1bbdfffaaaaaaaa\"},\"type\":\"device_connected\"}'"}
{"message":{"friendly_name":"0xec1bbdfffaaaaaaaa"},"type":"device_connected"}
{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{\"message\":\"interview_started\",\"meta\":{\"friendly_name\":\"0xec1bbdfffaaaaaaaa\"},\"type\":\"pairing\"}'"}
{"message":"interview_started","meta":{"friendly_name":"0xec1bbdfffaaaaaaaa"},"type":"pairing"}
{"level":"debug","message":"Device '0xec1bbdfffaaaaaaaa' announced itself"}
{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{\"data\":{\"friendly_name\":\"0xec1bbdfffaaaaaaaa\",\"ieee_address\":\"0xec1bbdfffaaaaaaaa\"},\"type\":\"device_announce\"}'"}
{"data":{"friendly_name":"0xec1bbdfffaaaaaaaa","ieee_address":"0xec1bbdfffaaaaaaaa"},"type":"device_announce"}
{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{\"message\":\"announce\",\"meta\":{\"friendly_name\":\"0xec1bbdfffaaaaaaaa\"},\"type\":\"device_announced\"}'"}
{"message":"announce","meta":{"friendly_name":"0xec1bbdfffaaaaaaaa"},"type":"device_announced"}
[... cut ...] um","values":["on","off","brightness_move_down","brightness_move_up","brightness_stop"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"E1743","vendor":"IKEA"},"friendly_name":"0xec1bbdfffaaaaaaaa","ieee_address":"0xec1bbdfffaaaaaaaa","status":"successful","supported":true},"type":"device_interview"}

```
This is with a CC2531 with extension cable on a Raspberry Pi 3B.

flatsiedatsie, 2021-02-01:
I tried with a different USB stick, the Conbee2. And now it does receive the messages. So it seems the pre-flashed Sonoff CC2531 USB stick isn't 100% compatible with Zigbee2MQTT.

Koenkk, 2021-02-01:
@flatsiedatsie probably outdated firmware, make sure you run the latest from: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default

flatsiedatsie, 2021-02-01:
Yes, undoubtedly.

Updating firmware on CC2531 sticks isn't an option for my use case though. I was hoping that my end-users could buy the pre-flashed Sonoff variety and be done. Unfortunately, that doesn't seem to be the case.