Koenkk/zigbee2mqtt #2345: IKEA LED1737R5 bulbs are not pairing

kiall, 2019-11-15:
# Bug Report
## What happened

IKEA LED1737R5 bulbs are not pairing to CC26x2R with 20191106 firmware. No log messages whatsoever after factory resetting and power cycling the bulb.

* log_level is set to `debug` (and debug messages are showing)
* DEBUG env var is set to `zigbee-herdsman*`, and herdsman messages are showing.
* permit_join is set to `true`
* Older devices of mine are pairing (i.e IKEA remotes and other bulbs)
* Absolutely nothing is happening when attempting to pair my LED1737R5 bulbs, no log messages at all.

## What did you expect to happen

Something .. *anything* :D

## Debug Info

zigbee2mqtt version: 1.7.0 and 1.7.1
CC26X2R1 firmware version: Z-Stack_3.x.0 ver 20191106

# configuration.yaml

    homeassistant: true
    permit_join: false
    mqtt:
      base_topic: zigbee2mqtt
      server: 'mqtt://172.29.0.117'
      include_device_information: true
    serial:
      port: /dev/ttyACM0
    advanced:
      channel: 11
      cache_state: true
      log_level: debug
      homeassistant_discovery_topic: homeassistant
      homeassistant_status_topic: hass/status
      last_seen: epoch
    devices:
      '0x14b457fffe3e3ac4':
        friendly_name: '0x14b457fffe3e3ac4'
      '0x90fd9ffffe515ff1':
        friendly_name: '0x90fd9ffffe515ff1'
      '0x000d6ffffe4d6da8':
        friendly_name: '0x000d6ffffe4d6da8'
    groups: {}

Koenkk, 2019-11-16:
- Is there any other zigbee network running? If yes, it could be that the bulb pairs to that
- Can you share the log when pairing the device when running with `DEBUG=zigbee-herdsman:*`

kiall, 2019-11-19:
> Is there any other zigbee network running? If yes, it could be that the bulb pairs to that

No, I even removed all the batteries from anything that used them

> Can you share the log when pairing the device when running with DEBUG=zigbee-herdsman:*

It's completely silent :) I've erased the CC26X2R1 + reflashed, removed all z2m config and state, started z2m with `log_level: debug`, `premit_join: true`, and `DEBUG=zigbee-herdsman*`, factory reset the bulb - and the log remains silent.

I paired two IKEA plugs, and 1 other older IKEA bulb to make sure things were working - nothing at all in the logs for the next 3 minutes - after that, I had the periodic `mgmtPermitJoinReq` and an `attributeReport` from one of the IKEA plugs.

I've included the log anyway below. The large gap is where I mashed enter a few times to make sure I would see the bulb joining.

	zigbee2mqtt    | zigbee2mqtt:debug 2019-11-19T10:24:23: Saving state to file /app/data/state.json
	zigbee2mqtt    | 2019-11-19T10:24:27.988Z zigbee-herdsman:controller:log Permit joining
	zigbee2mqtt    | 2019-11-19T10:24:27.990Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}
	zigbee2mqtt    | 2019-11-19T10:24:27.991Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228]
	zigbee2mqtt    | 2019-11-19T10:24:27.999Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82,254,3]
	zigbee2mqtt    | 2019-11-19T10:24:27.999Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82,254,3]
	zigbee2mqtt    | 2019-11-19T10:24:28.000Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [] - 82
	zigbee2mqtt    | 2019-11-19T10:24:28.000Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0}
	zigbee2mqtt    | 2019-11-19T10:24:28.001Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3]
	zigbee2mqtt    | 2019-11-19T10:24:28.003Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [69,182,0,0,0,240]
	zigbee2mqtt    | 2019-11-19T10:24:28.003Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240]
	zigbee2mqtt    | 2019-11-19T10:24:28.004Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [] - 240
	zigbee2mqtt    | 2019-11-19T10:24:28.004Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0}
	zigbee2mqtt    | 2019-11-19T10:24:28.005Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

	
	zigbee2mqtt    | 2019-11-19T10:27:47.996Z zigbee-herdsman:controller:log Permit joining
	zigbee2mqtt    | 2019-11-19T10:27:47.997Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}
	zigbee2mqtt    | 2019-11-19T10:27:47.999Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228]
	zigbee2mqtt    | 2019-11-19T10:27:48.014Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82,254,3]
	zigbee2mqtt    | 2019-11-19T10:27:48.015Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82,254,3]
	zigbee2mqtt    | 2019-11-19T10:27:48.015Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [] - 82
	zigbee2mqtt    | 2019-11-19T10:27:48.016Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0}
	zigbee2mqtt    | 2019-11-19T10:27:48.016Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3]
	zigbee2mqtt    | 2019-11-19T10:27:48.019Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [69,182,0,0,0,240]
	zigbee2mqtt    | 2019-11-19T10:27:48.020Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240]
	zigbee2mqtt    | 2019-11-19T10:27:48.020Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [] - 240
	zigbee2mqtt    | 2019-11-19T10:27:48.021Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0}
	zigbee2mqtt    | 2019-11-19T10:27:48.022Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
	zigbee2mqtt    | zigbee2mqtt:debug 2019-11-19T10:29:23: Saving state to file /app/data/state.json
	zigbee2mqtt    | 2019-11-19T10:30:11.040Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,226,141,0,237]
	zigbee2mqtt    | 2019-11-19T10:30:11.040Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,226,141,0,237]
	zigbee2mqtt    | 2019-11-19T10:30:11.041Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [�] - 237
	zigbee2mqtt    | 2019-11-19T10:30:11.041Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":36322,"relaycount":0,"relaylist":[]}
	zigbee2mqtt    | 2019-11-19T10:30:11.042Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
	zigbee2mqtt    | 2019-11-19T10:30:11.089Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0]
	zigbee2mqtt    | 2019-11-19T10:30:11.089Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0]
	zigbee2mqtt    | 2019-11-19T10:30:11.091Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [226,141,1,1,0,39,0,118,223,85,0,0,7,8,5,10,0,0,16,1,226,141,29,15]
	zigbee2mqtt    | 2019-11-19T10:30:11.092Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,226,141,1,1,0,39,0,118,223,85,0,0,7,8,5,10,0,0,16,1,226,141,29,15]
[... cut ...]
	zigbee2mqtt    | 2019-11-19T10:31:08.020Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [69,182,0,0,0,240]
	zigbee2mqtt    | 2019-11-19T10:31:08.021Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240]
	zigbee2mqtt    | 2019-11-19T10:31:08.021Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [] - 240
	zigbee2mqtt    | 2019-11-19T10:31:08.022Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0}
	zigbee2mqtt    | 2019-11-19T10:31:08.023Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

kiall, 2019-11-19:
I have now managed to get these to pair by ... turning off my wifi access points?

I've been using Zigbee channels 24, in order to avoid interference with my wifi APs which are on wifi channels 1 and 6 @ 40Mz wide channels. So, now I'm doubting my channel planning?

Wifi 1 & 6 @ 40Mhz is effectively consuming 2401-2443Mhz for ch 1 and 2461-2458Mhz for channel 6.

Zigbee ch 24 is effectively consuming 2467.5 - 2472.5 MHz.

Does this look correct? Any hints as to what I might have wrong in my logic heres?

(Yes, there is an ch 11 in the config paste above, that was last ditch attempt testing ;))

Anyway - They paired successfully, and I have some more reading to do on wifi+ZB channel overlap and interference.  

	zigbee2mqtt    | zigbee2mqtt:info  2019-11-19T11:13:36: Device '0x14b457fffe68575f' joined
	zigbee2mqtt    | zigbee2mqtt:info  2019-11-19T11:13:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x14b457fffe68575f"}}'
	zigbee2mqtt    | zigbee2mqtt:info  2019-11-19T11:13:36: Starting interview of '0x14b457fffe68575f'
	zigbee2mqtt    | zigbee2mqtt:info  2019-11-19T11:13:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x14b457fffe68575f"}}'
	zigbee2mqtt    | zigbee2mqtt:debug 2019-11-19T11:13:40: Device '0x14b457fffe68575f' announced itself
	zigbee2mqtt    | zigbee2mqtt:debug 2019-11-19T11:13:47: Received Zigbee message from '0x14b457fffe68575f', type 'readResponse', cluster 'genBasic', data '{"modelId":"TRADFRI bulb GU10 WS 400lm","manufacturerName":"IKEA of Sweden","powerSource":1}' from endpoint 1 with groupID 0
	zigbee2mqtt    | zigbee2mqtt:debug 2019-11-19T11:13:47: No converter available for 'LED1537R6' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"TRADFRI bulb GU10 WS 400lm","manufacturerName":"IKEA of Sweden","powerSource":1}'
	zigbee2mqtt    | zigbee2mqtt:info  2019-11-19T11:13:47: MQTT publish: topic 'homeassistant/light/0x14b457fffe68575f/light/config', payload '{"brightness":true,"color_temp":true,"schema":"json","command_topic":"zigbee2mqtt/0x14b457fffe68575f/set","state_topic":"zigbee2mqtt/0x14b457fffe68575f","name":"0x14b457fffe68575f_light","unique_id":"0x14b457fffe68575f_light_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x14b457fffe68575f"],"name":"0x14b457fffe68575f","sw_version":"Zigbee2mqtt 1.7.1","model":"TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (LED1537R6)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
	zigbee2mqtt    | zigbee2mqtt:info  2019-11-19T11:13:47: MQTT publish: topic 'homeassistant/sensor/0x14b457fffe68575f/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x14b457fffe68575f","json_attributes_topic":"zigbee2mqtt/0x14b457fffe68575f","name":"0x14b457fffe68575f_linkquality","unique_id":"0x14b457fffe68575f_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x14b457fffe68575f"],"name":"0x14b457fffe68575f","sw_version":"Zigbee2mqtt 1.7.1","model":"TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (LED1537R6)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
	zigbee2mqtt    | zigbee2mqtt:debug 2019-11-19T11:13:47: Received Zigbee message from '0x14b457fffe68575f', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2,"appVersion":32,"stackVersion":98}' from endpoint 1 with groupID 0
	zigbee2mqtt    | zigbee2mqtt:debug 2019-11-19T11:13:47: No converter available for 'LED1537R6' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":2,"appVersion":32,"stackVersion":98}'
	zigbee2mqtt    | zigbee2mqtt:debug 2019-11-19T11:13:47: Received Zigbee message from '0x14b457fffe68575f', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20190308","swBuildId":"2.0.022"}' from endpoint 1 with groupID 0
	zigbee2mqtt    | zigbee2mqtt:debug 2019-11-19T11:13:47: No converter available for 'LED1537R6' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1,"dateCode":"20190308","swBuildId":"2.0.022"}'
	zigbee2mqtt    | zigbee2mqtt:info  2019-11-19T11:13:47: Successfully interviewed '0x14b457fffe68575f', device has successfully been paired
	zigbee2mqtt    | zigbee2mqtt:info  2019-11-19T11:13:47: Device '0x14b457fffe68575f' is supported, identified as: IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (LED1537R6)
	zigbee2mqtt    | zigbee2mqtt:info  2019-11-19T11:13:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x14b457fffe68575f","model":"LED1537R6","vendor":"IKEA","description":"TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum","supported":true}}'

Koenkk, 2019-11-19:
Interference could indeed be the cause of this, this article explains it well: https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html

Please close this issue if your issue has been solved.

closed 2019-11-25 by kiall